raspberrypi/linux #1851: bluetooth keyboard does not work on raspberry pi 3

sassmann, 2017-02-19:
Trying to use a bluetooth keyboard with RPI 3.

root@kodipi:~ > uname -a
Linux kodipi 4.4.41 #1 SMP Wed Jan 11 15:08:33 CET 2017 armv7l GNU/Linux
root@kodipi:~ > bluetoothctl
[NEW] Controller B8:27:EB:8D:F7:E7 BlueZ 5.43 [default]
[bluetooth]# agent on
Agent registered
[bluetooth]# default-agent
Default agent request successful
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:8D:F7:E7 Discovering: yes
[NEW] Device 50:73:00:06:89:0D Bluetooth 3.0 Keyboard
[NEW] Device 5C:F6:DC:99:67:D0 TVBluetooth
[CHG] Device 6C:94:F8:D1:70:59 RSSI: -83
[bluetooth]# pair 50:73:00:06:89:0D
Attempting to pair with 50:73:00:06:89:0D
[CHG] Device 50:73:00:06:89:0D Connected: yes
[agent] PIN code: 448509
[CHG] Device 50:73:00:06:89:0D ServicesResolved: yes
[CHG] Device 50:73:00:06:89:0D Paired: yes
Pairing successful
[CHG] Device 50:73:00:06:89:0D ServicesResolved: no
[CHG] Device 50:73:00:06:89:0D Connected: no
[CHG] Device 50:73:00:06:89:0D Connected: yes
[Bluetooth 3.0 Keyboard]# quit

However no input recognized when typing on the keyboard by the RPI3.
dmesg shows the following errors.
[  656.277068] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.277166] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278160] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278198] Bluetooth: hci0 SCO packet for unknown connection handle 3
[  656.278223] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278277] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278357] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278443] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278530] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278617] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278703] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278791] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278970] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.278980] Bluetooth: hci0: Frame reassembly failed (-84)
[  656.279938] Bluetooth: hci0: Frame reassembly failed (-84)
[  694.010607] Bluetooth: hci0 command 0x2005 tx timeout
[  696.017272] Bluetooth: hci0 command 0x200b tx timeout
[  696.078998] Bluetooth: hci0: Frame reassembly failed (-90)
[  696.079024] Bluetooth: hci0: Frame reassembly failed (-84)
[  696.081156] Bluetooth: hci0: Frame reassembly failed (-84)
[  698.023918] Bluetooth: hci0 command 0x200c tx timeout
[  698.245320] Bluetooth: hci0: Frame reassembly failed (-84)

thejpster, 2017-03-12:
I get the same `hc0: Frame reassembly failed (-84)` error using a Dual Shock 3 with a Pi Zero W, as soon as I connect to the `/dev/input/js0` device.

JamesH65, 2017-05-18:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.
Please update to the latest kernel which may contain a fix for this issue.

sassmann, 2017-05-18:
Same problem still exists with kernel 4.9.

thejpster, 2017-07-01:
Works fine for me now on Kernel 4.9.

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65