raspberrypi/linux #1853: Kernel trace while deregistering 8812cu based Wi-Fi dongles

staroselskii, 2017-02-22:
There's two ways to reproduce:

1) Attach a dongle with the aforementioned chipset 
2) modprobe -r 8812cu

or

1) Attach a dongle with the aforementioned chipset 
2) Unplug the device

Either one yields the same kernel trace shown below:

```
[  670.735196] usbcore: deregistering interface driver rtl8192cu
[  670.736869] rtw_cmd_thread: DriverStopped(1) SurpriseRemoved(0) break at line 482
[  670.772394] ------------[ cut here ]------------
[  670.772597] WARNING: CPU: 3 PID: 5252 at net/wireless/sme.c:907 __cfg80211_disconnected+0x478/0x4b8 [cfg80211]()
[  670.772610] Modules linked in: 8812au(O) binfmt_misc rtl8xxxu mac80211 8192cu(-) brcmfmac brcmutil cfg80211 joydev evdev rfkill bcm2835_wdt spi_bcm2835aux spi_bcm2835 i2c_bcm2708 bcm2835_gpiomem uio_pdrv_genirq uio i2c_dev ipv6
[  670.772731] CPU: 3 PID: 5252 Comm: modprobe Tainted: G        W  O    4.4.36-da925b7-emlid-v7+ #29
[  670.772743] Hardware name: BCM2709
[  670.772781] [<800194b4>] (unwind_backtrace) from [<80014b30>] (show_stack+0x20/0x24)
[  670.772809] [<80014b30>] (show_stack) from [<80328d9c>] (dump_stack+0xbc/0x108)
[  670.772836] [<80328d9c>] (dump_stack) from [<800265e4>] (warn_slowpath_common+0x8c/0xc8)
[  670.772862] [<800265e4>] (warn_slowpath_common) from [<800266dc>] (warn_slowpath_null+0x2c/0x34)
[  670.773029] [<800266dc>] (warn_slowpath_null) from [<7f16c840>] (__cfg80211_disconnected+0x478/0x4b8 [cfg80211])
[  670.773342] [<7f16c840>] (__cfg80211_disconnected [cfg80211]) from [<7f141860>] (cfg80211_process_wdev_events+0x19c/0x1a4 [cfg80211])
[  670.773648] [<7f141860>] (cfg80211_process_wdev_events [cfg80211]) from [<7f13e5a0>] (cfg80211_netdev_notifier_call+0x31c/0x660 [cfg80211])
[  670.773815] [<7f13e5a0>] (cfg80211_netdev_notifier_call [cfg80211]) from [<80043dcc>] (notifier_call_chain+0x54/0x94)
[  670.773840] [<80043dcc>] (notifier_call_chain) from [<80043f30>] (raw_notifier_call_chain+0x28/0x30)
[  670.773865] [<80043f30>] (raw_notifier_call_chain) from [<804dd440>] (call_netdevice_notifiers_info+0x40/0x68)
[  670.773891] [<804dd440>] (call_netdevice_notifiers_info) from [<804e0f08>] (rollback_registered_many+0x188/0x318)
[  670.773915] [<804e0f08>] (rollback_registered_many) from [<804e10d0>] (rollback_registered+0x38/0x50)
[  670.773940] [<804e10d0>] (rollback_registered) from [<804e401c>] (unregister_netdevice_queue+0x64/0xbc)
[  670.773964] [<804e401c>] (unregister_netdevice_queue) from [<804e409c>] (unregister_netdev+0x28/0x30)
[  670.774138] [<804e409c>] (unregister_netdev) from [<7f2d9c80>] (rtw_unregister_netdevs+0x5c/0x7c [8192cu])
[  670.774439] [<7f2d9c80>] (rtw_unregister_netdevs [8192cu]) from [<7f2dac2c>] (rtw_dev_remove+0x2c/0x7c [8192cu])
[  670.774603] [<7f2dac2c>] (rtw_dev_remove [8192cu]) from [<80426718>] (usb_unbind_interface+0x80/0x278)
[  670.774629] [<80426718>] (usb_unbind_interface) from [<803b503c>] (__device_release_driver+0x94/0x110)
[  670.774653] [<803b503c>] (__device_release_driver) from [<803b51cc>] (driver_detach+0xbc/0xc0)
[  670.774675] [<803b51cc>] (driver_detach) from [<803b4190>] (bus_remove_driver+0x6c/0xe4)
[  670.774697] [<803b4190>] (bus_remove_driver) from [<803b5618>] (driver_unregister+0x38/0x58)
[  670.774717] [<803b5618>] (driver_unregister) from [<80425a3c>] (usb_deregister+0x70/0xe0)
[  670.774876] [<80425a3c>] (usb_deregister) from [<7f2eaed8>] (rtw_drv_halt+0x1c/0x28 [8192cu])
[  670.775049] [<7f2eaed8>] (rtw_drv_halt [8192cu]) from [<8009a518>] (SyS_delete_module+0x188/0x1f8)
[  670.775073] [<8009a518>] (SyS_delete_module) from [<80010640>] (ret_fast_syscall+0x0/0x1c)
[  670.775087] ---[ end trace bc140751e8952713 ]---
[  670.775100] ------------[ cut here ]------------
[  670.775255] WARNING: CPU: 3 PID: 5252 at net/wireless/core.c:1134 cfg80211_netdev_notifier_call+0x334/0x660 [cfg80211]()
[  670.775265] Modules linked in: 8812au(O) binfmt_misc rtl8xxxu mac80211 8192cu(-) brcmfmac brcmutil cfg80211 joydev evdev rfkill bcm2835_wdt spi_bcm2835aux spi_bcm2835 i2c_bcm2708 bcm2835_gpiomem uio_pdrv_genirq uio i2c_dev ipv6
[  670.775376] CPU: 3 PID: 5252 Comm: modprobe Tainted: G        W  O    4.4.36-da925b7-emlid-v7+ #29
[  670.775387] Hardware name: BCM2709
[  670.775411] [<800194b4>] (unwind_backtrace) from [<80014b30>] (show_stack+0x20/0x24)
[  670.775435] [<80014b30>] (show_stack) from [<80328d9c>] (dump_stack+0xbc/0x108)
[  670.775460] [<80328d9c>] (dump_stack) from [<800265e4>] (warn_slowpath_common+0x8c/0xc8)
[  670.775484] [<800265e4>] (warn_slowpath_common) from [<800266dc>] (warn_slowpath_null+0x2c/0x34)
[  670.775646] [<800266dc>] (warn_slowpath_null) from [<7f13e5b8>] (cfg80211_netdev_notifier_call+0x334/0x660 [cfg80211])
[  670.775811] [<7f13e5b8>] (cfg80211_netdev_notifier_call [cfg80211]) from [<80043dcc>] (notifier_call_chain+0x54/0x94)
[  670.775836] [<80043dcc>] (notifier_call_chain) from [<80043f30>] (raw_notifier_call_chain+0x28/0x30)
[  670.775859] [<80043f30>] (raw_notifier_call_chain) from [<804dd440>] (call_netdevice_notifiers_info+0x40/0x68)
[  670.775883] [<804dd440>] (call_netdevice_notifiers_info) from [<804e0f08>] (rollback_registered_many+0x188/0x318)
[  670.775906] [<804e0f08>] (rollback_registered_many) from [<804e10d0>] (rollback_registered+0x38/0x50)
[  670.775930] [<804e10d0>] (rollback_registered) from [<804e401c>] (unregister_netdevice_queue+0x64/0xbc)
[  670.775954] [<804e401c>] (unregister_netdevice_queue) from [<804e409c>] (unregister_netdev+0x28/0x30)
[  670.776113] [<804e409c>] (unregister_netdev) from [<7f2d9c80>] (rtw_unregister_netdevs+0x5c/0x7c [8192cu])
[  670.776411] [<7f2d9c80>] (rtw_unregister_netdevs [8192cu]) from [<7f2dac2c>] (rtw_dev_remove+0x2c/0x7c [8192cu])
[  670.776574] [<7f2dac2c>] (rtw_dev_remove [8192cu]) from [<80426718>] (usb_unbind_interface+0x80/0x278)
[  670.776597] [<80426718>] (usb_unbind_interface) from [<803b503c>] (__device_release_driver+0x94/0x110)
[  670.776619] [<803b503c>] (__device_release_driver) from [<803b51cc>] (driver_detach+0xbc/0xc0)
[  670.776641] [<803b51cc>] (driver_detach) from [<803b4190>] (bus_remove_driver+0x6c/0xe4)
[  670.776662] [<803b4190>] (bus_remove_driver) from [<803b5618>] (driver_unregister+0x38/0x58)
[  670.776682] [<803b5618>] (driver_unregister) from [<80425a3c>] (usb_deregister+0x70/0xe0)
[  670.776840] [<80425a3c>] (usb_deregister) from [<7f2eaed8>] (rtw_drv_halt+0x1c/0x28 [8192cu])
[  670.777002] [<7f2eaed8>] (rtw_drv_halt [8192cu]) from [<8009a518>] (SyS_delete_module+0x188/0x1f8)
[  670.777024] [<8009a518>] (SyS_delete_module) from [<80010640>] (ret_fast_syscall+0x0/0x1c)
[  670.777036] ---[ end trace bc140751e8952714 ]---
```

Even though, it looks scary one might reconnect the device and it will happily work ever after. But something is seriously wrong in here, isn't it?

I tried to look into the issue but to no avail yet. I guess there's a place where a handle was not released but I couldn't find it from the brief investigation. I hope someone more familiar with the Linux wireless stack's internals might want to take a look.

JamesH65, 2017-05-18:
Please update to the latest kernel which may contain a fix for this issue and report back.

staroselskii, 2017-05-20:
@JamesH65 
As a matter of fact I had wanted to share that I tested the same setup on `4.9.27-v7+` and unfortunately the issue persists. If you could point out the commit where the problem might have been fixed, I'd be more than happy to check it out!

JamesH65, 2017-05-22:
Your version already has the current set of fixes, so looks like this issue is still present. 

You could try the very latest 4.11 kernel (you will need to build it yourself from our github repository). There have been quite a few changes from upstream. That might be a useful data point.

JamesH65, 2017-09-13:
ping @staroselskii Any chance to test the latest kernels?

staroselskii, 2017-09-13:
Yeah, sure. I'll try it to have it tested by the end of the week.

JamesH65, 2018-06-28:
@staroselskii Did you have any luck with the most recent kernel?

staroselskii, 2018-06-28:
@james4765 

I haven't tested it on the latest. I can try it next week and get back to you.

But the 4.11 didn't fix it.