raspberrypi/linux #1857: backport patch for cve_2017_6074

arrfab, 2017-02-24:
Can we have a rebase to 4.4.51 in the rpi-4.4.y branch to be sure we have the needed patch for that CVE ? (dccp issue)
Probably the same for the new 4.9.x branch ? (4.9.12 if I'm correct).
Also, is now the "default" kernel for rpi2/3 the 4.9.x LTS one ?

6by9, 2017-02-24:
CONFIG_IP_DCCP is not set in the standard configs, so there is no issue for the vast majority of users. If you're building your own kernel with it enabled then you may as well cherry-pick that patch yourself.
The 4.9 branch will get bumped in the normal timeframes.

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934&start=75#p1117268 - rpi-update now provides a 4.9.x kernel for all Pi platforms.

arrfab, 2017-02-24:
Thanks for the answer .. I found CONFIG_NF_CT_PROTO_DCCP=m in https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/configs/bcm2709_defconfig and was wondering.

Wrt 4.9.x available now : does that mean that the rpi-4.4.y branch will never have any update anymore ?

6by9, 2017-02-24:
CONFIG_NF_CT_PROTO_DCCP - https://github.com/raspberrypi/linux/blob/rpi-4.4.y/net/netfilter/Kconfig#L144. Not the same module as affected by the bug.

You also have CONFIG_NETFILTER_XT_MATCH_DCCP=m, but that again is not the module affected - https://github.com/raspberrypi/linux/blob/rpi-4.4.y/net/netfilter/Kconfig#L1077

Not my call, but 4.4 is unlikely to get updated unless there is a very good reason or someone else takes on the task. Look at the 4.1 branch for comparison when 4.4 was adopted in mid-Feb 2016.
Pull requests will generally still be accepted, so anyone can take on the role.

betafive, 2017-03-03:
I'm currently looking at the feasibility of maintaining rpi-4.4.y unofficially in a linux-raspberrypi-lts repository until 4.4.y is EOL upstream. I've already merged up to 4.4.52 and will be testing it in the next couple of days. If you want to try it out, see https://gitlab.com/toganlabs/linux-raspberrypi-lts/tree/pbarker/merge-stable

arrfab, 2017-03-03:
Thanks, in fact in the mean time we've discussed (within the CentOS Project) to have a rebase to 4.9.x (LTS) for all our armhfp boards, including raspberrypi2/3 and we built 4.9.13 already.
More informations for the centos build/pkg here : https://lists.centos.org/pipermail/arm-dev/2017-February/002762.html

betafive, 2017-03-17:
@arrfab: We've now released the first version of our linux-raspberrypi-lts project: https://www.toganlabs.com/downloads/linux-raspberrypi-lts/rpi-4.4.53/RELEASE_NOTES.txt

We've also put up a blog post about our plans for this: https://toganlabs.wordpress.com/2017/03/16/announcing-linux-raspberrypi-lts/

Feel free to contact us if you have any further interest in this.

closed 2017-03-17 by popcornmix