raspberrypi/linux #1865: MAC Address Setting Appears Broken 4.9.13

Whobeu, 2017-02-28:
I run a feed to FlightAware and use the smsc95xx.macaddr setting in /boot/cmdline.txt to set the MAC address to b8:27:eb:89:93:xx which was the address of a Raspberry Pi 1 that I used to use (now using a Raspberry Pi 3). I do this because FlightAware uniquely defines our site numbers based upon our MAC address so changing my MAC address to the address of my original setup preserves my settings at FlightAware.

Under 4.9.11 everything works fine. Yesterday I upgraded to 4.9.13 and learned last night that my "site" was not feeding. Upon logging in to FlightAware I discovered a new site had been registered under my account with a MAC address of b8:27:eb:12:45:xx. I downgraded the kernel back to 4.9.11 this morning and the b8:27:eb:89:93:xx MAC address is working again. I never noted the real MAC address of my Raspberry Pi 3 but, regardless, it appears that in 4.9.13 the smsc95xx.macaddr setting is not working.

pelwell, 2017-02-28:
"MAC address overriding broken in 4.9.13" would be a more accurate title, but otherwise you are correct. That command line parameter was added by a downstream patch, but now that the smsc95xx driver can read its MAC address from the Device Tree the patch is no longer required. If the firmware detects a compatible .dtb file it will populate the local-mac-address property and not pass in the smsc95xx.macaddr parameter. However, this removes a mechanism to force the MAC address to a value of the user's choosing.

I'll get the firmware to read smsc95xx.macaddr from the cmdline.txt and use it as the preferred source of the Pi's MAC address, falling back to the value derived from the serial number when necessary. In the meantime you could, if you wanted, take the firmware from the 4.9.11 release and use it with the 4.9.13 kernel, but I'm not sure why you would.

kukabu, 2017-03-01:
@pelwell I think right way sets the mac-address property via DTB overlay.
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/Documentation/devicetree/bindings/net/ethernet.txt#L5

pelwell, 2017-03-01:
I would switch to that now, but the current dtparam mechanism doesn't have a syntax for an array of bytes. I could create 6 parameters - one per byte - but that's a bit ugly. For now I'll restore the ability to read it from the command line and think about a neater DT solution.

pelwell, 2017-03-01:
Can you try [this firmware](https://drive.google.com/file/d/0B_P-i4u-SLBXakdWa3lXNGZjaVU/view?usp=sharing) (elfs and dats)? It implements the change outlined above (checking for the module parameter in cmdline.txt and using that MAC address if present), with one addition - if the DTB (after merging overlays) includes a valid, non-zero local-mac-address property it doesn't overwrite it.

Whobeu, 2017-03-01:
This new firmware is working correctly.  I updated to 4.9.13 using rpi-update and then added these files in before reboot. The MAC address is now being set correctly to my cmdline.txt specified address.  I would consider this issue closed.

pelwell, 2017-03-01:
Let's keep it open until the fix makes it to the rpi-update channel, but thanks for the feedback.

aeruder, 2017-03-02:
Confirmed working here (running raspberrypi/firmware master)

closed 2017-03-02 by pelwell