raspberrypi/linux #1874: cc (Raspbian 4.8.2-21~rpi3rpi1) 4.8.2 Internal error when building MariaDB

Napoleon-BlownApart, 2017-03-05:
I'm attempting to build MariaDB from their git repository (10.1.22) on my Raspberry Pi 1 B+ (Wheezy).  After installing/building all the dependencies, the cmake eventually worked with the command `cmake . -DCMAKE_EXE_LINKER_FLAGS='-ljemalloc'  -DWITH_SAFEMALLOC=OFF`  ([cmake log available here](https://abms.net.au/public/mariadb__cmake.txt)).  But during the make, `cc (Raspbian 4.8.2-21~rpi3rpi1) 4.8.2` failed with the following error ([make log available here](https://abms.net.au/public/mariadb__make.txt)):

    /home/user/Downloads/mariadb/server/storage/mroonga/vendor/groonga/lib/expr.c:6816:7: note: in expansion of macro ‘GRN_PTR_POP’
       GRN_PTR_POP(&keywords, keyword);
       ^
    cc: internal compiler error: Killed (program cc1)
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
    storage/mroonga/vendor/groonga/lib/CMakeFiles/libgroonga.dir/build.make:254: recipe for target 'storage/mroonga/vendor/groonga/lib/CMakeFiles/libgroonga.dir/expr.c.o' failed
    make[2]: *** [storage/mroonga/vendor/groonga/lib/CMakeFiles/libgroonga.dir/expr.c.o] Error 4
    CMakeFiles/Makefile2:4417: recipe for target 'storage/mroonga/vendor/groonga/lib/CMakeFiles/libgroonga.dir/all' failed
    make[1]: *** [storage/mroonga/vendor/groonga/lib/CMakeFiles/libgroonga.dir/all] Error 2
    Makefile:149: recipe for target 'all' failed
    make: *** [all] Error 2

If more information is needed or I should be posting about this elsewhere, I would appreciated any assistance given as I haven't done this before.

ED6E0F17, 2017-03-05:
This is totally the wrong place, but:
````
-- Checking for module 'libzmq'
--   No package 'libzmq' found
````
Installing libzmg (dev?) should solve a lot of your problems, otherwise you need to upgrade your compiler.

popcornmix, 2017-03-05:
This repo if for linux kernel bugs. This isn't one. The [forum](https://www.raspberrypi.org/forums) would be a better place to post.
cc1 being killed may be an out of memory issue. Check dmesg for OOM message.
``gpu_mem=16``, increasing swap and reducing parallel jobs in make may help.

closed 2017-03-05 by popcornmix