raspberrypi/linux #1875: Please re-enable LZ4 in 4.9 kernel config

jayslovak, 2017-03-06:
Hi all,

I was using LZ4 compression for SWAP compression. This has stopped working after kernel upgrade to 4.9 because the functionality is no longer present:
```
diff 4.4.txt 4.9.txt
4d3
< CONFIG_ZRAM_LZ4_COMPRESS=y
8d6
< CONFIG_LZ4_COMPRESS=m
```
Please re-enable these options.

Thank you.

ED6E0F17, 2017-03-07:
The required option is:
````
config CRYPTO_LZ4
	tristate "LZ4 compression algorithm"
	select CRYPTO_ALGAPI
	select LZ4_COMPRESS
	select LZ4_DECOMPRESS
	help
	  This is the LZ4 algorithm.
````
It would be good to have this in the kernel.

pelwell, 2017-03-07:
Yes, it appears that as of 4.8 the ZRAM module uses the compression functions provided by the crypto module, and enabling CRYPTO_LZ4 just causes two small modules to be built:
```
-rw-rw-r-- 1 phil phil 6199 Mar  7 16:00 ./lib/lz4/lz4_compress.ko
-rw-rw-r-- 1 phil phil 5879 Mar  7 16:00 ./crypto/lz4.ko
```
@popcornmix Shall I add this config option?

jayslovak, 2017-03-07:
@pelwell, @ED6E0F17 are you sure we don't need ZRAM_LZ4_COMPRESS=y as well?

Thanks for looking into this.

pelwell, 2017-03-07:
defconfigs should only include the absolute minimum number of settings required to achieve a given configuration - any defaults, or settings which are implied by others, are omitted. Fortunately the savedefconfig make rule takes care of the pruning for us.

In this case, as @ED6E0F17's post above shows, setting CRYPTO_LZ4 automatically selects LZ4_COMPRESS, so it shouldn't appear in the defconfig.

popcornmix, 2017-03-07:
@pelwell no objections.

closed 2017-03-07 by pelwell

monetny, 2017-05-23:
add CRYPTO_LZ4HC=m too, because lz4hc faster then lzo, and better then lz4
(speed decompress lzo<lz4hc=lz4)

eccgecko, 2018-05-20:
Is this this able be committed to the the arch/arm64/configs/bcmrpi3_defconfig also? [https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm64/configs/bcmrpi3_defconfig]