raspberrypi/linux #1876: Gadget serial locks up.

ED6E0F17, 2017-03-06:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934&start=179

When using g_serial, or telnet over g_ether, the tty will lock up on certain text strings. (tested on Linux 4.9)
"dmesg|head" or "ls" tend to fail, while "dmesg|tail" or "ls -l" may succeed. This issue goes away when using ssh over g_ether.

This will be an upstream bug, since it affects the dwc2 USB driver.

phylter, 2017-03-07:
After updating to 4.9 none of the gadgets that I've tried (g_serial or g_ether) seem to work correctly. This is with a configuration that worked correctly before the update. It seems that they're no longer passing any data over them. The device functions as normal but the gadget interface doesn't.  

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934&start=177

ED6E0F17, 2017-03-08:
This is a "known issue": https://www.spinics.net/lists/linux-usb/msg150016.html

There may be a solution in 4.11 that would be better than reverting the broken commit.

closed 2017-03-13 by pelwell