raspberrypi/linux #1877: Ethernet Driver does not support netconsole (polling)

dererk, 2017-03-06:
Hi there.

Between some awesome features of linux, one of them is netconsole, which helps remote debugging and logging of critical systems:
Unfortunately the smsc95xx raspberry linux driver does not support polling, which is required for this feature to work, among other network features.

I consider this is of extreme importance due to the ability of a raspberry to be deployed remotely and, therefore, meeting one of the mentioned criteria:

Kernel version 
> root@rpi1:~# uname -a 
>Linux raspberrypi 4.9.13-v7+ #1 SMP Fri Mar 3 23:59:50 UTC 2017 armv7l GNU/Linux

Kernel msg
> [  360.725070] netpoll: netconsole: local port 6665
> [  360.727029] netpoll: netconsole: local IPv4 address 1.1.1.1
> [  360.729016] netpoll: netconsole: interface 'eth0'
> [  360.730985] netpoll: netconsole: remote port 6666
> [  360.732920] netpoll: netconsole: remote IPv4 address 1.1.1.2
> [  360.734875] netpoll: netconsole: remote ethernet address ff:ff:ff:ff:ff:ff
> [  360.737376] netpoll: netconsole: eth0 doesn't support polling, aborting

Any feedback on this would be awesome!

cheers,

P33M, 2017-03-06:
Does _any_ usb-ethernet device support polling?

https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/usb/usbnet.c#L1578

The struct net_device_ops in usbnet.c is missing the net_ll_poll member.

Electron752, 2017-03-06:
@dererk -

I suspect making this work would be rather difficult.  It would require making both the usb driver and the usb stack including bus enumeration work in polling mode.  

One option would be to add an expansion board that adds a very cheap 10M spi attached network controller that's attached through the GPIO ports.   Microchip actually makes a very old one(ENC28J60 - http://www.microchip.com/wwwproducts/en/en022889) that's available in DIP so it's hobbyist friendly:)

JamesH65, 2017-05-18:
Closing as not possible.

closed 2017-05-18 by JamesH65