raspberrypi/linux #1880: Quectel EC21 modem support

BaliSolutions, 2017-03-07:
Any chance support for Quectel EC21 model be added to the kernel?

**output for dmesg**:
```
[    7.575835] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[    7.715808] usb 1-1.4: New USB device found, idVendor=2c7c, idProduct=0121
[    7.715819] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    7.715824] usb 1-1.4: Product: Android
[    7.715827] usb 1-1.4: Manufacturer: Android
[    7.728580] usbcore: registered new interface driver cdc_wdm
[    7.735897] qmi_wwan 1-1.4:1.4: cdc-wdm0: USB WDM device
[    7.737829] qmi_wwan 1-1.4:1.4 wwan0: register 'qmi_wwan' at usb-3f980000.usb-1.4, WWAN/QMI device, ce:db:66:0e:df:26
```

**output for lsusb -t:**
```
root@raspberrypi:~# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 3: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 7, If 1, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 7, If 4, Class=Vendor Specific Class, Driver=qmi_wwan, 480M
        |__ Port 4: Dev 7, If 2, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 7, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 7, If 3, Class=Vendor Specific Class, Driver=, 480M
```

**output for lsmod:**
```
root@raspberrypi:~# lsmod
Module                  Size  Used by
qmi_wwan               13643  0 
cdc_wdm                12196  1 qmi_wwan
cfg80211              542899  0 
rfkill                 20851  1 cfg80211
sg                     20761  0 
mcp251x                10203  0 
can_dev                13682  1 mcp251x
ftdi_sio               32899  0 
usbserial              29943  1 ftdi_sio
i2c_bcm2835             7081  0 
spi_bcm2835             7596  0 
bcm2835_gpiomem         3940  0 
uio_pdrv_genirq         3923  0 
fixed                   3285  0 
uio                    10204  1 uio_pdrv_genirq
rtc_ds1307             13908  0 
hwmon                  10552  1 rtc_ds1307
i2c_dev                 6913  0 
i2c_bcm2708             5994  0 
fuse                   99603  1 
ipv6                  406215  38 
```

6by9, 2017-03-07:
The kernel has loaded a driver for it - cdc_wdm, with qmi_wwan on top of that.
What driver did you want to load instead?

BaliSolutions, 2017-03-07:
I need the qcserial driver to be loaded for it, like it is for the Quectel EC20
This is what i get with the Quectel EC20 modem:

```
root@raspberrypi:~# lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 1: Dev 5, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 2: Dev 4, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 3: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 4: Dev 7, If 1, Class=Vendor Specific Class, Driver=qcserial, 480M
        |__ Port 4: Dev 7, If 4, Class=Vendor Specific Class, Driver=qmi_wwan, 480M
        |__ Port 4: Dev 7, If 2, Class=Vendor Specific Class, Driver=qcserial, 480M
        |__ Port 4: Dev 7, If 0, Class=Vendor Specific Class, Driver=qcserial, 480M
        |__ Port 4: Dev 7, If 3, Class=Vendor Specific Class, Driver=qcserial, 480M
```
```
root@raspberrypi:~# lsmod
Module                  Size  Used by
cfg80211              542899  0
rfkill                 20851  1 cfg80211
qcserial                5583  0
qmi_wwan               13643  0
usb_wwan                9483  1 qcserial
cdc_wdm                12196  1 qmi_wwan
ftdi_sio               32899  0
usbserial              29943  3 qcserial,ftdi_sio,usb_wwan
sg                     20761  0
i2c_bcm2835             7081  0
bcm2835_gpiomem         3940  0
fixed                   3285  0
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
rtc_ds1307             13908  0
hwmon                  10552  1 rtc_ds1307
i2c_dev                 6913  0
i2c_bcm2708             5994  0
fuse                   99603  1
ipv6                  406215  32

```

```
[    5.290636] usbserial: USB Serial support registered for generic
[    5.305483] usbcore: registered new interface driver ftdi_sio
[    5.305688] usbserial: USB Serial support registered for FTDI USB Serial Device
[    5.305986] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
[    5.306187] usb 1-1.2: Detected FT-X
[    5.310574] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0
[    5.916931] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[    6.059687] usb 1-1.4: New USB device found, idVendor=05c6, idProduct=9215
[    6.059707] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.059715] usb 1-1.4: Product: Quectel LTE Module
[    6.059722] usb 1-1.4: Manufacturer: Quectel
[    6.110109] usbcore: registered new interface driver cdc_wdm
[    6.130769] qmi_wwan 1-1.4:1.4: cdc-wdm0: USB WDM device
[    6.131871] qmi_wwan 1-1.4:1.4 wwan0: register 'qmi_wwan' at usb-3f980000.usb-1.4, WWAN/QMI device, b2:4b:e4:19:1e:97
[    6.132041] usbcore: registered new interface driver qmi_wwan
[    6.144884] usbcore: registered new interface driver qcserial
[    6.145000] usbserial: USB Serial support registered for Qualcomm USB modem
[    6.151471] qcserial 1-1.4:1.0: Qualcomm USB modem converter detected
[    6.152080] usb 1-1.4: Qualcomm USB modem converter now attached to ttyUSB1
[    6.158647] qcserial 1-1.4:1.1: Qualcomm USB modem converter detected
[    6.159196] usb 1-1.4: Qualcomm USB modem converter now attached to ttyUSB2
[    6.160037] qcserial 1-1.4:1.2: Qualcomm USB modem converter detected
[    6.160480] usb 1-1.4: Qualcomm USB modem converter now attached to ttyUSB3
[    6.161144] qcserial 1-1.4:1.3: Qualcomm USB modem converter detected
[    6.161605] usb 1-1.4: Qualcomm USB modem converter now attached to ttyUSB4
```

6by9, 2017-03-07:
This is normally down to what services the USB device advertises, not something configured in the kernel.
What does ```lsusb -v``` give? How does it compare to when an EC20 is plugged in?
Given that the EC21 is reporting as an "Android:Android" device, it sounds like it needs some configuration or is incorrectly connected.

All the upstream modules appear to be configured in the 4.9 kernel (```uname -a```) which will be the standard very soon (```sudo rpi-update``` on a non-critical Pi before that).
There is a commit https://github.com/torvalds/linux/commit/9a765881bf3dcd32847d7108cf48cb04a4ed993f from October 2016 that claims to add some support for the EC21. As the 4.4 kernel is about to be deprecated don't expect those changes to be backported.

BaliSolutions, 2017-03-07:
lsusb -v on EC21:

```
root@raspberrypi:~# lsusb -v

Bus 001 Device 007: ID 2c7c:0121  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2c7c 
  idProduct          0x0121 
  bcdDevice            3.18
  iManufacturer           1 (error)
  iProduct                2 (error)
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          209
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
[... 391 lines cut ...]
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0000
  (Bus Powered)

```

6by9, 2017-03-07:
And your kernel version?

BaliSolutions, 2017-03-07:
```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
```

6by9, 2017-03-07:
In which case you should have all the config options enabled for it.

If there is still an issue, you need to either identify which additional module is required (unlikely any seeing as the EC20 works), or take it up with those upstream who support these devices - https://github.com/raspberrypi/linux/blob/rpi-4.9.y/REPORTING-BUGS

mypiandrew, 2017-03-09:
There's a patch here that appears to add the missing identifiers to option.c :

mypiandrew, 2017-03-09:
There's a patch here that appears to add the missing product identifiers to option.c 

https://lists.freedesktop.org/archives/libqmi-devel/2016-October/001781.html

Is this something that can be added?

6by9, 2017-03-09:
If it's not been merged upstream (and it appears not to have been), then not without **_very_** good reason.

BaliSolutions, 2017-03-29:
The changes were merged upstream, so 4.11.y has the modifications for Quectel EC21. Can those changes be added to 4.9.y?

pelwell, 2017-03-30:
That commit has been backported to rpi-4.9.y , and it will get merged onto 4.10 in due course.

BaliSolutions, 2017-04-03:
Any chance these will be available in 4.4.y?

JamesH65, 2017-05-18:
Unlikely, we have moved on to 4.9 now. There is a third party github tree that is claiming to keep support going for 4.4.

Closing this issue as question answered/issue resolved.

closed 2017-05-18 by JamesH65