raspberrypi/linux #1882: Kernel crashing with motion and USB Webcam

imp1sh, 2017-03-07:
Very similar to https://github.com/raspberrypi/linux/issues/1873 I have sporadic crashes. They lead not only to a non-functioning software (i.e. motion) but also make raspberry non-reachable in network.
Display: none attached
USB Webcam: QuickCam Vision Pro for Mac
In Contrast to #1873 I'm using motion not hawkeye and YUYV but not MJPEG.
v4l2_palette 15
width 800
height 600
framerate 2

This is happening on a Raspberry 2 
root@pi1:/etc/# /opt/vc/bin/vcgencmd version
Jan 31 2017 17:40:14
Copyright (c) 2012 Broadcom
version 8d94fbc3bbdde25c29b18a2572a5c60ee75c56f3 (clean) (release)

Full kernel log:
...
Mar  6 03:53:08 pi1 kernel: [221393.476689] ------------[ cut here ]------------
Mar  6 03:53:08 pi1 kernel: [221393.476762] WARNING: CPU: 0 PID: 7236 at fs/sysfs/group.c:237 sysfs_remove_group+0x98/0xa4()
Mar  6 03:53:08 pi1 kernel: [221393.476779] sysfs group c086a54c not found for kobject 'event0'
Mar  6 03:53:08 pi1 kernel: [221393.476792] Modules linked in: cfg80211 rfkill evdev uvcvideo videobuf2_vmalloc videobuf2_memops snd_usb_audio videobuf2_v4l2 videobuf2_core snd_hwdep v4l2_common snd_usbmidi_lib snd_rawmidi snd_seq_device videodev media snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio i2c_dev fuse ipv6
Mar  6 03:53:08 pi1 kernel: [221393.476916] CPU: 0 PID: 7236 Comm: motion Not tainted 4.4.47+ #961
Mar  6 03:53:08 pi1 kernel: [221393.476929] Hardware name: BCM2708
Mar  6 03:53:08 pi1 kernel: [221393.476994] [<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
Mar  6 03:53:08 pi1 kernel: [221393.477031] [<c0013c20>] (show_stack) from [<c02e48fc>] (dump_stack+0x20/0x28)
Mar  6 03:53:08 pi1 kernel: [221393.477067] [<c02e48fc>] (dump_stack) from [<c0021f0c>] (warn_slowpath_common+0x8c/0xc4)
Mar  6 03:53:08 pi1 kernel: [221393.477096] [<c0021f0c>] (warn_slowpath_common) from [<c0021f84>] (warn_slowpath_fmt+0x40/0x48)
Mar  6 03:53:08 pi1 kernel: [221393.477123] [<c0021f84>] (warn_slowpath_fmt) from [<c019bfd8>] (sysfs_remove_group+0x98/0xa4)
Mar  6 03:53:08 pi1 kernel: [221393.477167] [<c019bfd8>] (sysfs_remove_group) from [<c03723d8>] (dpm_sysfs_remove+0x58/0x60)
Mar  6 03:53:08 pi1 kernel: [221393.477207] [<c03723d8>] (dpm_sysfs_remove) from [<c0367b6c>] (device_del+0x48/0x224)
Mar  6 03:53:08 pi1 kernel: [221393.477260] [<c0367b6c>] (device_del) from [<bf1e31ac>] (evdev_disconnect+0x28/0x54 [evdev])
Mar  6 03:53:08 pi1 kernel: [221393.477338] [<bf1e31ac>] (evdev_disconnect [evdev]) from [<c0413a0c>] (__input_unregister_device+0xa8/0x14c)
Mar  6 03:53:08 pi1 kernel: [221393.477373] [<c0413a0c>] (__input_unregister_device) from [<c0414334>] (input_unregister_device+0x50/0x84)
Mar  6 03:53:08 pi1 kernel: [221393.477500] [<c0414334>] (input_unregister_device) from [<bf1d2ab0>] (uvc_status_cleanup+0x40/0x44 [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.477624] [<bf1d2ab0>] (uvc_status_cleanup [uvcvideo]) from [<bf1c831c>] (uvc_delete+0x1c/0x12c [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.477731] [<bf1c831c>] (uvc_delete [uvcvideo]) from [<bf1c8554>] (uvc_release+0x44/0x48 [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.478170] [<bf1c8554>] (uvc_release [uvcvideo]) from [<bf115760>] (v4l2_device_release+0xc0/0xe8 [videodev])
Mar  6 03:53:08 pi1 kernel: [221393.478432] [<bf115760>] (v4l2_device_release [videodev]) from [<c0367264>] (device_release+0x3c/0xa0)
Mar  6 03:53:08 pi1 kernel: [221393.478472] [<c0367264>] (device_release) from [<c02e6b54>] (kobject_release+0x50/0x84)
Mar  6 03:53:08 pi1 kernel: [221393.478500] [<c02e6b54>] (kobject_release) from [<c02e6bd4>] (kobject_put+0x4c/0x7c)
Mar  6 03:53:08 pi1 kernel: [221393.478527] [<c02e6bd4>] (kobject_put) from [<c036760c>] (put_device+0x24/0x28)
Mar  6 03:53:08 pi1 kernel: [221393.478772] [<c036760c>] (put_device) from [<bf115480>] (v4l2_release+0x60/0x88 [videodev])
Mar  6 03:53:08 pi1 kernel: [221393.479039] [<bf115480>] (v4l2_release [videodev]) from [<c012e724>] (__fput+0x98/0x1dc)
Mar  6 03:53:08 pi1 kernel: [221393.479075] [<c012e724>] (__fput) from [<c012e8d0>] (____fput+0x18/0x1c)
Mar  6 03:53:08 pi1 kernel: [221393.479112] [<c012e8d0>] (____fput) from [<c003b4c0>] (task_work_run+0x78/0xa8)
Mar  6 03:53:08 pi1 kernel: [221393.479155] [<c003b4c0>] (task_work_run) from [<c0013524>] (do_work_pending+0xe0/0xe4)
Mar  6 03:53:08 pi1 kernel: [221393.479186] [<c0013524>] (do_work_pending) from [<c000f848>] (slow_work_pending+0xc/0x20)
Mar  6 03:53:08 pi1 kernel: [221393.479203] ---[ end trace 46797110a8c742ff ]---
Mar  6 03:53:08 pi1 motion: [1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Mar  6 03:53:08 pi1 motion: [1] [NTC] [VID] vid_close: Closing video device /dev/video0
Mar  6 03:53:08 pi1 kernel: [221393.493098] Indeed it is in host mode hprt0 = 00021501
Mar  6 03:53:08 pi1 kernel: [221393.564769] ------------[ cut here ]------------
Mar  6 03:53:08 pi1 kernel: [221393.564857] WARNING: CPU: 0 PID: 7236 at fs/sysfs/group.c:237 sysfs_remove_group+0x98/0xa4()
Mar  6 03:53:08 pi1 kernel: [221393.564873] sysfs group c086a54c not found for kobject 'input0'
Mar  6 03:53:08 pi1 kernel: [221393.564902] Modules linked in: cfg80211 rfkill evdev uvcvideo videobuf2_vmalloc videobuf2_memops snd_usb_audio videobuf2_v4l2 videobuf2_core snd_hwdep v4l2_common snd_usbmidi_lib snd_rawmidi snd_seq_device videodev media snd_bcm2835 snd_pcm snd_timer snd bcm2835_gpiomem bcm2835_wdt uio_pdrv_genirq uio i2c_dev fuse ipv6
Mar  6 03:53:08 pi1 kernel: [221393.565049] CPU: 0 PID: 7236 Comm: motion Tainted: G        W       4.4.47+ #961
Mar  6 03:53:08 pi1 kernel: [221393.565063] Hardware name: BCM2708
Mar  6 03:53:08 pi1 kernel: [221393.565127] [<c0016cfc>] (unwind_backtrace) from [<c0013c20>] (show_stack+0x20/0x24)
Mar  6 03:53:08 pi1 kernel: [221393.565185] [<c0013c20>] (show_stack) from [<c02e48fc>] (dump_stack+0x20/0x28)
Mar  6 03:53:08 pi1 kernel: [221393.565225] [<c02e48fc>] (dump_stack) from [<c0021f0c>] (warn_slowpath_common+0x8c/0xc4)
Mar  6 03:53:08 pi1 kernel: [221393.565255] [<c0021f0c>] (warn_slowpath_common) from [<c0021f84>] (warn_slowpath_fmt+0x40/0x48)
Mar  6 03:53:08 pi1 kernel: [221393.565306] [<c0021f84>] (warn_slowpath_fmt) from [<c019bfd8>] (sysfs_remove_group+0x98/0xa4)
Mar  6 03:53:08 pi1 kernel: [221393.565354] [<c019bfd8>] (sysfs_remove_group) from [<c03723d8>] (dpm_sysfs_remove+0x58/0x60)
Mar  6 03:53:08 pi1 kernel: [221393.565396] [<c03723d8>] (dpm_sysfs_remove) from [<c0367b6c>] (device_del+0x48/0x224)
Mar  6 03:53:08 pi1 kernel: [221393.565466] [<c0367b6c>] (device_del) from [<c0413a88>] (__input_unregister_device+0x124/0x14c)
Mar  6 03:53:08 pi1 kernel: [221393.565504] [<c0413a88>] (__input_unregister_device) from [<c0414334>] (input_unregister_device+0x50/0x84)
Mar  6 03:53:08 pi1 kernel: [221393.565661] [<c0414334>] (input_unregister_device) from [<bf1d2ab0>] (uvc_status_cleanup+0x40/0x44 [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.565807] [<bf1d2ab0>] (uvc_status_cleanup [uvcvideo]) from [<bf1c831c>] (uvc_delete+0x1c/0x12c [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.565943] [<bf1c831c>] (uvc_delete [uvcvideo]) from [<bf1c8554>] (uvc_release+0x44/0x48 [uvcvideo])
Mar  6 03:53:08 pi1 kernel: [221393.566411] [<bf1c8554>] (uvc_release [uvcvideo]) from [<bf115760>] (v4l2_device_release+0xc0/0xe8 [videodev])
[... 101 lines cut ...]
Mar  6 03:53:08 pi1 kernel: [221393.592340] [<c0013524>] (do_work_pending) from [<c000f848>] (slow_work_pending+0xc/0x20)
Mar  6 03:53:08 pi1 kernel: [221393.592359] ---[ end trace 46797110a8c74303 ]---
Mar  6 03:53:08 pi1 kernel: [221393.932753] usb 1-1: new high-speed USB device number 5 using dwc_otg
Mar  6 03:53:08 pi1 kernel: [221393.933006] Indeed it is in host mode hprt0 = 00001101
Mar  6 03:53:08 pi1 kernel: [221394.274374] usb 1-1: New USB device found, idVendor=0424, idProduct=9512
Mar  6 03:53:08 pi1 kernel: [221394.274413] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Mar  6 03:53:08 pi1 kernel: [221394.288551] hub 1-1:1.0: USB hub found
Mar  6 03:53:08 pi1 kernel: [221394.288761] hub 1-1:1.0: 3 ports detected
Mar  6 03:53:09 pi1 kernel: [221394.572797] usb 1-1.1: new high-speed USB device number 6 using dwc_otg
Mar  6 03:53:09 pi1 kernel: [221394.673338] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
Mar  6 03:53:09 pi1 kernel: [221394.673376] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Mar  6 03:53:09 pi1 kernel: [221394.689861] smsc95xx v1.0.4
[... cut ...]
Mar  6 03:53:10 pi1 motion: [1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Mar  6 03:53:10 pi1 motion: [1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
Mar  6 03:53:10 pi1 motion: [1] [ALR] [VID] vid_v4lx_start: Failed to open video device /dev/video0:
Mar  6 03:53:10 pi1 kernel: [221396.027542] usb 1-1.2: Warning! Unlikely big volume range (=5120), cval->res is probably wrong.
Mar  6 03:53:10 pi1 kernel: [221396.027587] usb 1-1.2: [5] FU [Mic Capture Volume] ch = 1, val = 2816/7936/1
...

6by9, 2017-03-08:
Almost certainly no link to #1873 - one is using USB and the other the CSI2 interface via the GPU.

Those kernel logs don't have anything interesting. The 
```sysfs group c086a54c not found for kobject 'blah'```
messages are a known upstream bug if you close a USB webcam whilst something has a handle open on it.
https://www.spinics.net/lists/linux-media/msg108882.html

Something else that you haven't quoted has caused the USB stack to trip up (network is over USB as well), which results in these warnings.

imp1sh, 2017-03-08:
Here's what happened before:
...
Mar  6 03:40:00 pi1 motion: [1] [NTC] [EVT] event_newfile: File of type 2 saved to: /var/lib/motion/snapshot/20170306-034000-snapshot.jpg
Mar  6 03:40:00 pi1 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Mar  6 03:41:30 2017 [try http://www.rsyslog.com/e/2007 ]
Mar  6 03:46:42 pi1 kernel: [221008.300389] uvcvideo: Non-zero status (-5) in status completion handler.
Mar  6 03:46:42 pi1 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Mar  6 03:48:12 2017 [try http://www.rsyslog.com/e/2007 ]
Mar  6 03:50:17 pi1 motion: [0] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
Mar  6 03:50:30 pi1 motion: [1] [NTC] [ALL] motion_loop: Thread exiting
Mar  6 03:50:30 pi1 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Mar  6 03:52:00 2017 [try http://www.rsyslog.com/e/2007 ]
Mar  6 03:50:30 pi1 motion: [1] [NTC] [STR] stream_stop: Closing motion-stream listen socket & active motion-stream sockets
Mar  6 03:50:30 pi1 motion: [1] [NTC] [STR] stream_stop: Closed motion-stream listen socket & active motion-stream sockets
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] vid_close: Closing video device /dev/video0
Mar  6 03:50:30 pi1 motion: [0] [NTC] [ALL] main: Motion thread 1 restart
Mar  6 03:50:30 pi1 motion: [1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] vid_v4lx_start: Using videodevice /dev/video0 and input -1
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_get_capability: #012------------------------#012cap.driver: "uvcvideo"#012cap.card: "UVC Camera (046d:09a6)"#012cap.bus_info: "usb-20980000.usb-1.2"#012cap.capabilities=0x84200001#012------------------------
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_get_capability: - STREAMING
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_select_input: name = "Camera 1", type 0x00000002, status 00000000
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_select_input: - CAMERA
Mar  6 03:50:30 pi1 motion: [1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YUYV (800x600)
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_do_set_pix_format: Using palette YUYV (800x600) bytesperlines 1600 sizeimage 960000 colorspace 00000008
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980900, "Brightness", range 0,255
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"Brightness", default -8193, current 128
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980901, "Contrast", range 0,255
Mar  6 03:50:30 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"Contrast", default 57343, current 32
Mar  6 03:50:31 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980902, "Saturation", range 0,255
Mar  6 03:50:31 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"Saturation", default 57343, current 32
Mar  6 03:50:31 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: found control 0x00980913, "Gain", range 0,255
Mar  6 03:50:31 pi1 motion: [1] [NTC] [VID] v4l2_scan_controls: #011"Gain", default 57343, current 32
Mar  6 03:50:31 pi1 motion: [1] [NTC] [VID] vid_v4lx_start: Using V4L2
Mar  6 03:50:31 pi1 motion: [1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
Mar  6 03:50:31 pi1 motion: [1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 127.0.0.1 port: 8081
Mar  6 03:50:31 pi1 motion: [1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 127.0.0.1 port: 8081
Mar  6 03:50:31 pi1 motion: [1] [NTC] [ALL] motion_init: Started motion-stream server in port 8081 auth Enabled
Mar  6 03:53:07 pi1 dhcpcd[372]: eth0: carrier lost
Mar  6 03:53:07 pi1 rsyslogd-2007: action 'action 17' suspended, next retry is Mon Mar  6 03:54:37 2017 [try http://www.rsyslog.com/e/2007 ]
Mar  6 03:53:07 pi1 kernel: [221393.292917] usb 1-1: USB disconnect, device number 2
Mar  6 03:53:07 pi1 kernel: [221393.292956] usb 1-1.1: USB disconnect, device number 3
Mar  6 03:53:07 pi1 kernel: [221393.293778] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-20980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Mar  6 03:53:07 pi1 kernel: [221393.293875] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Mar  6 03:53:07 pi1 kernel: [221393.313505] usb 1-1.2: USB disconnect, device number 4
Mar  6 03:53:08 pi1 motion: [1] [ERR] [VID] v4l2_next: VIDIOC_QBUF:
...

P33M, 2017-03-08:
`Mar  6 03:53:08 pi1 kernel: [221394.274374] usb 1-1: New USB device found, idVendor=0424, idProduct=9512`

This device is not used on Pi 2. The lan9512 is exclusive to the Model B (v1.0 and v2.0). Are you using a Pi with only 2 USB ports?

imp1sh, 2017-03-08:
Sorry, mixed them up. This one is correct:
Raspberry Pi Model B (Rev 2.0, 512Mb): 000f

P33M, 2017-03-09:
Given that the entire USB device chain is getting disconnected, I suspect power brown-out as a primary cause. Can you replicate this behaviour with the camera plugged into a self-powered hub?

P33M, 2017-04-27:
No response in >1 month. Will reopen if OP responds.

closed 2017-04-27 by P33M

rooterkyberian, 2018-12-30:
> Given that the entire USB device chain is getting disconnected, I suspect power brown-out as a primary cause. Can you replicate this behaviour with the camera plugged into a self-powered hub?

Had the same issue with Logitech C920, powered USB hub fixed it for me.