raspberrypi/linux #1886: Zero W: bcm2708-rpi-0-w.dts has no property/override for pwr led

campag, 2017-03-08:
I'm needing a pulsing watchdog input to my circuit, & as early as possible in the boot process. The kernel led class pwr_led, with heartbeat trigger, was doing great on the previous Zeros, and on other Pi models.

On those models, I simply create & configure a pwr_led on my chosen GPIO by dtparams in config.txt. On the Zero W however, these are ignored, as evidenced in sudo vcdbg log msg.

Could the property & override be added?

Cheers.

pelwell, 2017-03-08:
Pi Zero W, like Pi Zero, has only one LED which does double duty as both a power and activity LED. Try setting act_led_trigger instead.

campag, 2017-03-08:
Hi & thanks. Yes, I can see that would work, but I'd lose the act led. My current workaround is a custom overlay to create another led. I'm curious why one can do this on the previous Zeros, but not on the latest W.

Likewise, I would hope that in spite of there being no analogue audio pins, I would be able to configure that on the 40-pin header.

pelwell, 2017-03-08:
Yes, I was going to suggest an overlay. Ideally it would be useful to other Pi's as well, but that may need some experimentation. Actually, there is already an overlay that does the job - the [wittypi overlay](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/wittypi-overlay.dts) - but since it also initialises an I2C-based clock you might want to just use it for parts.

There is already an overlay to remap the audio pins - `audremap`:

```
Info:   Switches PWM sound output to pins 12 (Right) & 13 (Left)
Load:   dtoverlay=audremap,<param>=<val>
Params: swap_lr                 Reverse the channel allocation, which will also
                                swap the audio jack outputs (default off)
        enable_jack             Don't switch off the audio jack output
                                (default off)
```
You can't change which pins to use because 12 and 13 are the only header pins that can carry the required PWM functions.

By the way, the reason the power LED is mentioned in the Pi Zero dtb is a consequence of the fact that Pi Zero shares the B+ dtb, leaving the power LED set to 35 which is disconnected.

campag, 2017-03-08:
Yes, had spotted the wittypi overlay, which is very much appropriate, as my board has similar functionality to that ie RTC with wake-up.

For  completeness, I would add my in/out pin to it, which is a simulated open-drain i/opin by way of setting it output-low then input-pullup. But the wittypi overlay will suffice for now. Thanks for helping me gain a better understanding.

closed 2017-03-08 by campag