raspberrypi/linux #1889: Raspberry Pi GPIO17 with DHT22 no longer working

kowenhouston, 2017-03-09:
Hi There, 

I have a few raspberry pi's wired with DHT22 sensors. It would be very expensive for me to wire them all back up. 

This is a working raspberry pi with raspbian:
- 4.1.20-v7+ #867 SMP Wed Mar 23 20:12:32 GMT 2016 armv7l GNU/Linux
- Debian 7.11
- Python 2.7.3
```
root@localhost:~# python AdafruitDHT.py 22 17
Temp=24.2*  Humidity=29.9%
```

The non working raspberry pi with rasbian is:
- 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux
- Debian 7.11
- Python 2.7.3
```
root@localhost:~# python AdafruitDHT.py 22 17
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 90, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 77, in read
    return platform.read(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/Beaglebone_Black.py", line 206, in read
    raise ValueError('Pin must be a valid GPIO identifier like P9_12 or GPIO1_28.')
ValueError: Pin must be a valid GPIO identifier like P9_12 or GPIO1_28.
```

The only difference between the two Raspberry Pis is that I ran **rpi-update** on one, which is now not working. The are both wired up identically, have the identical image (however I updated one).

My Wiring looks like this:
GPIO17 / Raspberry Pi Pin 11 - DHT22 Pin 2
+5VDC / Raspberry Pi Pin 2 - DHT22 Pin 1
GND / Raspberry Pi Pin 6 - DHT22 Pin 4

pelwell, 2017-03-09:
The filename Beaglebone_Black.py suggests the library has failed to identify that this is running on a Pi. There is a commit on Jan 20th to AdaFruit's Github repo for AdafruitDHT that looks like it might help, so can you update?

On [this page](https://github.com/Hexxeh/rpi-firmware/commits/master) you'll find all of the different firmware+kernel releases, along with a long SHA hash identifying the commit. You can get rpi-update to install a specific version by giving the hash. I suggest you start with the last release built from the 4.4 kernel, labelled "kernel: Bump to 4.4.50":
```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
```
if you are keen you can then jump backwards or forwards from that point to find the first failing commit, but just knowing the result of testing with that one release would help.

kowenhouston, 2017-03-09:
I've downgraded to the following, which as far as I can see is the last one I can downgrade to before breaking the raspberry pi 2 v1.2 compatibility 

> Commits on Sep 22, 2016
>  @popcornmix
> platform: Enable support for Pi2+
> 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82

```
rpi-update 21505f3fb5d8e2a5abbf95baf441a7a6415d4a82
```

Which still gives me an error, but it's different again.
```
python AdafruitDHT.py 22 17
Traceback (most recent call last):
  File "AdafruitDHT.py", line 41, in <module>
    humidity, temperature = Adafruit_DHT.read_retry(sensor, pin)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 90, in read_retry
    humidity, temperature = read(sensor, pin, platform)
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 76, in read
    platform = get_platform()
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/common.py", line 51, in get_platform
    import Raspberry_Pi_2
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 22, in <module>
    import Raspberry_Pi_2_Driver as driver
ImportError: No module named Raspberry_Pi_2_Driver
```

Is there any way to see how the pin mapping differs without getting to deep into the code? Or even better is there a way to enable Pi2 v1.2 compatibility but still regaining the ability to run this library?

pelwell, 2017-03-09:
It sounds like you haven't installed the library properly and/or need to update it for the 4.9 kernel. I'm sure the good people at Adafruit (or their forums) will be able to help.

kowenhouston, 2017-03-09:
Ok, Thank you for your time. I have opened an issue over on the library page too:
https://github.com/adafruit/Adafruit_Python_DHT/issues/63

closed 2017-03-09 by kowenhouston