raspberrypi/linux #1891: Rpi3 controlling red led

mkreisl, 2017-03-10:
Since commit https://github.com/raspberrypi/linux/commit/5d1da107bc619c1cc091ee5301f85cd89a6d92fb, /sys/class/leds/led1 is back to control red led for rpi3.

But when I do echo 0 > /sys/class/leds/led1/brightness, I get that undervoltage warning on screen: https://s9.postimg.org/b062013wb/under_volt.png

So my question, does this commit makes sense?

Using kernel 4.9.13

6by9, 2017-03-10:
Yes. 
https://github.com/raspberrypi/linux/issues/1332#issuecomment-280087412

mkreisl, 2017-03-10:
Hmm, I unterstand controlling of this red led is not trivial, but I would suppose that the driver would take care of that undervolting logic completely. 

And currently this is not the case, so IMO for users still not usable and very dangerous @pelwell already noted

6by9, 2017-03-11:
The GPU is monitoring the configured GPIO for undervoltage detection. The ARM is now configuring it as an output and changing the state. Two independent processors that know nothing of the other's use of the GPIOs.

Further checking of the firmware code it does appear to check that an internal GPIO is set as an input before reporting low voltage. That hasn't been done for GPIOs on the expander, so that should avoid the display (no one had noted that omission during review). I'll see what can be done.

mkreisl, 2017-03-12:
@6by9 I'm sure you'll find a suitable solution

6by9, 2017-03-14:
Updated firmware released last night that should apply a similar check of the pin being an input before using it for under voltage detection.
Seems to work for me, please verify and close if you're happy.

mkreisl, 2017-03-14:
Yeah, works great :smile: LED1 can be turned off without getting powerfail indication now.  heartbeat mode or maybe other trigger modes still crashes pi, but do not use thoose modes.
Thanks for your quick response :+1:

pelwell, 2017-03-30:
I found a bug in the leds_gpio module (one I put there long ago) that was causing timer- and interrupt-driven LED triggers to lock up. The kernel sources and the latest rpi-update firmware now include the fix for this, so feel free to put a heartbeat on your PWR LED.

6by9, 2017-04-18:
Broken in that it brings up the undervoltage warning (ie this issue), or broken in that it doesn't work at all (not this issue)?
The undervoltage warning is a function of the firmware, so what firmware version are you running?

mkreisl, 2017-04-18:
@6by9 Undervoltage warning, nothing else tested

> The undervoltage warning is a function of the firmware, so what firmware version are you running?

Using same firmware for 4.9 and 4.10 kernel, so it is kernel. Firmware files are from Apr 6

pelwell, 2017-04-18:
My Pi3 is on 4.10.10-v7+ and the PWR LED is happily flashing a heartbeat pattern.

mkreisl, 2017-04-18:
Hmmm, that's weird. Build kernel 4.10.10 today, and now it's working again. No idea why 4.10.8 from Apr 3 had this issue again :confused: 

Sorry for the noise

closed 2017-04-18 by mkreisl