raspberrypi/linux #1892: Solid red when using an initramfs bundled in kernel image

agherzan, 2017-03-10:
Kernel version and revision: 4.4.48 / 7ddf96fbb7d637b79b449c7bd1c8d35f00571e4b

Using `dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait`
- kernel image = boots OK
- kernel image bundled with initramfs using CONFIG_INITRAMFS_SOURCE + CONFIG_BLK_DEV_INITRD = some green leds and then after a couple of seconds solid red 

Using `dwc_otg.lpm_enable=0 console=console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait`
- kernel image bundled with initramfs using CONFIG_INITRAMFS_SOURCE + CONFIG_BLK_DEV_INITRD = some green leds and then after a couple of seconds solid red 

Using `dwc_otg.lpm_enable=0 console=console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait` and adding `enable_uart=1` to config.txt
- kernel image bundled with initramfs using CONFIG_INITRAMFS_SOURCE + CONFIG_BLK_DEV_INITRD = some green leds and then after a couple of seconds solid red 
- full comdline `Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa02082 bcm2709.serial=0x375ae59 smsc95xx.ma
caddr=B8:27:EB:75:AE:59 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3ea00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS
0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait` - similar to all the cases but the changes mentioned

Basically it seems to be a dependency in between enble_uart and bundled initramfs kernel images. Are you guys aware of this? Do we have a workaround for it?

pelwell, 2017-03-10:
Perhaps I'm being slow (it's been a long week), but from what I can see enable_uart doesn't make any difference to the outcome of your tests. Cut-and-paste error?

agherzan, 2017-03-10:
@pelwell fixed my report - sorry for this.

pelwell, 2017-03-10:
Have you tried without console=serial0,115200 ?

pelwell, 2017-03-10:
I'm assuming this is a Pi3 or Pi Zero W? You don't say.

pelwell, 2017-03-10:
Apart from enabling the UART, enable_uart=1 on a Pi 3 (and Zero W) fixes the core clock at 250MHz. Without adding enable_uart=1, try setting core_freq=250 to see if that makes a difference.

agherzan, 2017-03-10:
I am currently testing on RPI3, v1.2. I'll try to see how core_freq is influencing.

agherzan, 2017-03-10:
@pelwell added test results in the main description when using enable_uart alone. The led behaviour is different but still seems to be broken.

pelwell, 2017-03-10:
Hang on, the red light should not blink at all. That sounds like you have an undervoltage situation, and enabling the UART is slowing down the boot process so as to spread everything out and avoid a combined peak demand on the supply. With enable_uart=1, once it has booted run `vcgencmd get_throttled`. Any value other than zero is bad news, indicating throttling due to under-voltage or over-temperature.

agherzan, 2017-03-10:
I tried core_freq and it doesn't make any difference. Still the only thing that unblocks booting is enable_uart=1 + console=serial0,115200? Any other combination that I tried is trouble. That was the only hint I was thinking about. Let me see. Will come back. Thanks man!

agherzan, 2017-03-10:
root@6b5cf44:/opt/vc/bin# LD_LIBRARY_PATH=/opt/vc/lib/ ./vcgencmd get_throttled                                                                                                       
throttled=0x0

And still, without console and enable_uart the red led is solid on. I have no idea what to do next.

agherzan, 2017-03-10:
I tried: latest firmware - no go (solid red)
I tried: latest 4.9 + initramfs - no go (solid red + green blinks once per second)
I tried: latest 4.9 - no go (solid red + green blinks once per second)

pelwell, 2017-03-10:
Try adding `dtoverlay=pi3-miniuart-bt` - it will swap the two uarts and eliminate a few possibilities. You will need to use enable_uart=0 to disable uart0.

agherzan, 2017-03-10:
I added dtoverlay=pi3-miniuart-bt and disabled enable_uart (=0). Ending in the same solid red as the other tries. [here](https://www.youtube.com/watch?v=sIKvn-_himg) you can see the blinking pattern when I say "solid red". From sec 12 it remains solid red.

agherzan, 2017-03-10:
New test done: I used an image with 4.4 and bundled initramfs and only by changing console=tty (no uart, no uarts switch or anything), the board boots. So at this point it seems like console=null breaks kernel only if initramfs is bundled.

pelwell, 2017-03-10:
Does deleting console= change anything?

agherzan, 2017-03-10:
Removing it it goes directly to HDMI and everything seems OK.

pelwell, 2017-03-10:
Is that helpful? It might help if you explain a bit about your application.

agherzan, 2017-03-10:
OK. So what I get until no is that when console=null my initramfs scripts break the system. Here are the scripts I am running:

init script: https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/initrdscripts/initramfs-framework/init
This is a wrapper which runs enable and run for every of the following:
https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/initrdscripts/initramfs-framework/udev
https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/initrdscripts/initramfs-framework/rootfs
https://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-core/initrdscripts/initramfs-framework/finish

My current theory is that when passing console=null the switch_root with -c /dev/console breaks (or any other logging on that device).

JamesH65, 2017-05-18:
ping @pelwell

agherzan, 2017-05-18:
I think this can be closed. For reference the issue was in busybox and fixed with https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-core/busybox/busybox/0001-switch_root-don-t-bail-out-when-console-doesn-t-exis.patch . A similar version of this patch was already merged in busybox.

closed 2017-05-18 by pelwell