raspberrypi/linux #1893: ttyAMA0 (uart-pl011) gives "Port Busy" after first open, except if root

gohai, 2017-03-12:
I am attempting to use the UART on a Raspberry Pi 3 from user space (w/ [Processing](http://processing.org/)). I have the serial terminal disabled in the configuration utility, and enabled the port by running `sudo raspi-config nonint do_serial 1` and `sudo raspi-config nonint set_config_var enable_uart 1 /boot/config.txt`, and restarting. So far so good.

In the Processing sketch, I can open the Serial port the _first_ time after a reboot. But all subsequent invocations of the sketch will give back the error `Port busy`.

Oddly, this is not an issue when I run the sketch with root privileges. As root the port always opens. Unfortunately this is not a suitable workaround for Processing as it makes extensive use of the current user's home directory (e.g. for finding installed libraries).

I haven't found any information about this issue online, and would appreciate some information about what's going on here, and how we can circumvent this issue. Thanks in advance. (CC @spl237 @mrichardson23)

pelwell, 2017-03-12:
On a Pi3 ttyAMA0 is used to drive the Bluetooth modem, leaving ttyS0 for the console. If you still want to use BT you can add `dtoverlay=pi3-miniuart-bt` to swap them back, at the cost of reduced BT performance. `dtoverlay=pi3-disable-bt` turns the Pi3 into a Pi2 with regards to uarts - console on ttyAMA0, ttyS0 and BT disabled. Alternatively you can run `sudo systemctl disable hciuart` to prevent Bluetooth from starting, leaving it free to use for something else.

gohai, 2017-03-13:
Thanks for your reply @pelwell. I've tried to summarize the situation for our users here: https://github.com/processing/processing/wiki/Raspberry-Pi#serial

Is the serial port on models besides the Pi 3 named now also named `ttyAMA0`, or still `ttyACM0`?

pelwell, 2017-03-13:
Everything you've written is correct, but it isn't the whole story. The firmware and a udev script together create aliases to abstract away the serial port differences between the Pi models. /dev/serial0 is always the main console UART (ttyAMA0 on most models, ttyS0 on Pi 3 B and Pi Zero W), while /dev/serial1 is used for Bluetooth. Since enable_uart=1 is harmless on all Pis (assuming you want to use /dev/serial0) and is set by raspi-config,  you could simplify your instructions to:

> Open the Raspberry Pi Configuration (under Menu, Preferences), navigate to the Interfaces tab, make sure the serial shell hiding under the Serial option is disabled. After a restart, the serial port will be available to Processing's Serial library under the name /dev/serial0

I've not come across ttyACM0 before, but it looks like a USB-attached serial device - all the links I've seen have been Arduino-related.

gohai, 2017-03-13:
@pelwell `/dev/serial0` and `/dev/serial1` are good ideas in theory, but no program or library will pick this up as a serial port - everything, including Processing's Serial library will exclusively look for `/dev/tty...`. I hope you understand that it's not feasible for upstream projects to support these types of ad-hoc vendor customizations.

Regarding your simplified instructions (thanks for taking the time): At least on the Pi 3, disabling the Serial option in the GUI Configuration tool wasn't enough - I had to also force `enable_uart`. That was on the January release - did this get changed in more recent releases?

Thanks for the hint that the Pi Zero W works the same way as the 3!

pelwell, 2017-03-13:
After the initial question:
```
Would you like a login shell to be accessible over
serial?
```
raspi-config has a supplementary question:
```
Would you like the serial port hardware to be enabled?
```
If you answer "No" to the first question and "Yes" to the second the serial port should be configured as you want it to be. That feature was added to the RPI-Distro repo in January.

gohai, 2017-03-13:
I see. We're directing the users to the X11 GUI-type configuration tool currently. Will this receive the same type of prompt? Or should we just make everyone use `raspi-config` for all options?

pelwell, 2017-03-13:
Ah, I'm a console kind of guy. The GUI doesn't seem to have this feature so it looks like you'll have to run your shell command or stick to the textual raspi-config.

gohai, 2017-03-13:
Thanks for your help @pelwell. Closing this bug.

closed 2017-03-13 by gohai

gohai, 2018-03-16:
@pelwell Does enabling the Serial port still come with a performance penalty on the Raspberry Pi 3 with current releases of Raspbian? (What about 3+?) Thanks in advance!

pelwell, 2018-03-16:
Using UART1/mini-UART/ttyS0 requires that the core frequency doesn't change, otherwise a fluctuating baud rate will make it unusable. UART0/ttyAMA0 does not have this limitation.

Pi's with Bluetooth (3B, 3B+, 0W) use UART0 to drive the modem, so UART1 has to be used for the console. For this reason, setting `enable_uart=1` on a such a Pi locks the core to 250MHz (unless force_turbo=1, in which case it will be locked to the turbo speed - although undervoltage or thermal limiting would still cause a frequency change).