raspberrypi/linux #1894: Install old version of Linux for Raspberry Pi (4.1.18)

aurel94, 2017-03-13:
Hello everybody,

For a special project I need to have the 4.1.18 Linux version of the Raspberry Pi. When I use the following command : 
git  checkout  rpi-4.1.y 

I have the version 4.1.21. Do you know the methods to have the version 4.1.18.

Thanks,

pelwell, 2017-03-13:
It sounds like are are already on rpi-4.1.y, which is necessary otherwise you won't see all of the 4.1 point releases.
You are looking for a commit labelled "Linux 4.1.18":
```
$ git log --oneline | grep "Linux 4.1.18"
83fdace Linux 4.1.18
```
This is the first 4.1.18 commit, but you're probably better off with the last one because it will include any downstream patches we had to apply:
```
$ git log --oneline | grep -1 "Linux 4.1.19"
7e4f4cb Merge remote-tracking branch 'stable/linux-4.1.y' into rpi-4.1.y
b9a9cfd Linux 4.1.19
e9e9ccf Merge pull request #1304 from Cheong2K/rpi-4.1.y
```
Because logs are presented in reverse order, the commit you want is e9e9ccf:
```
$ git checkout -b rpi-4.1.18 e9e9ccf
Switched to a new branch 'rpi-4.1.18'
```

aurel94, 2017-03-13:
Thank you a lot for your answer !

pelwell, 2017-03-30:
Can we close this issue now?

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65