raspberrypi/linux #1895: Login prompt doesn't show up over serial console when UART pin is enabled.

fangfufu, 2017-03-13:
I noticed that if I have ``dtparam=spi=on`` in my ``/boot/config.txt``, login prompt doesn't show up in my serial console, although I can still see all the systemd boot messages. This can be fixed by setting ``dtparam=spi=off`` in ``/boot/config.txt``. However, this does mean that I can't use serial console, if I want to use the SPI bus. Is this a bug, or have I done something wrong?

pelwell, 2017-03-13:
Which model of Pi are you using?

fangfufu, 2017-03-13:
This is repeatable on both RPi 0 and RPi 0W.

pelwell, 2017-03-14:
I've checked on a Zero W with the latest Raspbian and tje serial console appears as I would expect it to.

1. Do you have any other settings in config.txt?

2. What is in cmdline.txt?

3. What hardware do you have attached to your pi, and on which header pins?

fangfufu, 2017-03-14:
Hi, these are the things you requested: 

    $ cat /boot/config.txt
    dtparam=spi=on
    dtparam=i2c_arm=on
    gpu_mem=16
    dtoverlay=dwc2
    enable_uart=1
    $
    $ cat /boot/cmdline.txt
    g_multi.file=/disk.img g_multi.host_addr=00:dc:c8:f7:75:05 g_multi.dev_addr=00:dd:dc:eb:6d:f1 dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_multi

Right now I don't have any hardware attached, other than a USB-TTL cable. It is attached in the same way as this image: 
[image]

pelwell, 2017-03-14:
Those options work for me, although I have omitted the g_multi settings.

What do you get from `dmesg | grep tty`?

fangfufu, 2017-03-14:
Here are the outputs: 

    $ dmesg | grep tty
    [    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2708.boardrev=0x9000c1 bcm2708.serial=0x61090b23 bcm2708.uart_clock=48000000 bcm2708.disk_led_gpio=47 smsc95xx.macaddr=B8:27:EB:09:0B:23 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  g_multi.file=/disk.img g_multi.host_addr=00:dc:c8:f7:75:05 g_multi.dev_addr=00:dd:dc:eb:6d:f1 dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait modules-load=dwc2,g_multi
    [    0.001475] console [tty1] enabled
    [    0.404754] console [ttyS0] disabled
    [    0.406637] 20215040.uart: ttyS0 at MMIO 0x20215040 (irq = 53, base_baud = 31250000) is a 16550
    [    1.021921] console [ttyS0] enabled
    [    1.447312] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 81, base_baud = 0) is a PL011 rev2
    [    3.278538] systemd[1]: Expecting device dev-ttyS0.device...
    [    3.290207] systemd[1]: Expecting device dev-ttyGS0.device...
    [    3.606270] systemd[1]: Starting system-getty.slice.
    [    3.623402] systemd[1]: Created slice system-getty.slice.
    [    3.632275] systemd[1]: Starting system-serial\x2dgetty.slice.
    [    3.646714] systemd[1]: Created slice system-serial\x2dgetty.slice.

pelwell, 2017-03-14:
That's the same as mine, apart from ttyGS0 which is a gadget serial interface.

What about `systemctl | grep getty`?

fangfufu, 2017-03-14:
There you go: 

    $ systemctl | grep getty
    getty@tty1.service                                                                                      loaded active running   Getty on tt
    serial-getty@ttyGS0.service                                                                             loaded active running   Serial Gett
    serial-getty@ttyS0.service                                                                              loaded active running   Serial Gett
    system-getty.slice                                                                                      loaded active active    system-gett
    system-serial\x2dgetty.slice                                                                            loaded active active    system-seri
    getty.target

pelwell, 2017-03-14:
Again, that looks sensible.

I'm running out of ideas - can you eliminate the only real difference to my system by disabling the gadget support?

fangfufu, 2017-03-14:
Turning off gadget support didn't help either. :(

pelwell, 2017-03-14:
Is it possible that you have some application or service that is trying to use SPI and stalling when it does? Other than that I have nothing else to suggest.

You might get some more ideas in the [Interfacing forum](https://www.raspberrypi.org/forums/viewforum.php?f=44).

fangfufu, 2017-03-14:
Okay, I figured out what was causing it. It was PaPiRus's epd-fuse daemon. (Things that are related to this: https://github.com/repaper/gratis/blob/master/PlatformWithOS/RaspberryPi/epd-fuse.sh) 

I am not pinning much hope on them resolving the issue. I filed a related bug report last night to them.
 (https://github.com/PiSupply/PaPiRus/issues/81) 

Sorry for the fuss. I am closing this issue.

closed 2017-03-14 by fangfufu