raspberrypi/linux #1900: C-How to enable Auto-RTS on SC16IS7xx

pmperez, 2017-03-15:
Hello

How could I enable and disable the RTS transimision and automatic reception described in the manual of the chip as hardware flow control?

According to what I see, you have to set the term c_cflag flag in CRTSCTS.

https://github.com/raspberrypi/linux/blob/rpi-4.1.y/drivers/tty/serial/sc16is7xx.c#L805

`	if (termios->c_cflag & CRTSCTS)
		flow |= SC16IS7XX_EFR_AUTOCTS_BIT |
SC16IS7XX_EFR_AUTORTS_BIT;`

The problem is that I manage to send data but I can not receive it because the RTS pin is always up.

Thank you very much

6by9, 2017-03-15:
You've linked to a Linux 4.1 source file when 4.4 is the Pi standard (soon to be 4.9). Are you really using such an old kernel?

Looking at the 4.9 tree
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/Documentation/serial/serial-rs485.txt
If you've enabled RS485 mode then it should automatically enable ```SC16IS7XX_EFCR_AUTO_RS485_BIT``` https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/tty/serial/sc16is7xx.c#L736

pmperez, 2017-03-15:
Thanks @6by9 for rapid response.
I'm using kernel 4.4 and I did not notice that.

I will try to update the kernel with 'sudo rpi-update' and to set the flag that you mention. Just that flag was not working for me in this version of the kernel.
Until new notice

6by9, 2017-03-15:
I didn't check where between 4.1 and 4.9 the driver changed, but you could check the history yourself if you so wished. 4.1 needed extra flags to enable auto RTS.

Do remember that rpi-update gets you the cutting edge firmware and kernel, so regressions are possible. The switch to 4.9 via apt-get is likely to be in the next couple of months anyway, so it should be pretty solid.

pmperez, 2017-03-15:
Effectively, the RTS pin changed after upgrading to 4.9. But even the work is not stable, since I must configure as you said in the serial_rs485 structure and I can not establish a good configuration. I keep checking what the problem is, then I'll comment.

Thanks for the information of when the next kernel version will be available, since I am in the development process and probably that version will serve me much more.

pmperez, 2017-03-15:
Well, I was able to solve my problem, thanks @ 6by9.

The code to control the hardware flow control that worked for me is like this:

`struct serial_rs485 rs485conf;`
`ioctl ( dev, TIOCGRS485, &rs485conf );`
`rs485conf.flags = ( SER_RS485_ENABLED | SER_RS485_RTS_ON_SEND | ~(SER_RS485_RTS_AFTER_SEND) | ~(SER_RS485_RX_DURING_TX));`
`octl ( dev, TIOCSRS485, &rs485conf ) `

Esto pone el modo RS485 en ON, en 1 el pin RTS antes de enviar, en 0 después de enviar y en 0 el modo escritura/lectura a la vez.

Thanks!
Pablo

closed 2017-03-15 by pmperez