raspberrypi/linux #1903: sc16is7xx module reset TX fifo

pmperez, 2017-03-16:
Hello!

After a power failure, the driver detects that the chip has part of the TX buffer with data and I stopped the communication. The message in the log is:
`sc16is7xx spi0.0: chip reports 255 free bytes in TX fifo, but it only has 64`

The driver has the definition of the bit that resets the queue and is set in the startup function, but it seems to me that it would not be working correctly.
In function:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/tty/serial/sc16is7xx.c#L965
Set:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/tty/serial/sc16is7xx.c#L973

I have tried rebooting the module with modprobe, I have removed any load of energy that can remain on the board and the pinout has not been modified.

I hope you have understood.
Thanks.
Pablo

pelwell, 2017-03-16:
I think you have two problems here, neither of which is due to an absent reset:

1. Something is causing SPI communication to the device to fail in a way which is persistent.

2. The SC16IS7XX driver is casting the returned value to a u8, turning error codes into numbers in the range 0-255 - -1=255, -2=254, etc.

What kind of power failure is occurring? Does it affect the whole system, or just the serial interface?

pmperez, 2017-03-16:
> What kind of power failure is occurring? Does it affect the whole system, or just the serial interface?

It was an unexpected general cut in the office that lasted a few minutes. Affect of general form to pi.

In same way I think sc16is7xx_startup is not running.

pelwell, 2017-03-17:
The startup method should be called by the TTY framework before any activity on the port. As I said, I think something else caused the SPI communication to fail, but because the driver just assumes that register accesses will work it continues through the whole startup procedure without actually talking to the device until it reaches this one bit of the code that looks at the returned value and misinterprets an error code as real, unexpected TX FIFO count.

Is this a repeatable problem? If so, what are the steps to reproduce it?

pmperez, 2017-03-17:
It can be repeated by executing a program, where possible in C, that is continuously sending and receiving data, with a few microseconds of difference, and then cut the energy in general.

It should be noted that, I am working on a Compute Module and a custom designed base. The selected gps are: 8 (chip select 0), 9 (spi0_mosi), 10 (spi0_miso), 11 (spi0_sclk), 16 (irq) and 17 (reset).

I am preparing a new image to be able to reproduce the error again.

pelwell, 2017-03-17:
Does your design include power-on-reset logic for the SC16IS752? The reset line should be low for at least 3us, after the supply (and clock, presumably) has stabilised.

pmperez, 2017-03-17:
The new image has no errors.
I think it's like your decision, it's a configuration issue of communication.
Pin reset 17, was not setting with high level and output for when the system starts, if not that custom used `sudo raspi-gpio set 17 op dh`.

pelwell, 2017-03-17:
Please close this issue if it has been resolved.

pmperez, 2017-03-17:
Thanks @pelwell

closed 2017-03-17 by pmperez