raspberrypi/linux #1905: RPi Zero W - can't use BT and UART and SPI1 at same time

NicolaP8, 2017-03-17:
Hi all,
with kernel 4.4.50 of march 03 I'm able to use only two of these peripherals at same time.
If all 3 are enabled when transmitting the first byte from UART the RPi hangs. Hard reset needed.
If BT enabled the serial0 uses TXD1 & RXD1.
If BT disabled (dtoverlay=pi3-disable-bt) the serial0 uses TXD0 & RXD0 and works.
Already tried to change the number and pins of SPI1 CS'. SPI0 haven't a problem.

Minimal changes to a virgin image to see the problem:
1) remove from cmdline.txt "console=/dev/serial0,115200"
2) add to config.txt
  enable_uart=1
  dtoverlay=spi1-1cs
minicom -b 9600 -o -D /dev/serial0

Thanks
      Nicola

pelwell, 2017-03-17:
This is a duplicate of #1573 and #1484. Writing a proper AUX IRQ controller (or adding that feature to the existing AUX driver) is on my list of things to do.

closed 2017-03-17 by pelwell

pelwell, 2017-03-23:
The current trees now include a modified bcm2835-aux driver that also acts as an interrupt controller, and DT modifications to activate it. I've tested it with the console over ttyS0 running an MFRC reader on SPI1.