raspberrypi/linux #1906: pypy-upsteam-dev install issues

Zzyzx-xx, 2017-03-17:
Greetings,
I was doing a dist-upgrade, when Raspbian failed to install pypy-upstream-dev, with the following error:
```
The following packages have unmet dependencies:
 pypy-upstream-dev : PreDepends: pypy-upstream (= 2.2.1-5) but 4.0.1+dfsg-1+rpi1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

Upon looking at http://archive.raspberrypi.org/debian/dists/jessie/main/binary-armhf/Packages, I noticed the following:
```
Package: pypy-upstream
Source: pypy
Version: 4.0.1+dfsg-1+rpi1
Architecture: all
Maintainer: Stefano Rivera <stefanor@debian.org>
Installed-Size: 76
Depends: pypy
Homepage: http://pypy.org/
Priority: extra
Section: oldlibs
Filename: pool/main/p/pypy/pypy-upstream_4.0.1+dfsg-1+rpi1_all.deb
Size: 19724
SHA256: 8d2505a8bd1e9bc65aa832e52c4e4c289d4a09b83baa4b1f8a48cffd832f245d
SHA1: e888e247c8b8ffaf9f268d0f715649686d295dd3
MD5sum: ef94c17fde7e6bd30779b6e2681e19fd
Description: transitional dummy package
  This is a transitional dummy package. It can safely be removed.

Package: pypy-upstream-dev
Source: pypy-upstream
Version: 2.2.1-5
Architecture: armhf
Maintainer: David Schneider <david.schneider@bivab.de>
Installed-Size: 158
Pre-Depends: pypy-upstream (= 2.2.1-5), dpkg (>= 1.15.6~)
Breaks: pypy-dev, pypy-upstream-dev
Replaces: pypy-dev, pypy-upstream-dev
Homepage: http://pypy.org/
Priority: extra
Section: python
Filename: pool/main/p/pypy-upstream/pypy-upstream-dev_2.2.1-5_armhf.deb
Size: 30314
SHA256: 80089e4771b9737186a9a8923bb69100c669cd82d45beab688552e89b72bb6dc
SHA1: 665f8b5d7104b79527a678335614b85f0bea7e7e
MD5sum: 055a32aed60450b541b445d95e2b2f40
Description: PyPy interpreter - Upstream development headers
 PyPy is a fast, compliant alternative implementation of the Python language
 (2.7.3). It has several advantages and distinct features:
  * Speed: thanks to its Just-in-Time compiler, Python programs often run faster on PyPy.
  * Memory usage: large, memory-hungry Python programs might end up taking less
    space than they do in CPython.
  * Compatibility: PyPy is highly compatible with existing python code.
    It supports ctypes and can run popular python libraries like twisted and django.
  * Sandboxing: PyPy provides the ability to run untrusted code in a fully secure way.
  * Stackless: PyPy comes by default with support for stackless mode, providing
    micro-threads for massive concurrency.  As well as other features.
```

Is pypy-upstream completely depreciated now?

pelwell, 2017-03-17:
This repo is for the linux kernel - you should ask the question here:  https://www.raspberrypi.org/forums/viewforum.php?f=31

or file a bug here: https://bugs.launchpad.net/raspbian/+filebug

Zzyzx-xx, 2017-03-17:
Sorry about that.

Cheers!

closed 2017-03-17 by Zzyzx-xx