raspberrypi/linux #1907: g_dbgp doesn't work

jpl888, 2017-03-17:
Hi, I'm trying to use the g_dbgp module Gadget driver to debug Chromebook firmware, and it doesn't work. The module loads without problem, and /dev/ttyGS0 gets created but there is no output.

Using the module with a BeagleBone Black works without problem and I can see output from the firmware on boot, so I know the EHCI debug port on the Chromebook side of things is configured and working correctly.

I enabled the module by enabling EHCI debug in menuconfig under rpi-source, then running: 

make modules_prepare
make M=drivers/usb/gadget, for f in $(find drivers/usb/gadget/ -name *.ko); do sudo cp $f /lib/modules/$(uname -r)/kernel/drivers/usb/gadget/ ; done
depmod -a
modprobe g_dbgp (with no other Gadget modules loaded).

Can you help me to get some debug output and get to the bottom of why it's not working?

Thanks.

jpl888, 2017-03-17:
I should also mention that dtoverlay=dwc2 is in /boot/config.txt and I have restarted to enable that.

jpl888, 2017-03-17:
Here's some example dmesg output:

[ 1562.027300] dwc2 20980000.usb: bound driver dbgp
[ 1562.215775] dwc2 20980000.usb: new device is high-speed
[ 1562.287750] dwc2 20980000.usb: new device is high-speed
[ 1562.348412] dwc2 20980000.usb: new address 5
[ 1801.893820] dwc2 20980000.usb: bound driver dbgp
[ 1802.085470] dwc2 20980000.usb: new device is high-speed
[ 1802.157601] dwc2 20980000.usb: new device is high-speed
[ 1802.218198] dwc2 20980000.usb: new address 6

popcornmix, 2017-03-17:
What kernel version are you using? i.e. ``uname -a``?

jpl888, 2017-03-17:
Linux raspberrypi 4.9.13+ #974 Wed Mar 1 20:04:40 GMT 2017 armv6l GNU/Linux

But I have also tried 4.4 and 4.9 on my way here.

I also had an idea that maybe my Chromecast Ultra power supply mightn't be giving it enough juice or something and just bought another 2.1 amp charger to power it off, swapping cables over in case it was something to do with the micro-usb cables too.

Also tried reading the ttyGS0 as root, in case it was some perms issue.

popcornmix, 2017-03-17:
Try updating. You probably want: https://github.com/raspberrypi/linux/pull/1887

jpl888, 2017-03-17:
Doing that now.

jpl888, 2017-03-17:
On Linux raspberrypi 4.9.14+ #977 Mon Mar 13 18:21:04 GMT 2017 armv6l GNU/Linux and it's still the same.

ED6E0F17, 2017-03-18:
Check the the gadget is showing as "VID 0x0525 , PID 0xc0de" when connected to a PC, and you could try enabling:
````
config USB_G_DBGP_PRINTK
	depends on USB_G_DBGP
	bool "printk"
	help
	  Directly printk() received data. No interaction.
````

jpl888, 2017-03-18:
c710@c710:~$ lsusb
Bus 002 Device 003: ID 0525:c0de Netchip Technology, Inc. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 064e:d251 Suyin Corp. 
Bus 001 Device 003: ID 0489:e04e Foxconn / Hon Hai 
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

With printk I still don't get any console/dmesg output from the debug port on the c710.

jpl888, 2017-03-18:
When the c710 reboots, I do get some more dmesg output about attached USB devices:

[  264.831851] dwc2 20980000.usb: bound driver dbgp
[  265.020440] dwc2 20980000.usb: new device is high-speed
[  265.092384] dwc2 20980000.usb: new device is high-speed
[  265.153059] dwc2 20980000.usb: new address 5
[  287.813186] dwc2 20980000.usb: new device is high-speed
[  287.953649] dwc2 20980000.usb: new address 127
[  288.203372] dwc2 20980000.usb: new device is high-speed
[  288.343681] dwc2 20980000.usb: new address 127
[  288.593328] dwc2 20980000.usb: new device is high-speed
[  288.733681] dwc2 20980000.usb: new address 127
[  289.039847] dwc2 20980000.usb: new device is high-speed
[  289.173573] dwc2 20980000.usb: new address 127
[  289.411227] dwc2 20980000.usb: new device is high-speed
[  289.544993] dwc2 20980000.usb: new address 127
[  289.782855] dwc2 20980000.usb: new device is high-speed
[  289.916417] dwc2 20980000.usb: new address 127
[  294.537204] dwc2 20980000.usb: new device is high-speed
[  294.551369] dwc2 20980000.usb: new address 2
[  304.679421] dwc2 20980000.usb: new device is high-speed
[  304.751412] dwc2 20980000.usb: new device is high-speed
[  304.811997] dwc2 20980000.usb: new address 3

jpl888, 2017-03-18:
I've enable debug, and I'm getting some more interesting output now:

[  794.098149] dwc2 20980000.usb: bound driver dbgp
[  794.286539] dwc2 20980000.usb: new device is high-speed
[  794.347200] dbgp gadget: setup: desc device
[  794.347262] dbgp gadget: setup complete: 0, 18/18
[  794.358502] dwc2 20980000.usb: new device is high-speed
[  794.419148] dwc2 20980000.usb: new address 4
[  794.439174] dbgp gadget: setup: desc device
[  794.439258] dbgp gadget: setup complete: 0, 18/18
[  794.439455] dbgp gadget: setup: failure req 6 v 200
[  794.439860] dbgp gadget: setup: failure req 6 v 200
[  794.440249] dbgp gadget: setup: failure req 6 v 200
[  854.492118] dwc2 20980000.usb: new device is high-speed
[  854.632079] dbgp gadget: setup: desc debug
[  854.632319] dbgp gadget: setup complete: 0, 4/4
[  854.632571] dwc2 20980000.usb: new address 127
[  854.882399] dwc2 20980000.usb: new device is high-speed
[  855.022241] dbgp gadget: setup: desc debug
[  855.022477] dbgp gadget: setup complete: 0, 4/4
[  855.022728] dwc2 20980000.usb: new address 127
[  855.272417] dwc2 20980000.usb: new device is high-speed
[  855.412235] dbgp gadget: setup: desc debug
[  855.412475] dbgp gadget: setup complete: 0, 4/4
[  855.412726] dwc2 20980000.usb: new address 127
[  855.716836] dwc2 20980000.usb: new device is high-speed
[  855.848641] dbgp gadget: setup: desc debug
[  855.848878] dbgp gadget: setup complete: 0, 4/4
[  855.849129] dwc2 20980000.usb: new address 127
[  856.084591] dwc2 20980000.usb: new device is high-speed
[  856.216440] dbgp gadget: setup: desc debug
[  856.216681] dbgp gadget: setup complete: 0, 4/4
[  856.216931] dwc2 20980000.usb: new address 127
[  856.452208] dwc2 20980000.usb: new device is high-speed
[  856.583943] dbgp gadget: setup: desc debug
[  856.584183] dbgp gadget: setup complete: 0, 4/4
[  856.584435] dwc2 20980000.usb: new address 127
[  861.216212] dwc2 20980000.usb: new device is high-speed
[  861.230394] dwc2 20980000.usb: new address 2
[  861.232475] dbgp gadget: setup: desc device
[  861.232518] dbgp gadget: setup complete: 0, 8/8
[  861.232535] dbgp gadget: setup: failure req 6 v 200

jpl888, 2017-03-18:
But apparently, even those "failure" messages are normal according to the EHCI Debug page.

https://www.coreboot.org/EHCI_Gadget_Debug

jpl888, 2017-03-18:
Enabling debug at the dwc2 level and using Gadget Only mode (was getting errors about "suspend" in dual mode) this is what I get:

[dwc2.txt](https://github.com/raspberrypi/linux/files/852419/dwc2.txt)

Also can't help feeling this is similar, if not the same to:

https://www.spinics.net/lists/arm-kernel/msg564614.html

ED6E0F17, 2017-03-18:
It is going to take me a while to read your logs, but the "setup: failure req 6 v 200" message proves that you are getting packets from the usb host.

The spinics bug report is due to the way that upstream have broken g-tx-fifo-size, and you need to be using the dwc2 overlay from rpi-linux-4.9.14 or later. You could try increasing  g-rx-fifo-size, but I do not think that is significant.

kmalkki, 2017-03-22:
You can configure your coreboot side (in menuconfig):
   Debugging->Output verbose USB 2.0 EHCI debug dongle messages

Once you get to OS, sudo cbmem -c should give you USB initialisation as seen from the c710 side.

Failing USB request is this one:
https://github.com/coreboot/coreboot/blob/master/src/drivers/usb/gadget.c#L188

In the dwc2.txt log that's line 137 and the request times out on line 152.

kmalkki, 2017-03-22:
In dwc2_hsotg_process_req_feature(), you need to handle USB_DEVICE_DEBUG_MODE succesfully (nothing to do there). It's just a parallel case next to USB_DEVICE_TEST_MODE.

jpl888, 2017-03-22:
Only thing that stood out to me was *without* Output verbose USB 2.0 EHCI debug dongle messages, and said:

ehci_wait_td error - status=80090d40

Both cbmem outputs are attached.

I don't really know what you mean in your last comment, but I will take a look.

[cbmem.txt](https://github.com/raspberrypi/linux/files/861177/cbmem.txt)
[cbmem_debug.txt](https://github.com/raspberrypi/linux/files/861176/cbmem_debug.txt)

kmalkki, 2017-03-22:
I think you either forgot to connect RPI or the USB port was otherwise completely stuck from previous boots. That cbmem_debug.txt should have been much more verbose.

My second comment refers to this:
https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/usb/dwc2/gadget.c#L1186
case USB_DEVICE_DEBUG_MODE: return 0;

jpl888, 2017-03-22:
You're right - forgot cable wasn't plugged in (had borrowed it for something else).

[cbmem_debug.txt](https://github.com/raspberrypi/linux/files/861247/cbmem_debug.txt)

jpl888, 2017-03-22:
Yep, adding that line did the trick. Although it is kicking out of the serial port halfway through output (presumably because I haven't applied what's left of the EHCI debug patches). Will add those later to confirm. Thank you.

jpl888, 2017-03-22:
Yep, that worked. Changed the kernel back to stock, except for enabling the debug gadget and your one line addition, and then applied a carefully selected (read: guessed) subset of the EHCI debug patches of yours, and all is good. :)

kmalkki, 2017-03-22:
Please take the approriate actions and post patch, linux-usb list probably. I have no time for that atm.

lategoodbye, 2017-04-29:
@jpl888 Do you plan to send an upstream patch?

JamesH65, 2017-05-18:
Closing this issue as questions answered/issue resolved.

closed 2017-05-18 by JamesH65

daiaji, 2023-02-14:
g_dbgp dbgp only work in dwc2? Does the dwc3 device not work at all?

pelwell, 2023-02-14:
You've come to the wrong place to ask questions, especially about dwc3 - Raspberry Pi devices don't use it.