raspberrypi/linux #1908: 4.9 Raspberry Pi0 W Dmesg causing crash

ConnorRigby, 2017-03-17:
# Problem
When trying to boot into a buildroot environment i dont get any kernel output on the gpio uart. 
If i enable usb  serial gadget i can get that to open, but still no kernel output (even if i configure cmdline to use `console=ttyGS0,115200`. when i get to a shell i seem to be missing `ttyS0` (but i see `ttyAMA0`)

```
# ls /dev/tty*
/dev/ttyprintk  /dev/tty53      /dev/tty38      /dev/tty22
/dev/ttyGS0     /dev/tty52      /dev/tty37      /dev/tty21
/dev/ttyAMA0    /dev/tty51      /dev/tty36      /dev/tty20
/dev/tty9       /dev/tty50      /dev/tty35      /dev/tty2
/dev/tty8       /dev/tty5       /dev/tty34      /dev/tty19
/dev/tty7       /dev/tty49      /dev/tty33      /dev/tty18
/dev/tty63      /dev/tty48      /dev/tty32      /dev/tty17
/dev/tty62      /dev/tty47      /dev/tty31      /dev/tty16
/dev/tty61      /dev/tty46      /dev/tty30      /dev/tty15
/dev/tty60      /dev/tty45      /dev/tty3       /dev/tty14
/dev/tty6       /dev/tty44      /dev/tty29      /dev/tty13
/dev/tty59      /dev/tty43      /dev/tty28      /dev/tty12
/dev/tty58      /dev/tty42      /dev/tty27      /dev/tty11
/dev/tty57      /dev/tty41      /dev/tty26      /dev/tty10
/dev/tty56      /dev/tty40      /dev/tty25      /dev/tty1
/dev/tty55      /dev/tty4       /dev/tty24      /dev/tty0
/dev/tty54      /dev/tty39      /dev/tty23      /dev/tty
# dmesg
[    0.000000] Booting Linux on physical CPU 0x0

```

then it hangs here forever. 
Opening `/dev/ttyAMA0` with a serial library also hangs forever.

I also see `AT` a couple times on the `ttyGS0` along with some random non utf characters at boot.

in `config.txt` i have: 
```
core_freq=250
enable_uart=1
dtoverlay=dwc2
dtoverlay=pi3-miniuart-bt
dtoverlay=pi3-disable-bt
```

my device tree files are: 
```
bcm2708-rpi-cm 
bcm2708-rpi-b-plus 
bcm2708-rpi-0-w
```

Let me know if im missing a device tree or something else simple.

pelwell, 2017-03-17:
> dtoverlay=pi3-miniuart-bt
> dtoverlay=pi3-disable-bt

You don't need both of these - pick one.

> when i get to a shell i seem to be missing ttyS0 (but i see ttyAMA0)

That's because of the `dtoverlay=pi3-disable-bt`

1. Delete the pi3-miniuart-bt reference and leave pi3-disable-bt - this will leave ttyAMA0 as the primary (console) uart.
2. Put "console=serial0,115200" in cmdline.txt. This will cause kernel output to go to ttyAMA0.
3. Check your cmdline.txt doesn't include `quiet` - upload the contents so I can see it.

ConnorRigby, 2017-03-17:
ok my `cmdline.txt` file looks like: 
```
console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=squashfs rootwait
```
right now. I thought `serial0` was created at runtime in raspbain?

pelwell, 2017-03-17:
Yes, but the firmware rewrites cmdline.txt before passing it to the kernel, according to the aliases in device tree.

ConnorRigby, 2017-03-17:
Oh thats cool i didnt know that. Will change to `serial0`

pelwell, 2017-03-17:
Pi 0 W needs `bcm2708-rpi-0-w.dtb`.

ConnorRigby, 2017-03-17:
heres a list of my dtb files on my boot partition.
```
➜  BOOT ls -la *.dtb
-rw-r--r-- 1 connor connor 15516 Mar 17 10:16 bcm2708-rpi-0-w.dtb
-rw-r--r-- 1 connor connor 15053 Mar 16 08:54 bcm2708-rpi-b.dtb
-rw-r--r-- 1 connor connor 15312 Mar 17 10:16 bcm2708-rpi-b-plus.dtb
-rw-r--r-- 1 connor connor 14772 Mar 16 08:54 bcm2708-rpi-cm.dtb
-rw-r--r-- 1 connor connor 16379 Mar 16 08:54 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 connor connor 17480 Mar 16 08:54 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 connor connor 16236 Mar 16 08:54 bcm2710-rpi-cm3.dtb
```

pelwell, 2017-03-17:
That looks fine.

Have you tried with just the one overlay?

Are you sure you've wired up the pins correctly?

ConnorRigby, 2017-03-17:
The pins are wired correctly. if i put a raspbain sdcard in it boots fine. 
Should i only put `bcm2708-rpi-0-w.dtb` on the boot partition?

pelwell, 2017-03-17:
I'm afraid I don't really have time to debug your buildroot for you, and this isn't the right place.

ConnorRigby, 2017-03-17:
No problem. Thank you for your time.

closed 2017-03-17 by ConnorRigby