raspberrypi/linux #1909: rpi3 doesn't power up without HDMI  after upgrading to 4.9.16-v7+

PieterGit, 2017-03-19:
After upgrading to kernel 4.9 my rpi won't boot without hdmi cable inserted. As soon as I insert the HDMI cable the rpi boots. I wan't to be able to boot my rpi3 without hdmi device installed.

```
$ grep hdmi /boot/config.txt | grep -v "#"
hdmi_safe=1
hdmi_force_hotplug=1
$ uname -a
Linux pi3 4.9.16-v7+ #978 SMP Sat Mar 18 13:59:01 GMT 2017 armv7l GNU/Linux
```

I tried the `hdmi_edid_file=1` setting I found on 
http://kodi.wiki/view/Raspberry_Pi_FAQ#TV_is_not_detected_unless_powered_on_first
as well, but that did not help either.

Is this a 4.9 problem, or is there something wrong with my `config.txt`

popcornmix, 2017-03-19:
Can you post complete config.txt?
What happens if you remove all custom settings from config.txt?
Are you sure it doesn't boot (rather than just not driving hdmi)?
What does ACT LED do when you try to boot?
Can you try to ssh in?

popcornmix, 2017-03-19:
Also can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this problem.

lategoodbye, 2017-04-17:
This looks similiar to this issue: https://github.com/anholt/linux/issues/95

JamesH65, 2017-05-19:
Closing due to lack of activity.  OP has not responded to information requests.

@PieterGit  Reopen if you feel this issue is still relevant.

closed 2017-05-19 by JamesH65