raspberrypi/linux #1911: Repeated kernel issues after 2.7 days runtime in 4.9 line (here 4.9.14+)

scalvin1, 2017-03-19:
After upgrading to the 4.9 kernel line, my weather station keeps locking up after 2 or 3 days.
Sometimes I cannot log in after this, but now I managed to scrape a few details.

No space/ memory/ swap issues. Used to work in same setup for over two years (with occasional Raspbian updates and rpi-update kernel updates). Moved to btrfs filesystem at around kernel 4.5 and it was fine until 4.9 came along. (Note, swap is on a separate partition.)

Any ideas how to approach this? Would be happy to try out a few things. It seems to be ok after a reboot. Will update to latest kernel 4.9.16 now.

Best regards,
Max

pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.9.14+ #977 Mon Mar 13 18:21:04 GMT 2017 armv6l GNU/Linux

dmesg except:
(Tofu below are square brackets)

[233136.361008] octave-cli: page allocation stalls for 10660ms, order:2, mode:0x24000c0(GFP_KERNEL)
[233136.370269] CPU: 0 PID: 21214 Comm: octave-cli Not tainted 4.9.14+ #977
[233136.377216] Hardware name: BCM2835
[233136.380868] [] (unwind_backtrace) from [] (show_stack+0x20/0x24)
[233136.388960] [] (show_stack) from [] (dump_stack+0x20/0x28)
[233136.396536] [] (dump_stack) from [] (warn_alloc+0xf8/0x120)
[233136.404164] [] (warn_alloc) from [] (__alloc_pages_nodemask+0xaec/0xd04)
[233136.415671] [] (__alloc_pages_nodemask) from [] (__get_free_pages+0x20/0x70)
[233136.430524] [] (__get_free_pages) from [] (pgd_alloc+0x24/0x104)
[233136.444421] [] (pgd_alloc) from [] (mm_init+0xcc/0x168)
[233136.454699] [] (mm_init) from [] (copy_process.part.5+0x968/0x16c8)
[233136.468976] [] (copy_process.part.5) from [] (_do_fork+0xb4/0x3a0)
[233136.483236] [] (_do_fork) from [] (SyS_clone+0x30/0x38)
[233136.493652] [] (SyS_clone) from [] (ret_fast_syscall+0x0/0x1c)
[233136.507669] Mem-Info:
[233136.513262] active_anon:7030 inactive_anon:7049 isolated_anon:1
active_file:19465 inactive_file:17091 isolated_file:31
unevictable:0 dirty:54 writeback:3 unstable:0
slab_reclaimable:2189 slab_unreclaimable:1903
mapped:10587 shmem:2810 pagetables:360 bounce:0
free:590 free_pcp:0 free_cma:0
[233136.564313] Node 0 active_anon:28120kB inactive_anon:28196kB active_file:77860kB inactive_file:68364kB unevictable:0kB isolated(anon):4kB isolated(file):124kB mapped:42348kB dirty:216kB writeback:12kB shmem:11240kB writeback_tmp:0kB unstable:0kB pages_scanned:0 all_unreclaimable? no
[233136.600418] Normal free:2360kB min:1868kB low:2332kB high:2796kB active_anon:28120kB inactive_anon:28196kB active_file:77860kB inactive_file:68364kB unevictable:0kB writepending:228kB present:245760kB managed:233620kB mlocked:0kB slab_reclaimable:8756kB slab_unreclaimable:7612kB kernel_stack:1760kB pagetables:1440kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[233136.645890] lowmem_reserve[]: 0 0
[233136.651882] Normal: 5544kB (UM) 188kB (UM) 016kB 032kB 064kB 0128kB 0256kB 0512kB 01024kB 02048kB 0*4096kB = 2360kB
[233136.668449] 39747 total pagecache pages
[233136.673433] 368 pages in swap cache
[233136.679284] Swap cache stats: add 9416, delete 9048, find 187356/188426
[233136.688374] Free swap = 250180kB
[233136.694109] Total swap = 265212kB
[233136.699709] 61440 pages RAM
[233136.704778] 0 pages HighMem/MovableOnly
[233136.710806] 3035 pages reserved
[233136.716150] 2048 pages cma reserved

culprit is 21214 octave-cli

pi@raspberrypi ~ $ ps aux --forest|cat
USER PID %CPU %MEM VSZ RSS TTY STAT START TIME COMMAND
root 2 0.0 0.0 0 0 ? S Mar16 0:00 [kthreadd]
root 3 0.0 0.0 0 0 ? S Mar16 1:26 _ [ksoftirqd/0]
root 5 0.0 0.0 0 0 ? S< Mar16 0:00 _ [kworker/0:0H]
root 7 0.0 0.0 0 0 ? S< Mar16 0:00 _ [lru-add-drain]
root 8 0.0 0.0 0 0 ? S Mar16 0:00 _ [kdevtmpfs]
root 9 0.0 0.0 0 0 ? S< Mar16 0:00 _ [netns]
root 10 0.0 0.0 0 0 ? S Mar16 0:00 _ [khungtaskd]
root 11 0.0 0.0 0 0 ? S Mar16 0:00 _ [oom_reaper]
root 12 0.0 0.0 0 0 ? S< Mar16 0:00 _ [writeback]
root 13 0.0 0.0 0 0 ? S Mar16 0:03 _ [kcompactd0]
root 14 0.0 0.0 0 0 ? S< Mar16 0:00 _ [crypto]
root 15 0.0 0.0 0 0 ? S< Mar16 0:00 _ [bioset]
root 16 0.0 0.0 0 0 ? S< Mar16 0:00 _ [kblockd]
root 17 0.0 0.0 0 0 ? S< Mar16 0:00 _ [watchdogd]
root 19 0.0 0.0 0 0 ? S< Mar16 0:00 _ [rpciod]
[... 111 lines cut ...]
root 639 0.0 0.6 12368 1496 ? Ss Mar16 0:00 /usr/sbin/cupsd -f
root 641 0.0 0.9 10204 2316 ? Ss Mar16 0:00 /usr/sbin/cups-browsed
root 653 0.0 1.5 39644 3604 ? Ssl Mar16 0:00 /usr/lib/policykit-1/polkitd --no-debug
root 666 0.0 0.5 4040 1232 tty1 Ss+ Mar16 0:00 /sbin/agetty --noclear tty1 linux
root 667 0.0 0.5 3860 1236 ? Ss+ Mar16 0:00 /sbin/agetty --keep-baud 115200 38400 9600 ttyAMA0 vt102
www-data 693 0.0 0.9 8140 2176 ? Ss Mar16 0:23 /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
root 713 0.0 0.5 2576 1228 ? Ss Mar16 0:11 /sbin/wpa_cli -B -P /run/wpa_action.wlan0.pid -i wlan0 -p /var/run/wpa_supplicant -a /sbin/wpa_action
root 751 0.0 0.7 8716 1664 ? Ss Mar16 0:00 dhclient -v -pf /run/dhclient.wlan0.pid -lf /var/lib/dhcp/dhclient.wlan0.leases wlan0
root 865 0.0 1.7 30616 3984 ? Ssl Mar16 0:00 /usr/sbin/console-kit-daemon --no-daemon
root 2656 0.0 0.7 5088 1648 ? Ds Mar16 0:18 /usr/sbin/cron -f
root 21191 0.0 1.1 7016 2672 ? S 12:19 0:00 _ /usr/sbin/CRON -f
root 21204 0.0 0.4 1900 1104 ? Ss 12:19 0:00 | _ /bin/sh -c cd /root && ./sensorplots
root 21206 0.0 1.1 4604 2636 ? S 12:19 0:00 | _ /bin/bash ./sensorplots
root 21214 0.0 15.6 126792 36612 ? D 12:19 0:13 | _ /usr/bin/octave-cli -H /root/sensorplots-20170319.m
root 21614 0.0 0.0 0 0 ? Z 12:29 0:00 _ [cron]
root 21616 0.0 1.1 7016 2672 ? S 12:29 0:00 _ /usr/sbin/CRON -f
root 21625 0.0 0.4 1900 1116 ? Ss 12:29 0:00 | _ /bin/sh -c cd /root && ./sensorcron
root 21629 0.0 1.1 4604 2648 ? S 12:29 0:00 | _ /bin/bash ./sensorcron
root 21676 0.0 0.7 4604 1820 ? D 12:29 0:04 | _ /bin/bash ./sensorcron
root 21695 0.0 1.0 7016 2528 ? D 12:30 0:00 _ /usr/sbin/CRON -f
pi 22206 0.0 1.4 4964 3320 ? Ss 18:58 0:00 /lib/systemd/systemd --user
pi 22209 0.0 0.8 7352 2056 ? S 18:58 0:00 _ (sd-pam)

Similar problem here, apparently not an out of memory or swap issue:
https://archlinuxarm.org/forum/viewtopic.php?f=31&t=6119

popcornmix, 2017-03-19:
To start with, do you have any non-default options in config.txt or cmdline.txt?

scalvin1, 2017-03-19:
Uh oh, looks like I have inadvertently been running the thing on 900MHz.
Undoing this blunder might fix it. I will report back here in a few days.

Thanks,
Max

Electron752, 2017-03-19:
Hi @scalvin1,

I'm just here to learn.  But I was wondering if you by any chance have VC4 as the display driver?  I noticed cma is being uses, and I'm not 100% sure anything would be using cma other then VC4.  You did mention that this is archlinux so they could be using that as their default.

cma is a really great idea, but it's a bit slow and buggy in it's current form.

scalvin1, 2017-03-19:
It is a headless box with a repeatedly upgraded Raspbian Wheezy -> Jessie. So nothing special going on here that I know about. Only the rpi-update kernels and btrfs on the SD Card.

Electron752, 2017-03-19:
Could you send what your kernel command line is:
cat /proc/cmdline

scalvin1, 2017-03-19:
pi@raspberrypi ~ $ cat /proc/cmdline 
bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0xfa00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=btrfs elevator=deadline rootwait rootflags=subvol=rootfs

popcornmix, 2017-03-19:
What does ``vcgencmd version`` report?

scalvin1, 2017-03-19:
(I have updated to latest rpi-update kernel since my first post here.)

pi@raspberrypi ~ $ vcgencmd version
Mar 18 2017 14:02:51 
Copyright (c) 2012 Broadcom
version 95a183d8579333fb96ba81afdc344dd74faf3bc7 (clean) (release)

Electron752, 2017-03-19:
As I'm trying to learn, I wonder if btrfs is part of the issue.  It doesn't get used much and I've heard it's still very much a work in progress.  

I noticed it adds in the SRCU kernel config option.  I've heard of issues with RCU on the RPI, so I wonder if that could be related. I haven't read all the docs yet.

scalvin1, 2017-03-19:
I have experimented with it on several Pis over the last couple of months and found it far more stable on weird SD cards than ext4. It also gives you peace of mind with the scrub function. Just check if it is all ok, no silent corruptions. On my Debian notebook I have been using it for quite some time now without problems. The snapshot function is pure genius (also on RPis).

Electron752, 2017-03-19:
Sounds cool, I'll have to look into it.  I've just heard it's still a work in progress.

Electron752, 2017-03-19:
So back to the initial issue.  How exactly is the RPI locked up?  Can you ssh into it(it being a headless system at all).  It is just some thing are locked up and some things still work.  Any details could be helpful.

popcornmix, 2017-03-19:
@scalvin1 I assume you are using a Pi1 and have arm_freq=900.
Is that with or without an over_voltage setting?

Overclocking the arm without increasing voltage is likely to be unstable.

scalvin1, 2017-03-19:
Note that I seem to have overclocked it to 900MHz for a probably over a year. Issues started after the kernel 4.9 transition. Symptoms are: After 2-3 days, my automated data gathering stops. I can still see the webserver, which keeps working, but data recording and processing stops (this is done by cron stuff). All further cron activity stops after this point. I could then usually not login by ssh anymore and power cycled the Pi. Today this happened for the 4th time and I realized that I could still log in for a change, thus the forensics.

popcornmix, 2017-03-19:
Is this with an over_voltage setting?

scalvin1, 2017-03-19:
The config.txt was like this. (I guess I used raspi-config to set this a long while back.)

#uncomment to overclock the arm. 700 MHz is the default.
arm_freq=900

# for more options see http://elinux.org/RPi_config.txt
gpu_mem=16
core_freq=250
sdram_freq=450
over_voltage=2

popcornmix, 2017-03-19:
Okay, run with that overclock disabled for a few days just to be sure that isn't the cause of the problem.

JamesH65, 2017-03-20:
Just a FYI, that elinux link is out of date (and the content is inaccurate), this link gives the latest config.txt information

https://www.raspberrypi.org/documentation/configuration/config-txt/README.md

Electron752, 2017-03-20:
So I'm actually writing a device driver for upstream that would go into vc04_services that simulates vcgencmd from the kernel and it's actually working.  The idea would be to collect information about the RPI from various locations and stuff it into some sysfs nodes.  Potentially to make it easy for people and possibly even other Linux driver to get to, but the primary purpose would be for diagnostic purposes. At this time it would be a completely read only view of the information.

Siting references is sort of important in upstream, so do have any other good links that could be useful off top top of your head?

I also have the old userland code for vcgencmd that I used as a starting point:
https://github.com/raspberrypi/userland/blob/master/interface/vmcs_host/vc_vchi_gencmd.c

Electron752, 2017-03-20:
Would some of the other userland utilities be good to hook in as well such as tvservice, or would it be better to leave that for the distant future.

Ideally, In the way future it would be good to be able to get some of the firmware log mirrored into the kernel log.  Just things that are catastrophic and actionable.  Potentially even rebooting the RPI if the firmware completely crashes or becomes unresponsive.

popcornmix, 2017-03-21:
I assume using the vc4-kms-v3d driver upstream is the norm, and in that situation tvservice is redundant (firmware does not control hdmi/composite display modes).

A simple watchdog script on the arm could reboot pi if firmware stops responding. Just do a mailbox or gencmd call every 10 seconds and poke the watchdog if it responds correctly. If it hangs or responds incorrectly then Pi will reboot.

Electron752, 2017-03-21:
OK, so I'm thinking simple watchdog would be nice.  The firmware logs would be hard to implement because that's all in vcdbg which is closed source and that's not really a direction I'm interested in going.

Do you think having the driver read the configuration as read from config.txt through vcgencmd would have any value for diagnostic usage?  I know it's a simple text file and people can just load up nano or such, but I'm reading that config.txt is getting some conditional and scripting ability so it may be good to have an easy way to tell exactly what was parsed and vcgencmd is not the most friendly tool around.

If something like that is too little value add, perhaps just the watchdog would be good for now.   I still think vcgencmd would be better then mailbox because those numbers aren't documented and it's too hard to explain where they are coming from.  vcgencmd commands at least gives a simple list.

Electron752, 2017-03-21:
You know what, I think this could all be implemented as a cron job or such in userland.  Maybe it's better that this just gets implemented there.

I think I'm going to put this issue aside for now.

scalvin1, 2017-03-23:
Issue has not appeared again for almost 4 days. So I think it was due to overclocking.
Thanks everyone and my apologies again for not checking this obvious source of trouble first.

pelwell, 2017-03-23:
Small changes in software can cause timing violations or power drains to be masked or exposed, which might explain why you'd not seen the issue before.

Feel free to close the issue if you are satisfied.

closed 2017-03-26 by scalvin1