raspberrypi/linux #1913: ttyAMA0 Problems with 4.9.x Kernel

mypiandrew, 2017-03-21:
I'm seeing problems when running an RS485 adapter circuit added to ttyAMA0 on the latest 4.9 Kernels. I don't see this problem on the 4.4.50 kernel.

I can't immediately see it's a hardware issue as I can flip the version back and forth from 4.4.50 to 4.9.x and the fault moves with that change.

To test I'm using this on the Pi

https://github.com/epsilonrt/mbpoll

and the below on a windows box with a proper serial port (not USB)

https://sourceforge.net/projects/modrssim2/

Looking here for commits  :

https://github.com/Hexxeh/rpi-firmware/commits/master

See below log if that helps

Can anyone suggest what I can do to isolate this further?

```
# rpi-update 52241088c1da59a359110d39c1875cda56496764

# uname -a
Linux raspberrypi 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><01><4D><00><00><00><00><00><00><91><8C>
[1]:    111
[2]:    333
[3]:    0
[4]:    0
[5]:    0

The 4.9.x commits all give a corrupt first character?

# rpi-update13711648ecf6f2e5aed380baaed00d47f56bcc3f

# uname -a
Linux raspberrypi 4.9.11-v7+ #971 SMP Mon Feb 20 20:44:55 GMT 2017 armv7l GNU/Linux

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<FE><01><03><0A><00><6F><01><4D>
Request for slave 254 ignored (not 1)
The responding slave 254 isn't the requested slave 1
Read output (holding) register failed: Response not from requested slave

# rpi-update 70d0e67030661cffbf863f66d71c59dff28fd20c

root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.16-v7+ #978 SMP Sat Mar 18 13:59:01 GMT 2017 armv7l GNU/Linux

# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<F8><01><03><0A><00><6F><01><4D>
Request for slave 248 ignored (not 1)
The responding slave 248 isn't the requested slave 1
Read output (holding) register failed: Response not from requested slave
```

mypiandrew, 2017-03-21:
Interestingly this only appears to affect ttyAMA0 (!)

If I swap out the UARTs and put ttyS0 on pins 32/33 instead of ttyAMA0 :
```

  GPIO 32: level=1 fsel=2 alt=5 func=TXD1
  GPIO 33: level=1 fsel=2 alt=5 func=RXD1
```

And re-run all is well

```
# uname -a
Linux raspberrypi 4.9.16-v7+ #978 SMP Sat Mar 18 13:59:01 GMT 2017 armv7l GNU/Linux

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyS0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyS0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyS0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><00><DE><00><00><00><00><00><00><F3><49>
[1]:    111
[2]:    222
[3]:    0
[4]:    0
[5]:    0

```

mypiandrew, 2017-03-21:
There's a recent commit with something to do with the PL011 UART is this related?

https://github.com/raspberrypi/linux/issues/148

pelwell, 2017-03-22:
I have the same kernel:
```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.9.16-v7+ #978 SMP Sat Mar 18 13:59:01 GMT 2017 armv7l GNU/Linux
```
After installing and building mbpoll and the windows server (I have a USB serial port) I get:
```
pi@raspberrypi ~ $ mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -vmbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1 
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><00><00><00><00><00><00><00><00><00><24><B6>
[1]: 	0
[2]: 	0
[3]: 	0
[4]: 	0
[5]: 	0
```
Running it for the first time after a boot, and repeatedly in a loop, I saw no errors.

You haven't said which Pi you have - my tests were with a Pi 3 using `dtoverlay=pi3-disable-bt` to free up ttyAMA0 from Bluetooth modem duty, and no mention of ttyAMA0 in cmdline.txt.

If reading this hasn't led you to a solution, please tell me what Pi you have, and post the (non-commented-out) contents of config.txt and cmdline.txt.

mypiandrew, 2017-03-22:
Hi Phil

Thanks for replying, appreciate your time

I'm running on a CM3 in a custom carrier board. 

The RS485 is opto-isolated from the main board, but I don't think that's relevant as I'll explain later

I've dug a bit further and found a kind of work around that might help us track this down.

Here's the config :

### config.txt ###
dtoverlay=mmc
dtdebug=on
dtoverlay=uart0,txd0_pin=32,rxd0_pin=33
dtoverlay=uart1,txd1_pin=14,rxd1_pin=15
dtparam=i2c1=on
dtoverlay=i2c-rtc,ds1307
core_freq=250

### cmdline.txt ###
dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait maxcpus=4

I was able to kind of get the system into a working state by disabling address 1 on the windows simulator, running the mbpoll program a few times, then re-enabling address 1 on the windows sim.

http://pastebin.com/z4si7qsY

This made me remember seeing some weird behaviour with ttyAMA0 when talking to a Microchip LoRa module the other week, in that case to get any sense out of the setup you had to do echo "anything" >/dev/ttyAMA0  and then toggle the LoRa module reset line after booting to get it in a constantly known working state.

So going down the same path, yes that appears to make it behave as expected.

In the log below I've powered off/on each time and then logged into the system via putty and run the commands shown.

http://pastebin.com/9BB4V8Ek

I don't understand why this should make it happy, that's really way beyond my software knowledge.

As I say this works fine without any of the above on 4.4.50, and even on 4.9.x ttyS0 works without issue (after removing the console and swapping pins over of course)

If you can see something obvious please let me know, or if there's a way I can step through commits using rpi-update can you point me at the web page to find the list?

Thanks!

pelwell, 2017-03-22:
You can find historical firmware versions here: https://github.com/Hexxeh/rpi-firmware/commits/master
By including the corresponding git commit hash in the rpi-update command you can switch to any version.

I suggest you start with the two commits either side of the jump to 4.9 - both on Feb 20th.
```
sudo rpi-update 52241088c1da59a359110d39c1875cda56496764 # 4.4.50
```
and
```
sudo rpi-update 13711648ecf6f2e5aed380baaed00d47f56bcc3f # 4.9.11
```

mypiandrew, 2017-03-22:
Hi 

Unless I'm missing something (which is entirely possible due to bad head cold) I tried both of those as I mention in the OP 

Any ideas as to why the echo-ing solves the problem?

I'm happy to work on the basis I've muffed something up, but I can't immediately see what it can be?

pelwell, 2017-03-22:
If you've tried adjacent commits and found one succeeds and the other fails then there is little point in trying any others.

Opening and closing the UART probably causes it to flush its FIFOs. The minimal command to try would be:
```
echo -n >/dev/ttyAMA0
```
i.e. echo nothing.

mypiandrew, 2017-03-22:
Yep, that does the same : 

```
root@raspberrypi:~# echo -n >/dev/ttyAMA0
root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><00><DE><00><00><00><00><00><00><F3><49>
[1]:    111
[2]:    222
[3]:    0
[4]:    0
[5]:    0
```

pelwell, 2017-03-22:
Can you post the output of `raspi-gpio get`?

mypiandrew, 2017-03-22:
```
root@raspberrypi:~# raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=4 alt=0 func=SDA1
  GPIO 03: level=1 fsel=4 alt=0 func=SCL1
  GPIO 04: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 05: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=0 alt=  func=INPUT
  GPIO 08: level=1 fsel=0 alt=  func=INPUT
  GPIO 09: level=0 fsel=0 alt=  func=INPUT
  GPIO 10: level=0 fsel=0 alt=  func=INPUT
  GPIO 11: level=0 fsel=0 alt=  func=INPUT
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=1 fsel=2 alt=5 func=TXD1
  GPIO 15: level=1 fsel=2 alt=5 func=RXD1
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 22: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=1 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 35: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 36: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 37: level=0 fsel=0 alt=  func=INPUT
  GPIO 38: level=0 fsel=0 alt=  func=INPUT
  GPIO 39: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 40: level=0 fsel=0 alt=  func=INPUT
  GPIO 41: level=0 fsel=0 alt=  func=INPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=0 fsel=0 alt=  func=INPUT
  GPIO 44: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=0 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 49: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 50: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 51: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 52: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 53: level=1 fsel=7 alt=3 func=SD1_DAT3

```

mypiandrew, 2017-03-22:
That's weird, I didn't get the -n to work this time 

```
login as: root
root@192.168.1.123's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Mar 22 21:24:50 2017 from user-hp
root@raspberrypi:~#  mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<00><01><03><0A><00><6F><00><DE>
ERROR CRC received 0xDE != CRC calculated 0x5DB1
Read output (holding) register failed: Invalid CRC

root@raspberrypi:~# echo -n >/dev/ttyAMA0
root@raspberrypi:~#  mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<F8><01><03><0A><00><6F><00><DE>
Request for slave 248 ignored (not 1)
The responding slave 248 isn't the requested slave 1
Read output (holding) register failed: Response not from requested slave

root@raspberrypi:~# echo -n >/dev/ttyAMA0
root@raspberrypi:~#  mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<F8><01><03><0A><00><6F><00><DE>
Request for slave 248 ignored (not 1)
The responding slave 248 isn't the requested slave 1
Read output (holding) register failed: Response not from requested slave

root@raspberrypi:~# echo "anything" >/dev/ttyAMA0
root@raspberrypi:~#  mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><00><DE><00><00><00><00><00><00><F3><49>
[1]:    111
[2]:    222
[3]:    0
[4]:    0
[5]:    0

```

pelwell, 2017-03-22:
Am I right in thinking that your uart0 overlay is based on the standard uart1 overlay, and preserves the pin pulls?

mypiandrew, 2017-03-22:
TBH I had quite forgotten about that, I only remember making one to add in the RTS/CTS 

Here's the source of the two overlay files I have for uart0

```
oot@raspberrypi:~/device-tree# cat uart0-overlay.dts
/dts-v1/;
/plugin/;
// *********************************************************************************************
// UART0 overlay can be compiled with the below and added via /boot/config.txt :
// dtc -@ -I dts -O dtb -o uart0-overlay.dtb uart0-overlay.dts && cp uart0-overlay.dtb /boot/overlays/uart0-overlay.dtb
//
// *********************************************************************************************

/{
        compatible = "brcm,bcm2708";

        fragment@0 {
                target = <&uart0>;
                __overlay__ {
                        pinctrl-names = "default";
                        pinctrl-0 = <&uart0_pins>;
                        status = "okay";
                };
        };

        fragment@1 {
                target = <&gpio>;
                __overlay__ {
                        uart0_pins: uart0_pins {
                                brcm,pins = <32 33>;
                                brcm,function = <7>; /* ALT3 */
                                brcm,pull = <0 2>;
                        };
                };
        };

        __overrides__ {
                txd0_pin = <&uart0_pins>,"brcm,pins:0";
                rxd0_pin = <&uart0_pins>,"brcm,pins:4";
        };
};
```

This one is for adding the RTS/CTS lines (not installed currently):  

```

root@raspberrypi:~/device-tree# cat uart0-full-overlay.dts
/dts-v1/;
/plugin/;
// *********************************************************************************************
//
// UART0 with full RTS/CTS support overlay can be compiled with the below and added via /boot/config.txt :
// dtc -@ -I dts -O dtb -o uart0-full-overlay.dtb uart0-full-overlay.dts && cp uart0-full-overlay.dtb /boot/overlays/uart0-full-overlay.dtb
//
// *********************************************************************************************

/{
        compatible = "brcm,bcm2708";

        fragment@0 {
                target = <&uart0>;
                __overlay__ {
                        pinctrl-names = "default";
                        pinctrl-0 = <&uart0_pins>;
                        status = "okay";
                };
        };

        fragment@1 {
                target = <&gpio>;
                __overlay__ {
                        uart0_pins: uart0_pins {
                                brcm,pins = <30 31 32 33>; /* UART0: CTS0 RTS0 TX0 RX0 */
                                brcm,function = <7>; /* alt3 */
                                brcm,pull = <0 0 0 2>;
                        };
                };
        };

        fragment@2 {
                target-path = "/aliases";
                __overlay__ {
                        serial0 = "/soc/uart@7e201000";
                };
        };

};

```

Is there a new one I can use now that works with pins 32/33?

mypiandrew, 2017-03-22:
for interest three test/reboot cycles :
http://pastebin.com/YUine619

mypiandrew, 2017-03-22:
root@raspberrypi:/boot/overlays# ls uart* -l
-rwxr-xr-x 1 root root  887 Oct  1 18:14 uart0.dtbo
-rwxr-xr-x 1 root root  869 Oct  1 17:53 uart0-full-overlay.dtb
-rwxr-xr-x 1 root root  887 Oct  1 19:22 uart0-overlay.dtb
-rwxr-xr-x 1 root root  980 Mar 21 21:18 uart1.dtbo
-rwxr-xr-x 1 root root 1012 Jul 19  2016 uart1-full-overlay.dtbo
-rwxr-xr-x 1 root root  996 Jan 17 11:01 uart1-overlay.dtb

pelwell, 2017-03-22:
There is no "official" uart0 overlay - create an issue (or a pull request) if you want one - but yours look fine.

pelwell, 2017-03-23:
Do you have a USB serial port? If so:
1. Plug both ends into the Pi - USB port and relevant GPIO pins - and verify that ttyUSB0 exists.
2. Download this file (https://drive.google.com/file/d/0B_P-i4u-SLBXRkRKcEhPbkhtdVk/view?usp=sharing) as ser.tgz.
3. `tar zxvf ser.tgz`
    This will extract two scripts (Perl - don't judge) - serrx and sertx.
4. `./serrx ttyAMA0 &`
    The Pi will now display the hex code of every character received on ttyAMA0, in rows of 16 bytes.
5. `./sertx ttyUSB0`
    Each time you run sertx you should see the hex digits 00-ff (256 bytes) being displayed by serrx.

I'm curious to see if this works for you - whether all data gets through correctly, and whether any extra characters are inserted. When you've finished the ttyUSB0->ttyAMA0 test, try the reverse direction.

mypiandrew, 2017-03-23:
Hi

See below 

```
root@raspberrypi:~/test# [ Listening on ttyAMA0 ]
0000: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
0010: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
0020: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
0030: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
0040: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
0050: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
0060: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
0070: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
0080: 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f
0090: 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f
00a0: a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af
00b0: b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf
00c0: c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf
00d0: d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df
00e0: e0 e1 e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef
00f0: f0 f1 f2 f3 f4 f5 f6 f7 f8 f9 fa fb fc fd fe ff
```

pelwell, 2017-03-23:
Was that the first access to ttyAMA0. as far as you know?

pelwell, 2017-03-23:
And the reverse direction?

mypiandrew, 2017-03-23:
Yes that's the first thing done once logged in

Here's the reverse flow 

```
[ Sending to ttyAMA0 ]
0000: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
0010: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
0020: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
0030: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
0040: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
0050: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
0060: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
0070: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
0080: 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f
0090: 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f
00a0: a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af
00b0: b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf
00c0: c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf
00d0: d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df
00e0: e0 e1 e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef
00f0: f0 f1 f2 f3 f4 f5 f6 f7 f8 f9 fa fb fc fd fe ff
```

Trying to make it count down rather than up (so we don't start with 00) I've modified the tx code to

```
for (my $i = 256; $i > 0; $i--)
{
    vec($msg, $i, 8) = $i;
}

```

But that's not working as I expected 

```
root@raspberrypi:~/test# [ Listening on ttyAMA0 ]
0000: 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
0010: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
0020: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
0030: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
0040: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
0050: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
0060: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
0070: 70 71 72 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f
0080: 80 81 82 83 84 85 86 87 88 89 8a 8b 8c 8d 8e 8f
0090: 90 91 92 93 94 95 96 97 98 99 9a 9b 9c 9d 9e 9f
00a0: a0 a1 a2 a3 a4 a5 a6 a7 a8 a9 aa ab ac ad ae af
00b0: b0 b1 b2 b3 b4 b5 b6 b7 b8 b9 ba bb bc bd be bf
00c0: c0 c1 c2 c3 c4 c5 c6 c7 c8 c9 ca cb cc cd ce cf
00d0: d0 d1 d2 d3 d4 d5 d6 d7 d8 d9 da db dc dd de df
00e0: e0 e1 e2 e3 e4 e5 e6 e7 e8 e9 ea eb ec ed ee ef
00f0: f0 f1 f2 f3 f4 f5 f6 f7 f8 f9 fa fb fc fd fe ff
0100: 00
```

pelwell, 2017-03-23:
That initialises the array to the same values but working backwards - try:
```
for (my $i = 0; $i < 256; $i++)
{
    vec($msg, 255-$i, 8) = $i;
}
```

pelwell, 2017-03-23:
Or:
```
for (my $i = 0; $i < 256; $i++)
{
    vec($msg, $i, 8) = 255-$i;
}
```

mypiandrew, 2017-03-23:
```
root@raspberrypi:~/test# ./serrx ttyAMA0 &
[1] 761
root@raspberrypi:~/test# [ Listening on ttyAMA0 ]
root@raspberrypi:~/test# ./sertx ttyUSB1
[ Sending to ttyUSB1 ]
0000: ff fe fd fc fb fa f9 f8 f7 f6 f5 f4 f3 f2 f1 f0
0010: ef ee ed ec eb ea e9 e8 e7 e6 e5 e4 e3 e2 e1 e0
0020: df de dd dc db da d9 d8 d7 d6 d5 d4 d3 d2 d1 d0
0030: cf ce cd cc cb ca c9 c8 c7 c6 c5 c4 c3 c2 c1 c0
0040: bf be bd bc bb ba b9 b8 b7 b6 b5 b4 b3 b2 b1 b0
0050: af ae ad ac ab aa a9 a8 a7 a6 a5 a4 a3 a2 a1 a0
0060: 9f 9e 9d 9c 9b 9a 99 98 97 96 95 94 93 92 91 90
0070: 8f 8e 8d 8c 8b 8a 89 88 87 86 85 84 83 82 81 80
0080: 7f 7e 7d 7c 7b 7a 79 78 77 76 75 74 73 72 71 70
0090: 6f 6e 6d 6c 6b 6a 69 68 67 66 65 64 63 62 61 60
00a0: 5f 5e 5d 5c 5b 5a 59 58 57 56 55 54 53 52 51 50
00b0: 4f 4e 4d 4c 4b 4a 49 48 47 46 45 44 43 42 41 40
00c0: 3f 3e 3d 3c 3b 3a 39 38 37 36 35 34 33 32 31 30
00d0: 2f 2e 2d 2c 2b 2a 29 28 27 26 25 24 23 22 21 20
00e0: 1f 1e 1d 1c 1b 1a 19 18 17 16 15 14 13 12 11 10
00f0: 0f 0e 0d 0c 0b 0a 09 08 07 06 05 04 03 02 01 00
```

I've just done a test by unplugging the USB cable (same as used above) from the Pi board and into my Windows box, reboot the Pi board and then ran mbpoll as the first thing from login. 

The same problem/fix as seen above also applies with that setup (which improves my sanity level a bit..)

If you can't replicate this on your setup then there must be a difference in the linux config between my system here and yours.

Here's the boot area on mine 
```
root@raspberrypi:/boot# ls -l
-rwxr-xr-x 1 root root   15516 Mar 21 21:18 bcm2708-rpi-0-w.dtb
-rwxr-xr-x 1 root root   15053 Mar 21 21:18 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   15312 Mar 21 21:18 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   14772 Mar 21 21:18 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   16379 Mar 21 21:18 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   17480 Mar 21 21:18 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   16236 Mar 21 21:18 bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root   50268 Mar 21 21:18 bootcode.bin
-rwxr-xr-x 1 root root     123 Mar 21 22:10 cmdline.txt
-rwxr-xr-x 1 root root    3891 Mar 22 21:49 config.txt
-rwxr-xr-x 1 root root   18693 Aug 21  2015 COPYING.linux
-rwxr-xr-x 1 root root    5982 Jan 16 14:07 dt-blob.bin
-rwxr-xr-x 1 root root    2575 Mar 21 21:18 fixup_cd.dat
-rwxr-xr-x 1 root root    6660 Mar 21 21:18 fixup.dat
-rwxr-xr-x 1 root root    9793 Mar 21 21:18 fixup_db.dat
-rwxr-xr-x 1 root root    9791 Mar 21 21:18 fixup_x.dat
-rwxr-xr-x 1 root root     103 Feb 26  2016 issue.txt
-rwxr-xr-x 1 root root 4574552 Mar 21 21:18 kernel7.img
-rwxr-xr-x 1 root root 4374784 Mar 21 21:18 kernel.img
-rwxr-xr-x 1 root root    1494 Nov 18  2015 LICENCE.broadcom
-rwxr-xr-x 1 root root   18974 Feb 26  2016 LICENSE.oracle
drwxr-xr-x 2 root root   16384 Mar 22 16:14 overlays
-rwxr-xr-x 1 root root  656260 Mar 21 21:18 start_cd.elf
-rwxr-xr-x 1 root root 4984708 Mar 21 21:18 start_db.elf
-rwxr-xr-x 1 root root 2847876 Mar 21 21:18 start.elf
-rwxr-xr-x 1 root root 3930820 Mar 21 21:18 start_x.elf

```

pelwell, 2017-03-23:
What's in your dt-blob.bin? Can you try without it if that makes sense?

mypiandrew, 2017-03-23:
Yeah I just tried that, no difference

What I'm going to do next is download the Jessie lite from the RPi website, flash that and upgrade it via rpi-update and see if it still breaks then

pelwell, 2017-03-23:
Do you have access to a logic analyzer? It would be interesting to 'scope the pins and see what's really going on. The Saleae Logic 4 is the cost of 3-4 Pis and can decode serial, I2C, SPI and CAN. The Bitscope Model 5 is cheaper and looks more capable.

mypiandrew, 2017-03-23:
Progress!

Flashed the stock jessie lite image and then copied across the overlay and blob files I've been using I found I saw the same problem.

So to test a hunch I removed the uart0 overlay files and dt-blob I've added and the altered the config.txt file to use pins 14/15 (which I think it turns out is irrelevant as the config looks hard coded)

That works fine!

```
login as: root
root@192.168.1.123's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Mar 23 15:39:06 2017 from user-hp
root@raspberrypi:~# raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=4 alt=0 func=SDA1
  GPIO 03: level=1 fsel=4 alt=0 func=SCL1
  GPIO 04: level=1 fsel=0 alt=  func=INPUT
  GPIO 05: level=1 fsel=0 alt=  func=INPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=0 alt=  func=INPUT
  GPIO 08: level=1 fsel=0 alt=  func=INPUT
  GPIO 09: level=0 fsel=0 alt=  func=INPUT
  GPIO 10: level=0 fsel=0 alt=  func=INPUT
  GPIO 11: level=0 fsel=0 alt=  func=INPUT
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=1 fsel=4 alt=0 func=TXD0
  GPIO 15: level=1 fsel=4 alt=0 func=RXD0
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=0 alt=  func=INPUT
  GPIO 22: level=0 fsel=0 alt=  func=INPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=1 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=0 fsel=0 alt=  func=INPUT
  GPIO 33: level=1 fsel=0 alt=  func=INPUT
  GPIO 34: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 35: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 36: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 37: level=0 fsel=0 alt=  func=INPUT
  GPIO 38: level=0 fsel=0 alt=  func=INPUT
  GPIO 39: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 40: level=0 fsel=0 alt=  func=INPUT
  GPIO 41: level=0 fsel=0 alt=  func=INPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=0 fsel=0 alt=  func=INPUT
  GPIO 44: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=0 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 49: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 50: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 51: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 52: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 53: level=1 fsel=7 alt=3 func=SD1_DAT3
root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><00><DE><00><00><00><00><00><00><F3><49>
[1]:    111
[2]:    222
[3]:    0
[4]:    0
[5]:    0

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<01><03><0A><00><6F><00><DE><00><00><00><00><00><00><F3><49>
[1]:    111
[2]:    222
[3]:    0
[4]:    0
[5]:    0

root@raspberrypi:~# ls -l /boot/overlays/uart*
-rwxr-xr-x 1 root root  980 Mar 21 21:18 /boot/overlays/uart1.dtbo
-rwxr-xr-x 1 root root 1012 Jul 19  2016 /boot/overlays/uart1-full-overlay.dtbo
-rwxr-xr-x 1 root root  996 Jan 17 11:01 /boot/overlays/uart1-overlay.dtb
root@raspberrypi:~# grep -v \# /boot/config.txt
dtoverlay=mmc
dtdebug=on
dtoverlay=uart0,txd0_pin=14,rxd0_pin=15
dtparam=i2c1=on
dtoverlay=i2c-rtc,ds1307
core_freq=250

```

mypiandrew, 2017-03-23:
However, I'm currently having a problem getting TXD0 and RXD0 to work when set to appear on 32/33 

Exact same hardware setup as before, (using USB-TTL) just moved to different pins and added in a new dtblob.bin and overlay file

```
root@raspberrypi:~# raspi-gpio get
BANK0 (GPIO 0 to 27):
  GPIO 00: level=1 fsel=0 alt=  func=INPUT
  GPIO 01: level=1 fsel=0 alt=  func=INPUT
  GPIO 02: level=1 fsel=4 alt=0 func=SDA1
  GPIO 03: level=1 fsel=4 alt=0 func=SCL1
  GPIO 04: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 05: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 06: level=1 fsel=0 alt=  func=INPUT
  GPIO 07: level=1 fsel=0 alt=  func=INPUT
  GPIO 08: level=1 fsel=0 alt=  func=INPUT
  GPIO 09: level=0 fsel=0 alt=  func=INPUT
  GPIO 10: level=0 fsel=0 alt=  func=INPUT
  GPIO 11: level=0 fsel=0 alt=  func=INPUT
  GPIO 12: level=0 fsel=0 alt=  func=INPUT
  GPIO 13: level=0 fsel=0 alt=  func=INPUT
  GPIO 14: level=0 fsel=0 alt=  func=INPUT
  GPIO 15: level=1 fsel=0 alt=  func=INPUT
  GPIO 16: level=0 fsel=0 alt=  func=INPUT
  GPIO 17: level=0 fsel=0 alt=  func=INPUT
  GPIO 18: level=0 fsel=0 alt=  func=INPUT
  GPIO 19: level=0 fsel=0 alt=  func=INPUT
  GPIO 20: level=0 fsel=0 alt=  func=INPUT
  GPIO 21: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 22: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 23: level=0 fsel=0 alt=  func=INPUT
  GPIO 24: level=0 fsel=0 alt=  func=INPUT
  GPIO 25: level=0 fsel=0 alt=  func=INPUT
  GPIO 26: level=0 fsel=0 alt=  func=INPUT
  GPIO 27: level=0 fsel=0 alt=  func=INPUT
BANK1 (GPIO 28 to 45):
  GPIO 28: level=1 fsel=0 alt=  func=INPUT
  GPIO 29: level=1 fsel=0 alt=  func=INPUT
  GPIO 30: level=0 fsel=0 alt=  func=INPUT
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 35: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 36: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 37: level=0 fsel=0 alt=  func=INPUT
  GPIO 38: level=0 fsel=0 alt=  func=INPUT
  GPIO 39: level=0 fsel=1 alt=  func=OUTPUT
  GPIO 40: level=0 fsel=0 alt=  func=INPUT
  GPIO 41: level=0 fsel=0 alt=  func=INPUT
  GPIO 42: level=0 fsel=0 alt=  func=INPUT
  GPIO 43: level=0 fsel=0 alt=  func=INPUT
  GPIO 44: level=1 fsel=1 alt=  func=OUTPUT
  GPIO 45: level=0 fsel=0 alt=  func=INPUT
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 49: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 50: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 51: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 52: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 53: level=1 fsel=7 alt=3 func=SD1_DAT3
root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<00><01><03><0A><00><6F><00><DE>
ERROR CRC received 0xDE != CRC calculated 0x5DB1
Read output (holding) register failed: Invalid CRC

root@raspberrypi:~# mbpoll -a 1 -b 9600 -P none -t 4 -r 1 -c 5 /dev/ttyAMA0 -1 -v
mbpoll 0.1-23 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2015 epsilonRT, All rights reserved.
This software is governed by the CeCILL license <http://www.cecill.info>

Opening /dev/ttyAMA0 at 9600 bauds (N, 8, 1)
Protocol configuration: Modbus RTU
Slave configuration...: address = [1]
                        start reference = 1, count = 5
Communication.........: /dev/ttyAMA0,       9600-8N1
                        t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[01][03][00][00][00][05][85][C9]
Waiting for a confirmation...
<F8><01><03><0A><00><6F><00><DE>
Request for slave 248 ignored (not 1)
The responding slave 248 isn't the requested slave 1
Read output (holding) register failed: Response not from requested slave)
```

Here's the config 

/boot/config.txt

```
dtoverlay=mmc
dtdebug=on
dtoverlay=uartzero
dtparam=i2c1=on
dtoverlay=i2c-rtc,ds1307
core_freq=250
```

dtblob source :

http://pastebin.com/W7yWD9nS

uartzero overlay 

http://pastebin.com/Dj045HXu

root@raspberrypi:~# ls /boot/overlays/uart* -l
-rwxr-xr-x 1 root root 887 Mar 23 16:27 /boot/overlays/uartzero.dtbo
-rwxr-xr-x 1 root root 887 Mar 23 16:27 /boot/overlays/uartzero-overlay.dtb

mypiandrew, 2017-04-13:
Still broken as of 

# uname -a
Linux raspberrypi 4.9.21-v7+ #986 SMP Tue Apr 11 17:06:55 BST 2017 armv7l GNU/Linux

mypiandrew, 2017-04-24:
Still no joy as of :

```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.9.24-v7+ #991 SMP Sat Apr 22 20:26:57 BST 2017 armv7l GNU/Linux

```

Looking at dmesg, this is interesting :

https://pastebin.com/KvVM9gLr

```
[    0.086536] uart-pl011 3f201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.086892] irq: no irq domain found for /soc/aux@0x7e215000 !
[    0.155144] bcm2835-dma 3f007000.dma: DMA legacy API manager at b780f000, dmachans=0x1
```

`[    0.316903] bcm2835-aux-uart 3f215040.serial: could not get clk: -517`

```
[    1.026062] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[    1.035283] console [ttyS0] disabled
[    1.042373] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 220, base_baud = 31250000) is a 16550
[    2.044293] console [ttyS0] enabled
```

Then when the first time the mbpoll command is run the below appears at the end of dmesg

`[  401.596628] uart-pl011 3f201000.serial: no DMA platform data`

Once you run echo "anything" >/dev/ttyAMA0 it comes good again, but nothing more that I can see is added to dmesg.

Does this give any more info?

pelwell, 2017-04-24:
Those "errors" are unfortunate but transient. and expected. They are a symptom of the fact that device load order is essentially undefined, and that sometimes (or with some configurations) a device that needs a resource may be loaded before the device that provides the resource. When that happens, the device probe functions returns -EPROBE_DEFER and the kernel tries again after some more devices have been loaded. The "no DMA platform data" message is always there.

The firmware behaves differently depending on whether it thinks uart0 (ttyAMA0) or uart1 (ttyS0) is the primary (console) uart. It decides the uart roles like this (in pseudo-code):
```
    if is_node(uart0) and is_alias(serial0) and (alias_value(serial0) != path_to(uart0)):
        primary = uart1
    else
        primary = uart0
```
What do you see after running these commands:
```
$ grep -a . /proc/device-tree/aliases/serial*
$ ls -l /dev/serial*
```

mypiandrew, 2017-04-24:
ah OK, was just looking for anything that appeared unusual 

What's puzzling here is if I put ttyAMA0 on pins 14/15 (as I show above) it works fine, but if I put ttyAMA0 on pins 32/33 that's when it breaks until I do the echo.  Oddly If I put ttyS0 on 32/33 that works fine.  

It's very strange behaviour...

Here's what you've asked for

pi@raspberrypi:/root $ grep -a . /proc/device-tree/aliases/serial*
/proc/device-tree/aliases/serial0:/soc/serial@7e201000
/proc/device-tree/aliases/serial1:/soc/serial@7e215040

pi@raspberrypi:/root $ ls -l /dev/serial*
total 0
drwxr-xr-x 2 root root 60 Jan  1  1970 by-id
drwxr-xr-x 2 root root 60 Jan  1  1970 by-path

And one level down

pi@raspberrypi:/root $ ls -l /dev/serial/*
/dev/serial/by-id:
total 0
lrwxrwxrwx 1 root root 13 Jan  1  1970 usb-FTDI_FT230X_Basic_UART_DN017X67-if00-port0 -> ../../ttyUSB0

/dev/serial/by-path:
total 0
lrwxrwxrwx 1 root root 13 Jan  1  1970 platform-3f980000.usb-usb-0:1.2:1.0-port0 -> ../../ttyUSB0

pelwell, 2017-04-24:
The output tells me that the firmware thinks that ttyAMA0 is the primary (console) UART, and also that you aren't running Raspbian (if you were, /dev/serial0 would be a symlink to /dev/ttyAMA0 and /dev/serial1 to /dev/ttyS0). This is probably not what you want, since you want ttyAMA0 to be left for your own use. 

If you look at the [pi3-miniuart-bt overlay](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/pi3-miniuart-bt-overlay.dts) you will see that it configures the aliases to be the standard (non-Pi3) way round. If you reverse the assignments in one of your overlays it should make your CM3 behave like a Pi3 with respect to UARTs.

mypiandrew, 2017-04-24:
Hi Phil

I really appreciate your time on this

It is Raspbian, honest!  :)

It's Jessie Lite (if my memory serves correct from March 2016 that's had rpi-update through 4.1 => 4.4 => 4.9)

I don't know why the symlinks are not there (or what generates them) but I had the same issue on this even with the latest core image version downloaded from the website. I tried to narrow down if it was an obvious OS problem earlier.

Pardon my slowness here but I'm not quite up to speed on the reasoning here, sorry.

If I'm following this, are you saying that the problem here lies with how the firmware assignment of uart0 & 1 to serial0 &1 alias has been reversed by default - presumably to suit the Pi3 board 

So what we need to do is create a "unswap" overlay, borrowing from the bt overlay you reference, to "correct" this back to how we expect them to be for CM3 usage. 

This could be done by just creating an overlay with a single a section like fragment@4 :

```
fragment@0 {
		target-path = "/aliases";
		__overlay__ {
			serial0 = "/soc/serial@7e215040";
			serial1 = "/soc/serial@7e201000";
		};
	};
```

I'm really confused how, if this is the problem, that it works at all at the moment.  That does not make sense and the more I think about it the more my head hurts!

For what it's worth here's the  kernel cmdline currently being used 

root@raspberrypi:~# cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=serial1,115200  console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

pelwell, 2017-04-24:
I'm not at the point of saying where the problem lies, I'm just eliminating possibilities by eliminating differences. What I _am_ saying is that currently you are using the UARTs backwards compared to how they are meant to be used - serial0 is meant to designate the console UART and serial1 the BT/spare UART. I am suggesting that you swap the roles in your system so that you can use them as they are intended to be used. It might not make any difference to your problem, but it removes a difference to my test setup.

To that end, your overlay looks correct to me.

mypiandrew, 2017-04-24:
YES!

We have a winner :)

So I created a unswap overlay :

```
/dts-v1/;
/plugin/;
/{
        compatible = "brcm,bcm2708";

        fragment@0 {
                target-path = "/aliases";
                __overlay__ {
                        serial0 = "/soc/serial@7e215040";
                        serial1 = "/soc/serial@7e201000";
                };
        };
};
```

`dtc -@ -I dts -O dtb -o unswap.dtb unswap.dts && cp unswap.dtb /boot/overlays/unswap.dtbo`

Added that after the serial port assignments 

```
dtoverlay=mmc
dtdebug=on
dtoverlay=uart0,txd0_pin=32,rxd0_pin=33
dtoverlay=uart1,txd1_pin=14,rxd1_pin=15
dtoverlay=unswap
dtparam=i2c1=on
dtparam=i2c1=on,i2c1_baudrate=180000
dtoverlay=i2c-rtc,ds1307
core_freq=250
```

Next had to do was swap the cmdline console assignment:

`dwc_otg.lpm_enable=0 console=serial0,115200  console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait maxcpus=4
`

Now works first time, interesting there's no error the first time you run it either

I really can't fathom why something like this should /kinda/ work if it's wrong, or what at 4.9 was the straw that broke the camels back either.

Thank you very much for your patience here

pelwell, 2017-04-24:
Thank goodness for that. I would prefer you to call your overlay uart_swap since it points serial0 at uart1 and serial1 at uart0.  Even though that is the default configuration on a Pi 3, it is still swapped with respect to other Pis.

mypiandrew, 2017-04-24:
Sure, will do 

I'm just as thankful we've sorted it too :)

I understand the main thrust of the work done here is aimed at the standard Pi units, but is it worth having some standardised CM overlays?

pelwell, 2017-04-24:
Most overlays have been submitted or requested. Feel free to do either.

closed 2017-04-24 by mypiandrew

danilkorotkov, 2017-06-11:
hi everyone!
i have similar problem with RS485 mode support with usb-485 converter - no response&timeout.
i installed ubuntu mate and my script works! i think it is raspbian bug: broken RS485 mode

jcwren, 2017-09-14:
@pelwell, I'm running into an issue related to this, and I'm trying to figure out how you managed to get what you did in https://github.com/raspberrypi/linux/issues/1913#issuecomment-288416579 working at all. 

The -5 option causes ```MODBUS_RTU_RS485_RTS_ON_SEND``` to be passed to  ```modbus_rtu_set_serial_mode()```. This, in turn passes ```SER_RS485_ENABLED | SER_RS485_RTS_ON_SEND``` to ```ioctl(ctx->s, TIOCSRS485, &rs485conf)```. 

mbpoll isn't checking the return result from ```modbus_rtu_set_serial_mode()```, which for me, is failing with ENOTTY (Inappropriate ioctl for device).

This is with kernel 4.9.35-v7+. Unless I'm reading something wrong (more than entirely possible), the pl011 driver has no support for the RS485 mode (unlike the SC16IS7xx  driver, for example).

The serial port is configured using ```pi3-disable-bt```, ```raspi-gpio get``` shows UART0 assigned to pins 8 & 10, libmodbus is opening /dev/serial0 (which is linked to /dev/ttyAMA0), and I see data being transmitted on the o'scope. 

But given how there's no RS485 support in the pl011 driver, I'm puzzled as to how both transmit and receive worked for you. Perhaps you were not using actual RS485, and since there was no RS485 transceiver actually being switched transmit and receive, it worked as a RS232 port would?

mypiandrew, 2017-09-14:
Hi @jcwren

I was using a MAX13487 which doesn't require RTS flow control.

FWIW I don't think ttyAMA0 comes up with RTS/CTS lines by default, so you'll need to enable them (using a device tree overlay for example) if you want to use them I'd expect

pelwell, 2017-09-14:
@jcwren You're right - I was using a straight RS232 connection. If the use of an RS485 adapter alters the requirements on the RS232 interface then I'm sorry - that wasn't obvious from the original post.

@mypiandrew is correct that flow control is not normally mapped onto the header pins. You can either use an overlay or a utility such as raspi-gpio to change that:
```
raspi-gpio set 16 a3
raspi-gpio set 17 a3
```

jcwren, 2017-09-14:
@mypiandew, @pelwell, thanks. I've got RTS0 configured to pin 17, so I'm good there. Somewhere there's a fork of libmodbus that either uses a GPIO or manually farbles RTS to control the RS485 transceiver. Unfortunately, it's somewhat behind libmodbus master, so I'll have to figure out what it takes to bring it up to date. 

Is there any technical reason the pl011 driver couldn't support the RS485 modes? There's a lot of conflicting information out there that the pl011 does/doesn't support CTS/RTS, does/doesn't support automatic flow control, etc. With a casual glance at the SC16IS7xx, it looks like the RS485 support is not particularly hardware dependent, and perhaps could be ported into the pl011 driver. 

The problem with using a GPIO line and doing it outside the driver means that you now have to account for the delay in the FIFO. And that means either long enough fixed delays (slowing down overall throughput) or risking too short a delay and switching the transceiver back to RX mode before it's actually finished transmitting (bad data).

pelwell, 2017-09-14:
The BCM2835 pl011 does do automatic flow control (although I can't rule out there being a subtle bug). We're using the standard upstream driver largely unmodified, so any serious change like you are suggesting, if possible, would have to go upstream.

Have you considered using a USB RS485 interface? It might be easier and more performant in the long run.

jcwren, 2017-09-14:
What's a little frustrating is that the autoconf tools can't tell the driver doesn't support RS485, only whether the kernel does or not. So when libmodbus builds on a Pi, it says "Yeah, sure, I have TIOCSRS485!" and merrily builds away. Of course, it doesn't know ahead of time that the actual port driver doesn't support it, but still...

I already use a USB adapter, but I was trying to eliminate it, primarily because of size and awkwardness of building into to an enclosure. I laid out a Pi hat PCB using a MAX13451E (I use that on a lot of my other RS485 hardware) plus some other bits, with the expectation that because RTS was available, it should be easily doable. It wasn't until I actually had hardware I could test against that I discovered that the pl011 driver wasn't capable enough. Yeah, sure, maybe some deeper research would have revealed this, but RS485 over a UART with RTS control is pretty basic stuff.

I haven't done any kernel development since the 2.x kernels, but I may take a poke at integrating it in. Seems there's a fair amount of interest in RS485 on Pi's out there, and a fully integrated driver would make that easier.

pelwell, 2017-09-14:
From what you said you may be able to study the SC16IS7xx driver and apply the same logic to pl011 by analogy. Fingers crossed it is that easy, but there may be a reason why it hasn't been done already.

Ooh - this may be helpful: https://www.spinics.net/lists/arm-kernel/msg597518.html

jcwren, 2017-09-14:
Nice find! I've followed up with him to see if there's any other changes. The patch is missing a lot of the supporting infrastructure, but looks pretty cookie cutter to add. He's concerned about the mdelay() call while waiting for the FIFO to clear, but I can live with that. At 115200 baud, that shouldn't be more than a 100us stall.

chinmaymetanoia, 2017-10-05:
Hey sorry for the long wall of text - I really need some help here and I want to include everything I can.
I'm trying to access /dev/serial0 from some sample code for an rfid reader (connected to GPIO 14/15 TxD0/RxD0) but i get the following error in /var/log/syslog and dmesg:- 

`raspberrypi kernel: [ 416.693171] uart-pl011 3f201000.serial: no DMA platform data`

after which the Pi restarts. This is the last and only message I see in syslog after my code begins executing.

Also while booting I'm getting the error : 
```
[FAILED] Failed to start Configure Bluetooth Modems connected by UART.
See 'systemctl status hciuart.service' for details.
```
```
me@raspberrypi:~ $ systemctl status hciuart.service 
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-10-05 06:47:33 IST; 54min ago
  Process: 354 ExecStart=/usr/bin/btuart (code=exited, status=1/FAILURE)

Oct 05 06:47:33 raspberrypi systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Oct 05 06:47:33 raspberrypi btuart[354]: Can't get port settings: Input/output error
Oct 05 06:47:33 raspberrypi btuart[354]: Can't initialize device: Input/output error
Oct 05 06:47:33 raspberrypi systemd[1]: hciuart.service: Control process exited, code=exited status=1
Oct 05 06:47:33 raspberrypi systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
Oct 05 06:47:33 raspberrypi systemd[1]: hciuart.service: Unit entered failed state.
Oct 05 06:47:33 raspberrypi systemd[1]: hciuart.service: Failed with result 'exit-code'.
```

I'm using a Pi 3 
Note below 
1. I have toggled the **enable_uart to 1** using raspi-config and set **dtoverlay=pi3-disable-bt** in **/boot/config.txt** - These are the only changes I made to /boot/config.txt. I have included all of it since I just started using a Pi and might have missed something.
2. I manually **removed console=serial0,115200 from /boot/cmdline.txt**

to switch AMA0 and s0

```
me@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
me@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

me@raspberrypi:~ $ ls -al /dev/serial*
lrwxrwxrwx 1 root root 7 Oct  5 06:47 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Oct  5 06:47 /dev/serial1 -> ttyS0

me@raspberrypi:~ $ ls -al /dev/ttyA*
crw-rw---- 1 root dialout 204, 64 Oct  5 06:47 /dev/ttyAMA0

me@raspberrypi:~ $ ls -al /dev/ttyS0 
crw-rw---- 1 root dialout 4, 64 Oct  5 06:47 /dev/ttyS0

me@raspberrypi:~ $ cat /boot/cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=75cb9c9c-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
me@raspberrypi:~ $ cat /boot/config.txt  | less
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi
dtoverlay=pi3-disable-bt

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
enable_uart=1
````

Any chance I could be doing something wrong here? What should I investigate? Thanks in advance.

chinmaymetanoia, 2017-10-05:
Never mind, it was due to excessive power consumption.