raspberrypi/linux #1914: 2017-03-02 Raspbian build CPU usage bug

xaos11, 2017-03-24:
Using this (latest) build of Raspbian, a CPU usage bug is encountered when using the Amazon AVS Sample App's "wakeWordAgent". This has been confirmed by failing to reproduce the bug with a previous build, and also in a minimal build "DietPi".  Many users have reported this issue. However as we are mere users we wouldn't know how to track down this bug. 

https://github.com/alexa/alexa-avs-sample-app/issues/605
https://forums.developer.amazon.com/questions/61907/avs-sample-app-wake-word-cpu-usage.html

popcornmix, 2017-03-24:
This repo is for linux kernel issues. Your problem doesn't sound like a kernel issue.
But, to be sure, can you report output of ``top`` when the cpu is high?

popcornmix, 2017-03-24:
Also, if this is a regression in the kernel, then can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with this problem.

xaos11, 2017-03-24:
Thanks for the info @popcornmix . A user mentioned on the issue that his is caused by the preinstalled version of OpenBLAS. Closing this issue.

closed 2017-03-24 by xaos11

XECDesign, 2017-03-24:
If this is the OpenBLAS issue, it should be fixed now.