raspberrypi/linux #1916: Possibly accelerating DWC_OTG driver from the VPUs(on uploaded firmware blob)

Electron752, 2017-03-24:
So since ARM64 is winding down, I was thinking about one of the big pieces of the puzzle of getting things into upstream is the dwc_otg driver.  In it's current form, dwc_otg is probably never going to be upstreamed.

I've heard leaks on the Internet and such that the firmware has a RTOS kind of deal going on under the hood.   In wondering if perhaps in the way distant future the FIQ based part of dwc_otg wouldn't be better to run on the VPU itself?  Especially since Linux doesn't have the concept of interrupt priorities or nested interrupts.

I've also noticed from leaks on the Internet, that the same hardware registers are available from the VPU just with a slightly different numbering.  I've also thought the boot process is somewhat backward.  Perhaps something could trigger an upload of firmware bits to accelerate dwc_otg that gets loaded on demand to enable better FIQ support.  I think would be accepted in upstream actually.

It just so happens I've done a bunch of home brew experiments with USB device side on various Microchip PIC processors starting from the PIC18F up to the PIC32.  I realize these processors are intended for a different type of application(and are in a completely different ballpark and zipcode BTW), I suspect some of the same design challenges may be similar.   

This could actually be something that could be implemented using the leaked open source VC compiler that's available on the Internet, or perhaps it would be better done by someone else I don't know.  If this is an open source driver or is completely closed and gets checked into upstream in the /lib/firmware tree is an independent issue.  I could see things working either way.

popcornmix, 2017-03-24:
While the gpu runs an rtos (threadx) is isn't hard real-time.
Interrupts do get disabled for significant periods of time (e.g. callback handling from ISRs) and spinlocks. We had an informal rule that 1ms was the limit for disabling interrupts.

As such it wasn't considered suitable as a way of handling the USB interrupts, compared to the much lower latency of the arm FIQ.

Obviously if you don't care about gpu features, like camera/video decode/3d and are replacing the firmware with something custom you could improve this, but the official firmware isn't suitable for this job.

Electron752, 2017-03-24:
Like I said, I'm not sure replacing the firmware is a path I'm really interesting in going down.  Obviously, a great deal of effort has been invested in it(I read something like 50 people over at least 10 years), I'm not sure starting from scratch is the best approach.

It's just the GPU has those VPUs sitting around and perhaps some of the work could be offloaded.

Obviously, the best solution would be to get Linux to support what's needed such as nested interrupts and possibly interrupt priorities.  I suspect that would take a lot of arguing with people and I'm not sure it would be productive to go down that path.

Electron752, 2017-03-24:
You know, uploading firmware is completely accepted by upstream and I'm not sure people like Intel even run the firmware where they claim they are running it(It is actually really running on the Intel Processor for example).  I think everyone realizes this and nobody cares.

Does anything prevent the FIQ based handler which is coded in ARM code from being part of the uploaded firmware.

popcornmix, 2017-03-24:
> Does anything prevent the FIQ based handler which is coded in ARM code from being part of the uploaded firmware.

What do you mean by firmware? If you mean GPU code, then GPU has no FIQ mechanism.
If you mean is hiding arm code that handles the fiq in a firmware file acceptable to upstream, then you would have to ask upstream.

Electron752, 2017-03-24:
I'm not sure who to ask in upstream actually.  I just know that I see changes being accepted for sound devices and such from people like Intel that are blasting these firmware blobs into who knows what.  It maybe be a FPGA, it might be Intel code, maybe microcode, or it could simply be a license key that's magic and all it does it unlock something that was already around to begin with.  

Nobody knows, nobody cares, and it appears to be completely accepted.

In the case of the RPI, some of it could be FIQ code(because that isn't part of Linux), some of it could run on the VPU, or how knows where it runs actually.   It's just stuff that makes things work.  I would suggest scrambling the firmware somehow.

closed 2017-03-26 by Electron752