raspberrypi/linux #1917: SPIdev cloch polarity ignored.

hugeone, 2017-03-24:
When using SP1 mode is ignored and clock is always inactive low. SOC datasheet shows that microSPI (SPI1 & 2) have this capability.

pelwell, 2017-03-25:
Although the driver does seem to set the CPOL bit when requested, it's possible that calling ..._reset_hw from ..._unprepare_message is causing the clock to be inactive low between messages when you are expecting it to be inactive high. Can you confirm whether polarity within a message is correct?

pelwell, 2017-03-26:
The existing polarity handling code was only added in Linux 4.6. Raspbian is still on 4.4, but if you `sudo rpi-update` you will get a 4.9 kernel with the updated driver.

hugeone, 2017-03-26:
Thank you. After update it works great 🥇 .

> Can you confirm whether polarity within a message is correct?
Unfortunately before update it did not. In desperation I wrote low level driver :)

[image]

pelwell, 2017-03-26:
Can we close this issue? We'll be switching to 4.9 once we feel it's had
sufficient testing.

closed 2017-03-26 by hugeone