raspberrypi/linux #1920: kernel7.img can't boot on 4.11.0-rc3, but works fine on 4.10.5

jeffersonchua, 2017-03-26:
Hi,

Tried to compile kernel for 32-bit 4.11.0-rc3, but the compiled kernel7.img won't boot on pi3. There's nothing displayed on the console except for an 'underscore".

The same config compiled for 4.10.5 booted up and worked fine on pi3.

Both config attached. What am I missing?

Thanks,
Jeff

[config-4105.txt](https://github.com/raspberrypi/linux/files/871057/config-4105.txt)
[config-4110rc3.txt](https://github.com/raspberrypi/linux/files/871058/config-4110rc3.txt)

popcornmix, 2017-03-27:
Can you test with the default config? (bcm2709_defconfig)

timpro, 2017-03-27:
Are you using any overlays ?  Those seem to be missing in 4.11 .

jeffersonchua, 2017-03-27:
> Can you test with the default config? (bcm2709_defconfig)
Tested, still can't boot.

> Are you using any overlays ? Those seem to be missing in 4.11 .

I'm only specifying device_tree=bcm2710-rpi-3-b.dtb in config.txt
and copying arch/arm/boot/dts/overlays/*.dtbo to /boot/overlays

Jeff

pelwell, 2017-03-27:
With the serial port connected to a Pi 2 I get output as far as:
```
[    1.142449] Waiting for root device /dev/mmcblk0p2...
```
then it stalls.

I'm investigating.

pelwell, 2017-03-27:
Am I right in thinking that you are using the mkknlimg script? In the 4.11 tree it is failing to detect some downstream-only strings, which is causing it to load the upstream dtb  (e.g. bcm2836-rpi-2-b.dtb) and failing because the kernel doesn't match.

Official RPi firmware builds no longer use mkknlimg, but my personal build script still does which is why I saw the same problem. Try replacing the mkknlimg step with:
```
cp arch/arm/boot/zImage kernel7,img
```
or
```
sudo cp arch/arm/boot/zImage /boot/kernel7,img
```
and see if you can boot again. You won't need to rebuild the modules - just copy the new kernel onto the SD card. Alternatively, you can force the downstream DTB in config.txt:
```
[pi3]
device_tree=bcm2710-rpi-3-b.dtb
[pi2]
device_tree=bcm2709-rpi-2-b.dtb
[all]
```

pelwell, 2017-03-27:
The mkknlimg in the 4.11 tree should now detect a downstream build again.

timpro, 2017-03-27:
Are you using any overlays ? Those seem to be missing in 4.11 .

Sorry for the noise, there is a quirk in the Giithub web interface: the "overlays" subdirectory is "missing" from the top of arch/arm/boot/dts because the list of files gets cut short after "omap*" at the bottom. I can navigate to it manually though.

Electron752, 2017-03-27:
I'm getting a crash on arm64 now with the overlays.

It's crashing in a DMA/CMA related path to the new mmc driver.

pelwell, 2017-03-27:
It's a WARN rather than a crash, and it's next on the list to look at. My current thinking of that this is another system-wide change that has affected all users of CMA, and the sdhost driver is just the first, but I should know more tomorrow.

pelwell, 2017-03-27:
And can you _try_ to stay on-topic?

jeffersonchua, 2017-03-27:
> Am I right in thinking that you are using the mkknlimg script?

Yes! I'm using "scripts/mkknlimg arch/arm/boot/zImage /boot/kernel7.img"

> cp arch/arm/boot/zImage kernel7,img

Interesting ... didn't know that. Learning something new everyday!

> The mkknlimg in the 4.11 tree should now detect a downstream build again.

Ok, will try that.

Thanks,
Jeff.

Electron752, 2017-03-27:
I'm still having troubling, but I think I know what to do now.  Give me a bit.

jeffersonchua, 2017-03-28:
>  cp arch/arm/boot/zImage kernel7,img
>  The mkknlimg in the 4.11 tree should now detect a downstream build again.

I just tried both, still showing just "_" (underscore) at bootup.

I'm going back to test using default config bcm2709_defconfig.

Jeff

popcornmix, 2017-03-28:
> cp arch/arm/boot/zImage kernel7,img

Note that should be ``kernel7.img``

Electron752, 2017-03-28:
OK, I think we have successfully passed Pluto and are about to engage the warp driver here.

Exactly how many layers of firmware/kernels are actually running on the RPI 3?  That would certainly explain why basic things like displaying video on the screen can't work at all without a mailbox call or something.  I'm not convinced at this point that what people call the kernel isn't just a process running on the inner whatever and is drawing on a full screen window.  Are these processors actual hardware processors or are they just threads of some kind?

Does that RPI board have any kind of NAND or EEPROM on the board that is actually loading the initial layers?

popcornmix, 2017-03-28:
There is no NAND or EEPROM on the Pi.
There is 32k of bootrom embedded in the 2835 (or 2836 or 2837) SoC.
It is *ROM* We cannot change it without changing metal layers in the silicon and taping out a new SoC.

You apply power to the Pi. The VideoCore GPU automatically runs code from the bootrom.
That will detect the sdcard and load bootcode.bin from it where execution continues.
That will load start.elf which will eventually load kernel.img and finally apply power and clocks to the arm where linux will start to boot.

ED6E0F17, 2017-03-28:
None of which gets me any closer to understanding why:

    DMA: failed to allocate 16384 KiB pool for atomic coherent allocation

jeffersonchua, 2017-03-29:
> I'm going back to test using default config bcm2709_defconfig.

default config worked. Now I've to try to see which options are missing in 4.11 as the same config worked for 4.10.5

> Note that should be kernel7.img

Yep. Sorry, my typo. But my .config not working still. At least I'm 4.11 and just need to know what's missing.

Thanks,
Jeff.

Electron752, 2017-03-29:
I'm getting closer here.  I can get Debian arm64 to boot, but Raspbian doesn't work anymore.   It hangs after init runs and I see just a blinking _.

Sound doesn't work any more though.   I'm going to try putting the linux-next version in the tree and see what happens.

I do see some new things that I didn't expect to see.   I think I need to play a bit with the device-trees and such.  I think the .config is correct at this point though.

pelwell, 2017-03-30:
Can we close this issue?

jeffersonchua, 2017-03-31:
Can you give me a day or two? I just need to find out which configs are the one required for 4.11 that's not needed for 4.10? Been really busy.

Thanks,
Jeff.

pelwell, 2017-03-31:
OK - when you find out, please report back and hit Close.

jeffersonchua, 2017-04-04:
After I pulled the latest 4.11.0-rc4 27712d4d73e45bb6ba0ae80cdb88005c54e42c9d, I could reused my old 4.10.6 config and the new kernel booted fine.

Thanks,
Jeff

closed 2017-04-04 by jeffersonchua