raspberrypi/linux #1928: Can't find default configuration "arch/x86/configs/bcm2709_defconfig"!

djrocktarun, 2017-03-29:
Hi all,

I am trying to build kernel for RASPBERRY PI 2/3 and facing this error.

  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
***
 *** Can't find default configuration "arch/x86/configs/bcm2709_defconfig"!
 ***
scripts/kconfig/Makefile:112: recipe for target 'bcm2709_defconfig' failed
make[1]: *** [bcm2709_defconfig] Error 1
Makefile:546: recipe for target 'bcm2709_defconfig' failed
make: *** [bcm2709_defconfig] Error 2

pelwell, 2017-03-29:
You need to include "ARCH=arm" on the kernel make command line. See https://www.raspberrypi.org/documentation/linux/kernel/building.md for full instructions.

djrocktarun, 2017-03-30:
Thanks @pelwell.

Closing this issue. As i was not following instructions.

closed 2017-03-30 by djrocktarun

pabigot, 2020-08-23:
FWIW: the current content of https://www.raspberrypi.org/documentation/linux/kernel/building.md says nothing about `ARCH=arm`, but the fix still works.

pelwell, 2020-08-23:
The building instructions describe compiling _and_ cross-compiling; the cross-compilation instructions are careful to include `ARCH=arm`. You just need to read a bit further.

pabigot, 2020-08-23:
> You just need to read a bit further.

Quite right; I fumbled the search with `ARCG=`.  Thanks.