raspberrypi/linux #1932: Can't get desktop to start on latest Raspbian Jessie Lite image (lightdm fail)

valentt, 2017-03-30:
Steps to reproduce:
1. copy latest Raspbian Jessie Lite image to sdcard
2. install updates, reboot
3. install lightdm package
4. use raspi-config to swith from cli to desktop
5. reboot

After these steps cli is only loaded and lightdm service shows only this:

`sudo systemctl status lightdm
● lightdm.service - Light Display Manager
   Loaded: loaded (/lib/systemd/system/lightdm.service; enabled)
   Active: failed (Result: start-limit) since Thu 2017-03-30 11:05:44 UTC; 39s ago
     Docs: man:lightdm(1)
  Process: 854 ExecStart=/usr/sbin/lightdm (code=exited, status=1/FAILURE)
  Process: 849 ExecStartPre=/bin/sh -c [ "$(cat /etc/X11/default-display-manager 2>/dev/null)" = "/usr/sbin/lightdm" ] (code=exited, status=0/SUCCESS)
 Main PID: 854 (code=exited, status=1/FAILURE)

Mar 30 11:05:44 raspberrypi systemd[1]: Failed to start Light Display Manager.
Mar 30 11:05:44 raspberrypi systemd[1]: Unit lightdm.service entered failed state.
Mar 30 11:05:44 raspberrypi systemd[1]: lightdm.service holdoff time over, scheduling restart.
Mar 30 11:05:44 raspberrypi systemd[1]: Stopping Light Display Manager...
Mar 30 11:05:44 raspberrypi systemd[1]: Starting Light Display Manager...
Mar 30 11:05:44 raspberrypi systemd[1]: lightdm.service start request repeated too quickly, refusing to start.
Mar 30 11:05:44 raspberrypi systemd[1]: Failed to start Light Display Manager.
Mar 30 11:05:44 raspberrypi systemd[1]: Unit lightdm.service entered failed state.
`

pelwell, 2017-03-30:
That's correct - one of the things that makes Jessie Lite "lite" is the absence of the Pixel Desktop (and no XServer, either).

closed 2017-03-30 by pelwell

valentt, 2017-04-03:
I understand that it comes without Xserver, but why does raspi-config suggests installing only "lightdm" package which when choosing "boot into Desktop" from boot menu?

valentt, 2017-04-03:
Should I open a new bug for raspi-config?

pelwell, 2017-04-03:
What do you think, @spl237?

spl237, 2017-04-03:
If you want a desktop, use the full image not the light image - the raspi-config "boot into desktop" option is intended for people running the full image; it is not intended to add everything required to convert a light image into a full image. (lightdm is used to manage the process of booting into the desktop, which is why raspi-config warns if it is not installed.)

pelwell, 2017-04-03:
I think the question is (now) whether the option should not appear, or appear but report an error,  in raspi-config.

spl237, 2017-04-03:
Given that this is the only report we've seen of confusion caused by this, and the difficulty of reliably telling a light image apart from a slightly broken full one, I'm inclined not to fiddle...