raspberrypi/linux #1933: alacarte / main menu editor does not work

muhahacz, 2017-03-30:
```
pi@raspberrypi:~ $ alacarte 

** (alacarte:2468): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files

(alacarte:2468): Gtk-CRITICAL **: gtk_accel_label_set_accel_closure: assertion 'gtk_accel_group_from_accel_closure (accel_closure) != NULL' failed

(alacarte:2468): Gtk-CRITICAL **: gtk_accel_label_set_accel_closure: assertion 'gtk_accel_group_from_accel_closure (accel_closure) != NULL' failed
Traceback (most recent call last):
  File "/usr/bin/alacarte", line 26, in <module>
    main()
  File "/usr/share/alacarte/Alacarte/MainWindow.py", line 468, in main
    app = MainWindow()
  File "/usr/share/alacarte/Alacarte/MainWindow.py", line 44, in __init__
    self.tree.add_from_file(os.path.join(config.pkgdatadir, 'alacarte.ui'))
GLib.Error: gtk-builder-error-quark: Invalid property: GtkScrolledWindow.overlay_scrolling on line 113 (11)
```

```
pi@raspberrypi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2017-03-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f563e32202fad7180c9058dc3ad70bfb7c09f0fb, stage4
```

JamesH65, 2017-03-30:
This is currently being looked at. Note there is a dedicated repo for Pixel stuff.

https://github.com/raspberrypi-ui

JamesH65, 2017-05-18:
Closing, as this is the wrong repo for this report.

closed 2017-05-18 by JamesH65