raspberrypi/linux #1937: 4.9.19 gets kernel panic on Pi Zero (original model)

DougieLawson, 2017-03-30:
https://github.com/Hexxeh/rpi-firmware/commit/edd3e6c00078c2a8c28196030a7f2dd151c277e7 starts to boot on my original model Zero, gets lots of services starting messages but then the screen turns to a cross hatched pattern and the system dies.

Booting to a console there's a kernel panic but not enough details emerge before the system locks up solid to even bother taking a photo.

Currently rolling back to https://github.com/Hexxeh/rpi-firmware/commit/41561448388d90f19f2e40ac62ce1c21c2e064ec

config.txt has
```
[0xebd5f1e8]
# ** Falcon     : ebd5f1e8 **
#dtoverlay=enc28j60,int_pin=25,speed=20000000
gpu_mem=64
dtparam=act_led_activelow=on
disable_audio_dither=1
# ...
[all]
dtparam=audio=on
dtparam=spi=on
dtparam=i2c_arm=on
```

MrEngman, 2017-03-30:
Similar issue with my Pi Zero v1.3. Don't have a display connected. Checking the card on a Pi 3B and it runs OK. config.txt file only has `dtparam=audio=on` enabled.

After trying on the Pi Zero I checked the log files using a card reader on the Pi 3B and it looks as though no log files were created/updated when booting on the Pi Zero.

Same update working OK on my other Pi's, Pi Zero W, Pi 1 B 512MB, B+, 2B and 3B.

pelwell, 2017-03-31:
I see the same problem with the latest firmware on a Zero 1.3, but reverting just bootcode.bin to [the previous version from Mar 22nd](https://github.com/Hexxeh/rpi-firmware/blob/0ead2bdf97eaf8312e08febb3a75addf59010ed0/bootcode.bin) works for me.

pelwell, 2017-03-31:
I've narrowed the problem down to a single commit, which is reverted in [this bootcode.bin](https://drive.google.com/file/d/0B_P-i4u-SLBXcEpRbDhSaVNJQ3c/view?usp=sharing).

popcornmix, 2017-03-31:
Fix pushed to rpi-update firmware.

DougieLawson, 2017-03-31:
Thank you @pelwell & @popcornmix for a very quick fix.

closed 2017-03-31 by DougieLawson