raspberrypi/linux #1939: 4.9.19 can't start with CONFIG_DWCOTG=y

10robinho, 2017-03-31:
As title says, latest kernel from 4.9.y can't start (hanging on rainbow screen) when `CONFIG_DWCOTG=y`. If I disable this option everything works normal.

There is no output on console, I have no idea how to debug it.

Am I missing something or this is a bug?

popcornmix, 2017-03-31:
I'm running latest rpi-update kernel and:
```
pi@domnfs:~ $ uname -a
Linux domnfs 4.9.19-v7+ #983 SMP Thu Mar 30 14:46:28 BST 2017 armv7l GNU/Linux
pi@domnfs:~ $ sudo modprobe configs
pi@domnfs:~ $ zgrep DWCOTG /proc/config.gz 
CONFIG_USB_DWCOTG=y
```
So it is certainly possible to boot with CONFIG_USB_DWCOTG=y.

What type of pi are you running on?
Does latest rpi-update kernel boot for you?
Does it work if you build the kernel using rpi2709_defconfig?

10robinho, 2017-03-31:
> So it is certainly possible to boot with CONFIG_USB_DWCOTG=y.

Sure it is, but I think there is something wrong with kernel configuration.

I build kernel from sources using `rpi2835_defconfig` (that should be correct one for reference, right?).
I just added few options that are unrelated to `USB_DWCOTG` but config is basically the same and very short/clean.
It's for RPI3.

I will investigate more and post here.
It may be:

- my mistake somewhere
- something wrong with `bcm2835_defconfig`
- something wrong with other kernel options required for `USB_DWCOTG` but not auto-selected or required

You mentioned `rpi2709_defconfig`, I will check it. (it is `bcm2709_defconfig` in sources but I guess it's the same thing).

popcornmix, 2017-03-31:
No, use bcm2709_defconfig for pi2/pi3.
bcm2835_defconfig is for building an upstream kernel which supports much fewer drivers (no fiq and so no dwcotg).

popcornmix, 2017-03-31:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

10robinho, 2017-03-31:
@popcornmix thanks for helping out.

closed 2017-03-31 by 10robinho