raspberrypi/linux #1940: Add dtparam to rpi-display-overlay for backlight pin config?

shawaj, 2017-03-31:
Currently, the rpi-display-overlay uses GPIO18 for the led-gpio (for the screen backlight)...which is also an I2S pin...so this stops the board from being used with products that allow for I2S audio. 

Was wondering whether it would be acceptable to adapt the [rpi-display-overlay.dts](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/rpi-display-overlay.dts) to include a dtparam that allows you to override the default setting of GPIO 18 to any other pin of choice.

Happy to do the work / PR but just wanted to check this would be an acceptable change before doing so?

pelwell, 2017-03-31:
But the display is a HAT - how are you planning on rewiring the board?

shawaj, 2017-04-02:
More thinking that we would spin up a similar board that could use the same driver but different backlight pin.

And/or we could manually wire one or use an interface board that moves that pin to another one... If that makes sense.

pelwell, 2017-04-02:
Well in that case, add some parameters and PR away. The last thing I want is two overlays where one would do.

JamesH65, 2017-05-19:
Closing this issue as questions answered/issue resolved.

closed 2017-05-19 by JamesH65

shawaj, 2017-12-28:
just for future reference, have sorted this in #2321