raspberrypi/linux #1941: defconfig differences

ali1234, 2017-04-01:
I ran a diff of bcmrpi_defconfig and bcm2709_defconfig in 4.11.y and I found some differences that don't seem to make sense. This is the full diff:

```
-ARCH_MULTI_V6 y
-ARCH_MULTI_V7 n
-CRYPTO_AES_ARM m
-CRYPTO_CRYPTD m
-CRYPTO_SHA1_ARM m
-CRYPTO_SHA512 m
-USB_AUDIO m
-USB_CDC_COMPOSITE m
-USB_ETH m
-USB_GADGETFS m
-USB_G_ACM_MS m
-USB_G_HID m
-USB_G_MULTI m
-USB_G_PRINTER m
-USB_G_SERIAL m
-USB_G_WEBCAM m
-USB_MASS_STORAGE m
-USB_MIDI_GADGET m
-USB_ZERO m
+CPU_FREQ_GOV_SCHEDUTIL y
+CRYPTO_AES_ARM_BS m
+CRYPTO_SHA1_ARM_NEON m
+GPIO_BCM_EXP y
+GPIO_BCM_VIRT y
+KERNEL_MODE_NEON y
+LOCALVERSION "-v7"
+NEON y
+RCU_TRACE n
+SMP y
+TOUCHSCREEN_EKTF2127 m
+VMSPLIT_2G y
```

Some of these are obvious: the different CPU architecture, the USB gadget stuff, NEON stuff, optimized crypto routines. But then there's the stuff I don't understand:

Why does only bcmrpi have CRYPTO_CRYPTD and CRYPTO_SHA512?

Why does only bcm2709 have CPU_FREQ_GOV_SCHEDUTIL and RCU_TRACE? Are these only for SMP systems?

Why does only bcm2709 have TOUCHSCREEN_EKTF2127?

pelwell, 2017-05-15:
I just independenty noticed the TOUCHSCREEN_EKTF2127 discrepancy. The option must have been added to bcm2709_defconfig somewhere between 4.8 and 4.9, but the commit has been squashed (unless it was a merge error). Do you remember this, @popcornmix?

Apart from that:

1. RCU_TRACE=n is the default on BCM2708, so it won't appear in the defconfig since it only includes non-default options, but something else in BCM2709 configurations is causing it to be enabled - hence the need for it to be explicitly disabled.

2. I'm going to guess that the NEON crypto configs make the other CRYPTO_ configs redundant or unwanted - ah yes, CRYPTO_AES_ARM_BS selects CRYPTO_SIMD which selects CRYPTO_CRYPTD.

3. And the SCHEDUTIL governor option depends on SMP.

popcornmix, 2017-05-16:
I couldn't find a reference to ``TOUCHSCREEN_EKTF2127`` ever being requested. As such I've now removed it (but would be willing to add it to both if anyone actually uses it).

JamesH65, 2017-05-23:
I think this can be closed as answered. All happy with that? Reopen if not!

closed 2017-05-23 by JamesH65