raspberrypi/linux #1942: Problem with bootcode.bin

Matthaiks, 2017-04-01:
My Pi 1 Model B+ won't boot (blank screen, no disk activity) with the latest (commit e5ca26f) bootcode.bin.

popcornmix, 2017-04-01:
Are you sure that is the version you are using? Can you report:
```md5sum bootcode.bin```
The previous version of bootcode did have a problem on Pi1.

Matthaiks, 2017-04-01:
Bad: 60009c193abdc91edc6439223266b00d
Good: 90f09cba99869fd2d4eda0b533546ffe

For sure.

popcornmix, 2017-04-01:
Can you confirm if this works:
https://github.com/raspberrypi/firmware/blob/3e82d90123edb0f2225eff2dede74652c18c8040/boot/bootcode.bin?raw=true

Matthaiks, 2017-04-01:
Yes, it's OK.

popcornmix, 2017-04-01:
What is on sdcard? Raspbian jessie? Are you using noobs?
Any custom config.txt options?

Matthaiks, 2017-04-01:
Only Arch Linux ARM. Default config.txt didn't help.

popcornmix, 2017-04-01:
Ping @XECDesign @pelwell 
Any suggestions other than reverting bootcode change?

pelwell, 2017-04-01:
I don't think we have much choice but to revert the change until we can work out what the problem is.

ED6E0F17, 2017-04-02:
I am picking up a rather strange issue where one of my PiZeros will refuse to start (from rpiboot -d) on a warm reboot;  this has been reported in other issues, so be careful what you are testing.

In my case the issue is only apparent on Linux 4.11, and appears to be resolved by switching back to my  "bcm2708-rpi-b-plus.dtb" from linux 4.9 

I noticed that the 4.11 dtb file was a lot larger, but I have not examined the changes.

pelwell, 2017-04-02:
Does reverting bootcode.bin make any difference? If not, can you create a separate issue?

pelwell, 2017-04-03:
Can you try with [these firmware files](https://drive.google.com/file/d/0B_P-i4u-SLBXWDRKTFp1MzR3MkE/view?usp=sharing)? Functionally they should be the same as pre-issue firmwares, but after booting if you run `sudo vcdbg log msg |& grep DEBUG` it will show some register values that will help to debug this problem.

pelwell, 2017-04-05:
The latest rpi-update firmware release should fix this problem.

closed 2017-04-05 by Matthaiks