raspberrypi/linux #1943: Kernel 4.9.x crash using USB composite gadget HID function

mame82, 2017-04-01:
`usb_f_hid.ko` crashes on `hidg_bind`, when calling `alloc_ep_req`.

Code causing the issue is [here](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/usb/gadget/function/f_hid.c#L663)

Possible patch is described [here](https://patchwork.kernel.org/patch/9494173/) but marked with "revert"

Could you please Point out what is the latest kernel with working HID Composite Gadget function (I was on 4.4.50+ before testing latest kernel)

popcornmix, 2017-04-01:
The 4.4 is the current stable tree for raspbian.
We will be moving to the 4.9 kernel in the near future and that is the testing (rpi-update) kernel.

ED6E0F17, 2017-04-01:
This issue affects all versions of 4.9 - it may be fixed in 4.11, but there are other issues with that release.

Expected fix: https://github.com/raspberrypi/linux/commit/749494b6bdbbaf0899aa1c62a1ad74cd747bce47

popcornmix, 2017-04-01:
@mame82 are you able to build kernel from source?
If you can confirm a patch fixes the issue we may be able to pull it in before it reaches the 4.9 tree.

mame82, 2017-04-01:
@popcornmix 

Unfortunately I haven't got the time to do this in the next weeks (even I would like to in my own interest). Maybe I could ask somebody to test the patch. I'll report back

mame82, 2017-04-01:
A colleague of mine is willing to test the patch tommorow. Could you provide a starting point (tutorial) on cross compiling and deploying the firmware binaries to RPi to save him some time?  I'm sure he is able to do this on its own, but not without initial guidance (newcomer on ARM, but Linux expert otherwise)

popcornmix, 2017-04-01:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

popcornmix, 2017-04-01:
I think if you run ``rpi-update`` and then download [usb_f_hid.ko](https://drive.google.com/uc?id=0B-6zmEDJwxZEV0VsSXFxb3FjVUk&export=download)
Replace the one in ``/lib/modules/4.9.19+/kernel/drivers/usb/gadget/function/usb_f_hid.ko``
and it should include the linked patch.

mame82, 2017-04-01:
@popcornmix thanks for the links. @markusschultheis agreed to dive into this tommorow. I myself had to revert to kernel 4.4.50+, because as said I'm currently working on a major release of a Pi project relying on usb_f_hid (time is running and I need it working for now)

mame82, 2017-04-01:
BTW. I looked into the newer kernel, because of another bug in current USB_f_rndis.ko. Reconnecting the Pi makes the detected bitrate drop from 425Mbps to 0Mbps. A description of the problem is [here](https://github.com/mame82/ratepatch/blob/master/README.md). I haven't opened an issue for this, because I patched the module binary, but had hopes that it would be gone in newer release

JamesH65, 2017-05-19:
@mame82 @popcornmix What is the current status of this one?

popcornmix, 2017-05-19:
I'm waiting to hear is the usb_f_hid.ko I posted helps the issue.
Currently https://github.com/raspberrypi/linux/commit/749494b6bdbbaf0899aa1c62a1ad74cd747bce47 is not in the 4.9 kernel tree, but we can add it if it is confirmed to help.

Fyren, 2017-06-27:
@popcornmix 
I just tried your posted module after running rpi-update, but when attempting to add a HID function I see syslog complains that `usb_f_hid: disagrees about version of symbol module_layout`.  I'm not currently able to build the kernel myself.  I should be able to test a different build of the module if you can make another.
if it matters, my kernel/firmware:
```
$ uname -a
Linux raspberrypi 4.9.34+ #1013 Sun Jun 25 16:41:15 BST 2017 armv6l GNU/Linux
$ /opt/vc/bin/vcgencmd version
Jun 26 2017 17:15:37
Copyright (c) 2012 Broadcom
version 431e1ed6e291705a1a6bfcd62082170e7fafd122 (tainted) (release)
```

BlanketFortGames, 2017-07-10:
I had **success** using the posted module, but only after getting the same kernel version by using
`sudo rpi-update f9c1f6aab19cb827acb4e654f363014c0edfece6`

popcornmix, 2017-07-12:
@BlanketFortGames so you are happy with test module and it fixes the crash for you?

@Fyren here is modules that should work with latest rpi-update kernel (4.9.36):
https://drive.google.com/uc?id=0B-6zmEDJwxZESVVNZm9vMm5DUGc&export=download

@pelwell okay with including upstream patch (https://github.com/raspberrypi/linux/commit/749494b6bdbbaf0899aa1c62a1ad74cd747bce47) in 4.9 tree?

pelwell, 2017-07-12:
No objections from me.

BlanketFortGames, 2017-07-12:
I am happy with the test module.   The original one fixed the crash for kernel 4.9.19, and the new one fixes the crash on 4.9.36.  Thank you for your help.

Fyren, 2017-07-12:
@popcornmix Thanks.  The module also seems to work for me.  It does not crash when creating a HID and the hid_gadget_test sample in the kernel docs works for sending keyboard reports.

popcornmix, 2017-07-12:
Okay pushed to rpi-4.9.y branch. Will be in next rpi-update kernel.

mame82, 2017-07-23:
Hey guys, 
first: many thanks for solving this issue.
As expected my project isn't usable anymore, if installed on current raspbian release.
The user is presented with a kernel crash, as discussed here.
Including `rpi-update` into the [install routine](https://github.com/mame82/P4wnP1/blob/devel/install.sh) of the P4wnP1 devel branch solved the issue (as long as the script is able to finish and the kernel is updated to 4.4.39+ before reboot).

So there're two questions left:
- When will the new kernel be integrated into the raspbain release ?
- As the (headless) install procedure of my project gets more an more complicated (Windows 10 detects the g_ether module as "Serial COM device", the kernel patch needs to be applied to make HID Gadget work before reboot etc. etc.), I wonder under what conditions it is allowed to publish a patched raspbain Image with [P4wnP1](https://github.com/mame82/P4wnP1) preinstalled? I've been asked several times by P4wnP1 users, if it is possible to provide a "prebuilt" image. Till now I haven't taken this step, because as far as I can see, this will cause license issues (Need to follow copyleft GPL). So to keep the question simple:
What conditions have to be met, allowing the publication of a raspbain Image, with P4wnP1 pre installed on github?

One more Thing: as mentioned earlier in this issue the RNDIS composite device (usb_f_rndis.ko) gets detected with a bitrate of 0 bps after unplugging and replugging the Pi to a Windows 10 64bit host. Unfortunately this issue still resides in kernel 4.9.39+.

So again ... many thanks for fixing the HID gadget problem, great community  !!! Keep on going

drudru, 2017-07-31:
Hi - is there a single file copy/headless solution for this?
I have a Pi Zero running the public Raspbian Jessie Lite on Kernel 4.9.35+. I cannot rpi-update easily (at least not for a while).

Can I just drop in a file mentioned above into a particular location?

I have serial console going via a bus pirate. I can easily download the file on my mac, copy it to /boot. Then I can boot the Pi and manually copy the file into its final location (and reboot again).

popcornmix, 2017-08-01:
No - in general a kernel module built for a different kernel version won't load.
You may be able to force it with ``modprobe -f`` but that may or may not have issues. (It is not guaranteed to work).
I'd suggest you run rpi-update when convenient if you want this fix.
Note: you can also run rpi-update from a different linux PC if that is more convenient.

drudru, 2017-08-01:
@popcornmix - thanks - I suspected that might be an issue. I'll do the rpi-update on Pi 2 that I have with keyboard, hdmi, and network.

mame82, 2017-08-01:
A little thank you from the red teaming world, for fixing this module

https://www.youtube.com/watch?v=7fCPsb6quKc

JamesH65, 2017-09-13:
Closing this issue as questions answered/issue resolved.

closed 2017-09-13 by JamesH65