raspberrypi/linux #1944: IPv6 UDP checksum error when sending small packets

njh, 2017-04-02:
Hello,

I think I have found a UDP checksum bug, for packets less than 3 bytes, when sending over IPv6. I have been testing using socat on a Raspberry Pi and Wireshark on a MacBook.

Three byte UDP payload succeeds: 
```
echo -n 'off' | socat -u - udp6-sendto:[2001:8b0:ffd5:3:6959:620b:9ca4:34d1]:5000
```

But a one or two byte payload fails:
```
echo -n 'on' | socat -u - udp6-sendto:[2001:8b0:ffd5:3:6959:620b:9ca4:34d1]:5000
echo -n '1' | socat -u - udp6-sendto:[2001:8b0:ffd5:3:6959:620b:9ca4:34d1]:5000
```

Doing the same with IPv4 succeeds for all payload sizes:
```
echo -n 'off' | socat -u - udp4-sendto:10.108.65.13:5000
echo -n 'on' | socat -u - udp4-sendto:10.108.65.13:5000
echo -n '1' | socat -u - udp4-sendto:10.108.65.13:5000
```

nick.

popcornmix, 2017-04-02:
Does disabling hardware checksums avoid the issue?
```sudo ethtool --offload eth0 rx off tx off```

njh, 2017-04-02:
Just tested and yes it does. Disable hardware checksums and it works fine.

For my purposes, I am not too bothered if there is any performance degradation, so it is a good work-around for me.

Thanks!

Shall I leave this open for a proper fix?

popcornmix, 2017-04-02:
Leave it open. There is a similar issue here: https://github.com/raspberrypi/linux/issues/1832
Probably not exactly the same issue, but microchip has said they will be submitted a patch for #1832 (and a couple of other issues) upstream within about a month.

I'll let them know about this report - perhaps they already have a fix.

popcornmix, 2017-05-09:
Can you test latest rpi-update firmware?
The fix for #1832 disables HW TX CSUM for IPV6 so probably avoids your issue too.

JamesH65, 2017-05-19:
@njh Did you have a chance to test this? If all is OK now please close the issue.

njh, 2017-05-19:
Hi! Will try and do this tomorrow. If it just disables the hardware checksum for IPv6, then it is very likely to be fixed.

njh, 2017-05-22:
Updated my kernel to `4.9.29-v7+` using `rpi-update` and I can confirm that my problem is now fixed. (thanks for making that so easy!)

For reference, the output of `ethtool -k eth0` is:

```
Features for eth0:
rx-checksumming: on
tx-checksumming: on
	tx-checksum-ipv4: on
	tx-checksum-ip-generic: off [fixed]
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: off
	tx-scatter-gather: off [fixed]
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: off [fixed]
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-sctp-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
busy-poll: off [fixed]
hw-tc-offload: off [fixed]
```

closed 2017-05-22 by njh