raspberrypi/linux #1945: 4.11.y doesn't seem to boot over network or usb

ali1234, 2017-04-02:
After building with the default configs, 4.11.y cannot be booted over USB with rpiboot to a zero, or over the network from a tftp server to a pi 3. The serial port produces no console output. It can however be booted using a regular SD card method. In each case the exact same set of /boot files were used, including the same config.txt and overlays (including pi3-disable-bt so that the serial console should work).

This also seems to affect 4.10.y, but 4.9.y and 4.4.y work okay.

popcornmix, 2017-04-02:
I have tested an nfs boot from 4.10 and 4.11 which are fine (but this is using sdcard for /boot files).

@ghollingworth any thoughts?

ghollingworth, 2017-04-03:
So it gets past the splash screen and actually runs the linux kernel?  I'm guessing it could be a dtb problem (i.e. a problem with the firmware loading up the dtb files pre-kernel execution)

You say it produces no output from serial which would be a surprise because you're not even getting early_printk stuff

Gordon

ali1234, 2017-04-03:
It stops on the rainbow screen.

ali1234, 2017-04-03:
Netboot is working today for some reason. This is my testing procedure. I have scripts which automate all of the build process and produce the final boot/ directory:

1. Tested hardware:

* Raspberry Pi 2B connected on ethernet, HDMI, serial UART.
* Raspberry Pi 3B connected on ethernet, HDMI, serial UART.
* Raspberry Pi 0 W connected on USB, HDMI, serial UART.

2. Procedure for building boot:

* Clone firmware repo and copy boot/
* Clone kernel version to test.
* Build kernel with bcmrpi_defconfig and install kernel.img, dts, overlays to boot/
* Build kernel with bcm2709_defconfig and install kernel7.img, dts, overlays to boot/
* Create cmdline.txt:

`dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 elevator=deadline rootwait consoleblank=0`

* Create config.txt:

```
dtoverlay=pi3-disable-bt
dtparam=i2c_arm=on

# RTC
dtoverlay=i2c-rtc,ds3231

# Internal Audio
#dtparam=audio=on

gpu_mem=64
lcd_rotate=2
```

3. Testing:

* Copy boot/* to a blank SD card with only an empty FAT partition (ie the default format they are supplied with)
* Test this SD card in all three systems system.
* Copy just bootcode.bin to another blank SD card.
* Start a tftp server serving the contents of boot/
* Test this on the Raspberry Pi 2B and 3B
* Test the Pi Zero W with: "sudo rpiboot -d boot/"

4. Results:

* Expected result: kernel should boot, print stuff to serial and HDMI console, then panic because there is no root file system.
* Full SD boot: Works correctly in all three systems.
* tftpboot: Works correctly in 2B and 3B.
* rpiboot: Works correctly only with rpi-4.4.y to rpi-4.9.y. With rpi-4.10.y and rpi-4.11.y it hangs at rainbow screen and prints nothing on serial.

ali1234, 2017-04-03:
Here is the script to build boot files:

https://gist.github.com/ali1234/46a53d796b8faa01cf4e67d484fe00d5

Run it like "./build-boot.sh rpi-4.9.y && ./build-boot.sh rpi-4.11.y"

It will produce a boot directory for each commit or branch you ask it to build. These directories contain everything needed for sd/pxe/usb booting except for a root filesystem.

ED6E0F17, 2017-04-08:
I have an intermittent issue with Linux 4.11-rc4 not starting with rpiboot on PiZero, that is resolved by replacing the bcm2708-rpi-b-plus.dtb file with one built for 4.9.20

There seems to be a strong correlation between it working when the PiZero is cold, and it failing if power is re-applied within a few seconds of being disconnected.

ED6E0F17, 2017-04-10:
My issue with warm booting Linux 4.11 with rpiboot is resolved by removing unused features from the decompiled dtb to get the size back to 16k.

The issue is not present when booting from sdcard.

JamesH65, 2017-05-19:
@ali1234 Are you happy this issue can be closed? It seems that you are now working ok.

closed 2017-06-02 by ghollingworth