raspberrypi/linux #1946: Kernel 4.9.x up to 4.11.0~rc4: RPi2 and RPi3 freeze when usind Kodi and pvr.mythtv addon

mkreisl, 2017-04-03:
This issue happens, if you are starting live-TV stream or recording using Kodi v17 and mythtv backend. 
Starting of stream is ok, but after some seconds stream is stopped and usually system is freezing. Neither network or disk access is possible. 

Nothing can be found in logs before freezing, so unfortunately I can not provide more information

Pi1 is working, but Pi2 (using usb wlan dongle) and Pi3 (using onboard wlan) showing this issue.

Kernel up to 4.8 is working, but using .config from this kernel did not make any difference

Standard .config for 4.9 kernel can be found here: https://github.com/xbianonpi/xbian-package-kernel/blob/master/extra-files/rpi2-4.9.y/.config

Issue reported first here: https://github.com/xbianonpi/xbian/issues/837

popcornmix, 2017-04-05:
Does it occur with raspbian using Kodi 17 (e.g. installed from pplware repo).

mkreisl, 2017-04-05:
No, I'm using XBian distribution. Don't use Raspbian

popcornmix, 2017-04-05:
As xbian uses custom kernel and kodi patches it is hard to offer advice.
If you can reproduce the issue using raspbian then we'll know the issue is present in our kernel and we can try to reproduce.
If this works correctly on raspbian then the issue is likely to reside in xbian's kernel and that should help narrow down the problem.

mkreisl, 2017-04-05:
I know that this is very difficult to reproduce. 

Our kernel is based on your defconfig, just using BTRFS as root fs (so build in kernel). Already build standard kernel wiithout any additional patches or config changes (only btrfs in kernel), but did not make any difference

mkreisl, 2017-04-05:
Will try to install a standard Raspbian and pplware tonight.

Btw, I'm wondering why Raspbian image still uses kernel 4.4 ???

popcornmix, 2017-04-05:
You can run rpi-update to get raspbian up to 4.9.
We will bump the default apt-get kernel to 4.9 for the next relase (expected within the next month).

popcornmix, 2017-04-05:
Can you provide some more info on what leads to the crash?

Does it still occur when using ethernet rather than wifi? 
What is format of video (e.g. mpeg-2/H.264? HD/SD?). 
If deinterlace is disabled does the problem still occur?
If you play the recording through the file interface, rather than MythTV add-on, does the problem still occur?
Any different with omxplayer enabled or disabled?

mkreisl, 2017-04-05:
I do not use ethernet, tested with different wlan dongles. But I suppose that @ssakatausa is using ethernet.
Used ethernet now: does not work. Seems eth is much more critical than wlan, was not able to start live-tv

Tested both formats, SD (mpeg) and HD (720p), does not make difference. Already tried  mmal and omxplayer and disabled both, nothing helped. 
Deinterlace is set to Auto, so it should not be used for 720p

I did not test file interface right now, but @ssakatausa did it (see this post: https://github.com/xbianonpi/xbian/issues/837#issuecomment-291019827)
Quick test now. Seems to work here

Usually I have mmal enabled and omxplayer disabled (have no license for my rpi3)

popcornmix, 2017-04-05:
Please provide a kodi.log.
I want to know the video format and whether deinterlace is actually active (whether or not it should or shouldn't be).

mkreisl, 2017-04-05:
Ok, will do it, but have to 'repair' my rpi3 first :disappointed:

mkreisl, 2017-04-05:
So, back to wlan, using kernel from today, same issue:

mmal enabled, omx disabled, deinterlace disabled, playing 720p live tv
Kodi was playing stream for some seconds and then stop

Here is kodi log http://paste.ubuntu.com/24321134/

popcornmix, 2017-04-05:
I can't see any obvious errors in log. If the problem still occurs with mmal/omplayer disabled then can you capture that log, with "video" component specific debug also enabled.

You do have a number of errors like:
```
Apr  5 18:01:37 kmxbilr2 198  73.584488 T:1525674752   ERROR: PVR - GetChannelGroupMembers - addon ':http://kmxbmc:6544' returned an error: invalid parameters for this method
```
Are you sure mythtv add-on and backend are using compatible versions?
Where is mythtv backend running? On a separate machine on network?

mkreisl, 2017-04-05:
Separate machine, running under openSUSE 13.1

When start testing, it was a stone old version 0.26.1. Now I'm running 0.27.3 (updated last Monday).

I had to scan all channels (used dvb-t 4 years ago, and now using dvb-c) and no channel editing done after scan. Just using all channels mythtv found

FYI: RPi1 is running well and i.MX6 (running kernel 4.8) has absolutely no issue with that configuration

lategoodbye, 2017-04-05:
@mkreisl Do you see any kernel issues via UART? What happen to the ACT LED?

mkreisl, 2017-04-05:
@lategoodbye 
Unfortunately I do not have access to the UART (USB -> TTL hw currently unavailable)

ACT LED is still active (flashes as usual)

As already told, Kodi is not freezing until disk or network access - time is is updated and is still responding on OK button

popcornmix, 2017-04-05:
If you have ssh-ed in from before the freeze, can you still run commands when it freezes?

mkreisl, 2017-04-05:
No, no response. Network and disk i/o (sc-card and usb disk - tested both) completely dead

mkreisl, 2017-04-05:
Last test for today:

Built Kodi v17.1 rc2 from scratch without all the XBian patches - it's Kodi vanilla now - and fresh .kodi configuration (nothing special configured everything default, only mythtv addon activated), but result is as usual: freeze when starting live-tv :rage:

ssakatausa, 2017-04-05:
Let me add a couple of things. 
- mkreisl is right that I am using ethernet, not wlan. 
- My mythtv backend is v0.28.

mkreisl, 2017-04-06:
Did another test:

1st step: Installed Raspbian Lite and updated Kernel to 4.9.20-v7
2nd step: copied (rsynced --delete) my XBian installation over that root partition, but left /lib/modules unchanged
3rd step: boot XBian with original Raspbian kernel and /boot partition
4th step: test live-tv from MythTV backend again: same issue: freeze after starting stream

So, it's not XBian kernel, it's not XBian /boot partition and it's not BTRFS file system

popcornmix, 2017-04-06:
What is the best way to reproduce this?
I can install MythTV on another linux machine, but I don't have PVR hardware.
It is possible to download a suitable file and have MythTV serve it up?
Does it matter what the file is?

e.g. can you download, say [this file](https://drive.google.com/file/d/0BwxFVkl63-lETXlOa3hEMmVyZGM/view) and have kodi play it through MythTV? Does that exhibit the crash?

mkreisl, 2017-04-06:
> What is the best way to reproduce this?

When having running MythTV backend should not be problem to reproduce

> I can install MythTV on another linux machine, but I don't have PVR hardware.
> It is possible to download a suitable file and have MythTV serve it up?

Don't know, being not MythTV expert. MythTV stores a lot in mysql database, so I don't believe that you can include recordings into MythTV easily - but
it also provides a Video section, but never played with that bc using Kodi library. 

> Does it matter what the file is?

No, can be mpeg or h.264. MythTV stores recordings and live-tv as (AFAIK already demuxed) .mpg, not as .TS files

ssakatausa, 2017-04-06:
@popcornmix 

I think that I've found a way to set up a testing environment without requiring a tuner.  In short, you can use an HLS recorder in lieu of TV tuner in mythtv backend (but with a little bit of twists). More concretely, you can configure the mythtv backend in the following way.

1. In the "Capture Card" section, choose "IPTV" for "card type" and use the attached m3u file for the m3u URL (suitably adjusting for your environment).

2. In the "Video Source Setup", name the video source, say, "HLS", and choose "No grabber" for "Listing grabber". 

3. In the "Input Connections" section, choose the one staring with "[FREEBOX ..." (there should be no other choice). Choose "HLS" for "Video source" and perform "Scan for channels" by "M3U import".

If things go well, mythtv backend now should offer three channels.

Unfortunately, the above setup does not work for the live TV viewing in kodi, probably because pvr.mythtv has no idea how to talk withl MythTV's HLS recorder. So, you need to take one extra step, i.e., youu need to install and configure a mythtv frontend. Using the frontend, watch a channel for, say, 5 minutes . This makes mythtv backend create a 5-minute TV recording in the live TV folder, which shows up in the "Recently played channel" area on the TV screen in Kodi. You can now use this recorded TV clip in investigating how Kodi behaves dealing with mythtv.

[playlist-hls.m3u.txt](https://github.com/raspberrypi/linux/files/904161/playlist-hls.m3u.txt)

mkreisl, 2017-04-07:
Good news before weekend :smile: 

Seems found culprit:

XBian uses dynamic priority setting per default, means that Kodi becomes higher priority when playing media (`schedtool -R -p 3 ...` does this)

Seems, that kernels >= 4.9 have different behavior or issue in case of priority change. Turning it off and voila, all live-tv and recordings are played without any freezes.

I already reported when playing live-tv or recordings on RPi1 no freeze occurred. That's because setting is already turned off since weeks. I did that due to Kodi 17 final did not start correctly when setting is on (already used 4.9 kernel). But unfortunately forgot to check if this is active per default and on my RPi3.

popcornmix, 2017-04-07:
Interesting. After removing schedtool when playing recordings what is the cpu usage like?

mkreisl, 2017-04-07:
In top, kodi uses ~40% and cpu is > 85% idle when playing 720p live tv (rpi3, not overclocked, kernel 4.10.8)

popcornmix, 2017-04-07:
Strange. I could understand very high cpu along with "realtime" priority basically killing the system but I wouldn't have expected it with low cpu.

Perhaps it holds off kernel tasks "enough" to cause issues.

mkreisl, 2017-04-07:
I have absolutely no idea. But it should never happen that system does not respond to pings any more

mkreisl, 2017-04-07:
Maybe schedutil does not deal with ondemand governor. Perhaps schedutil governor is required or simply does not support arm's. Many many ???

JamesH65, 2017-05-19:
@mkreisl @popcornmix Anything further required on this issue? Can we close it or is there some ongoing investigation requirred?

closed 2017-09-13 by JamesH65