raspberrypi/linux #1948: vc4-kms-v3d-overlay + HDMI -> OOps in rpi_touchscreen_dsi_probe

schnitzeltony, 2017-04-05:
Kernel 4.9.16 build in Yocto/OE environment from this source complains:

[kernel-log.txt](https://github.com/raspberrypi/linux/files/900781/kernel-log.txt)

I see this issue with all my setups:

- Pi3 + audioinjector-stereo
- Pi2 + hifiberry-plus

Similar mentioned in #1869

pelwell, 2017-04-06:
@anholt I think there is a bug in `rpi_touchscreen_dsi_probe` - `of_find_i2c_device_by_node` (and hence `rpi_touchscreen_get_i2c`) can return NULL, which isn't caught by `IS_ERR(tc->bridge_i2c)`, and leads to a NULL pointer dereference in i2c_smbus_read_byte_data via rpi_touchscreen_i2c_read.

JamesH65, 2017-05-19:
@schnitzeltony Has the linked fix fixed this issue? Please report any further issues or close please.

schnitzeltony, 2017-05-19:
On Fri, May 19, 2017 at 11:19 AM, James Hughes <notifications@github.com>
wrote:

> @schnitzeltony <https://github.com/schnitzeltony> Has the linked fix
> fixed this issue? Please report any further issues or close please.
>
I was under the impression I had closed this issue - will check..

closed 2017-05-19 by schnitzeltony