raspberrypi/linux #1951: Wireless adapter not recognized by applet even works on command line

tomasnovotnyspsse, 2017-04-07:
I have TP-LINK TL-WN725N WiFi dongle.
When I connect it, I'm able to see Wi-Fi networks by iwlist command, but the applet write "No wireless interfaces found". It used to worked with no problem, but now I'm not able to configure it through the default applet.

$ iwconfig 
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

$ iwlist wlan0 scanning
wlan0     Scan completed :
          Cell 01 - Address: 00:21:D8:4A:02:22
                    ESSID:"eduroam"
                    Protocol:IEEE 802.11gn
                    Mode:Master
                    Frequency:2.412 GHz (Channel 1)
                    Encryption key:on
                    Bit Rates:144 Mb/s
                    Extra:wpa_ie=dd180050f20101000050f20201000050f20201000050f2010000
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (1) : TKIP
                        Authentication Suites (1) : 802.1x
                    Extra:rsn_ie=30140100000fac020100000fac040100000fac012800
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : 802.1x
                    Quality=0/100  Signal level=77/100

pelwell, 2017-04-07:
That's a GUI issue, and should be reported in the [lxpanel repo](https://github.com/raspberrypi-ui/lxpanel/).

closed 2017-04-07 by pelwell

MilanKerslager, 2017-04-07:
Filled as https://github.com/raspberrypi-ui/lxpanel/issues/12