raspberrypi/linux #1952: gpio-poweroff disables gpio3 to gnd reset

skarbat, 2017-04-07:
Hi all,

I am using the wonders of the `gpio-poweroff` module. It works wonderfully well, signaling a custom gpio pin after a poweroff.

Problem is, I am using a BCM-3 to GND with a button as well, to reboot the board from the power-off state. This was working great, but anytime I enable `gpio-poweroff` it's not working anymore, at least with gpio-26 and gpio-16.

Any ideas for a workaround? I could help with patching the driver if that would make sense.

Cheers!
Albert

JamesH65, 2017-05-19:
@pelwell For you, or better served on the forum?

pelwell, 2017-05-19:
@skarbat How do you expect the gpio-poweroff (GPO) mechanism to interact with the restart switch (RS) on GPIO 3?

During a normal power-off, the processor resets, then the bootcode running on the VPU waits in a low power state for GPIO 3 to be grounded (or for the power to be switched off).

With the GPO overlay loader, the GPO module drives a GPIO in a clear active-low pulse for 100ms (i.e. high on either side of a distinct low). The intentions is that the low causes an external device to switch off the power source. A kernel WARN is generated 

There are some obvious problems when trying to use both:

1. With GPO and an external power switch you can waggle the GPIOs all you like but it won't do anything because the processor is not powered.

2. With GPO and no external power switch the chip never resets, so RS can't do anything.

Sure we could modify GPO to force a reset in the event that the power isn't pulled, but then you are abusing the mechanism which is designed to keep the ARMs in a state where it is safe to pull the plug.

skarbat, 2017-05-23:
@pelwell that makes a lot of sense. Misunderstood the functionality alltogether. Thanks!

closed 2017-05-23 by skarbat