raspberrypi/linux #1954: Dropped RX packets - RPi 1, 2, and 3

stamster, 2017-04-08:
Hello,
I have multiple RPi's running on two locations. I noticed that each of them report dropped packets, but only in RX direction.

RPi 3:

```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:xx:xx:xx  
          inet addr:192.168.100.20  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::906e:f3:xxxx:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:275242 errors:0 dropped:15336 overruns:0 frame:0
          TX packets:71301 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:38165404 (36.3 MiB)  TX bytes:23844353 (22.7 MiB)
```

RPi 2:
Just rebooted after latest rpi-firmware patch, and only 4 mins of uptime getting: `RX packets:689 errors:0 dropped:5 overruns:0 frame:0`

RPi 1 model B:
`RX packets:647 errors:0 dropped:28 overruns:0 frame:0`

RPi 1:
`RX packets:209844 errors:0 dropped:14892 overruns:0 frame:0`

The only single RPi which does not seem to be affected by this issue is another RPi 1 model B, where I didn't upgraded firmware.
So if it's a firmware related issue, the last known good firmware/kernel w/o RX drops is this one:

```
4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux

Mar 15 2016 14:48:20 
Copyright (c) 2012 Broadcom
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 (clean) (release)
```

10 days uptime, and not a single drop:

```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:xx:xx:xx  
          inet addr:192.168.100.30  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1335268 errors:0 dropped:0 overruns:0 frame:0
          TX packets:721742 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:81649608 (77.8 MiB)  TX bytes:185224562 (176.6 MiB)
```

RPi's behave exactly the same on other networks, so it's not LAN related, since other PC's and devices do not have any drops with uptime of 250 days.

rusher81572, 2017-04-17:
I am having the exact same problem. Major amounts of dropped packets. My PiCluster is very unstable.

JamesH65, 2017-05-18:
Any relevant messages in syslog?

rusher81572, 2017-05-18:
Hello @JamesH65 . There was no relevant logs. It happens sporadically throughout the day

JamesH65, 2017-05-19:
A quick glance and some minor debugging at the driver level seems to indicate that it is not the driver itself that is dropping the packet - the code that increments the rx_dropped counter is not called. So presumably somewhere else in the stack is deciding to drop the packet. I'm not sure how to find out where in the stack though! Will investigate further.

stamster, 2017-05-21:
> It's quite possible this is harmless. The dropped packet count is not only
> used for errors, but also flagging up packets that are dropped because the
> linux stack doesn't handle them (some IPv6 stuff if you don't have IPv6
> enabled for example). This might be one of those messages. I will try and
> find out which it is.

It's not IPv6 related as first thing I usually do after installation is
disabling IPv6 on all interfaces via sysctl.conf kernel parameter.

Is it harmless? You never really know, dropped packets shouldn't be there
for sure.

JamesH65, 2017-05-22:
OK, not IPv6, but that was only given as an example of some packets that are simply discarded by the network stack. Dropped packets are not, necessarily, an error. If the stack gets a correctly formatted packet it doesn't implement, then it is dropped. It's not an error as such. So it is perfectly reasonable to be getting dropped packets and not worry about them. However, in this case I would like to know whether this is the case here.

JamesH65, 2017-05-22:
Here is some (probably quite old) text on why a packet may be dropped.

> Softnet backlog full -- (Measured from /proc/net/softnet_stat)
> 
> Bad / Unintended VLAN tags
> 
> Unknown / Unregistered protocols
> 
> IPv6 frames when the server is not configured for IPv6
> 
> If any frames meet those conditions, they are dropped before the protocol stack and the rx_dropped counter is incremented.

JamesH65, 2017-05-25:
After much faffing, I have build a utility I found called dropwatch, rebuilt the kernel to turn on a particular form of net logging and now have a list of addresses where packets are being dropped. However, cross references those addresses to the kernel isn't giving valid results, so i suspect all the drops are in modules.

What is worth noting is that the ifconfig dropped packets counter is a small subset of the actual number of packets dropped. I can see which address in my list is causing the ifconfigs drops, just need to figure out what code corresponds to that location.

JamesH65, 2017-05-25:
OK, thanks to a timely intervention from @pelwell I do have some idea of the code that is dropping the packets. The ifconfig dropped packet counter appears to be incremented in the __netif_receive_skb_core function, approx line 4214. Still some effort required in backtracking from there to determine the reason for the dropped packets.

sharky1337, 2017-05-25:
I also see these these drops on my pi3 

`pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.13-v7+ #974 SMP Wed Mar 1 20:09:48 GMT 2017 armv7l GNU/Linux`

stamster, 2017-05-26:
@JamesH65 so we can conclude issue is in kernel core? On some cloud providers I also experience dropped packets in RX line.

JamesH65, 2017-05-26:
I don't think any conclusion can be made since I don't yet know the cause of the dropped packets - it could be entirely benign. 

I think everyone sees the dropped packets on the Pi, I certainly have been seeing them for years, but ignored then.

stamster, 2017-05-26:
Well, it started to happen only after kernel/firmware update year ago or so.

On this version I have zero dropped packets:
```4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux

Mar 15 2016 14:48:20 
Copyright (c) 2012 Broadcom
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 (clean) (release)
```

Heh, it might be that this old version has a bug of not incrementing counter :laughing:

JamesH65, 2017-05-26:
Or newer version added more places where the counter could be incremented...

popcornmix, 2017-05-26:
stamster, can you identify the exact update which caused this. See:
https://github.com/Hexxeh/rpi-firmware/commits/master

If you click on each commit the end of the url contains a git hash. Run
```sudo rpi-update <hash>```
to revert back to that version. Report the first version with the packets dropped error.

(I suspect it will be one of the major bumps - e.g. the first 4.4 kernel)

stamster, 2017-07-13:
I've been super busy recently - I'll try to experiment with down-grades this weekend.

JamesH65, 2017-07-13:
I was doing some testing today with the latest Raspbian release, 5GB of data transferred with no dropped packets. This was on ethernet. If doing testing check the latest release first.

rusher81572, 2017-07-13:
I upgraded to Kernel 4.9.36-v7+ and have not seen any dropped packets yet. Almost 24 hours.

Node:rpi-1

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    59432983   260094   0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    59432983   260094   0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:97:e5:60 brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    1265482694 3594177  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    1228352705 3564027  0       0       0       0      

Node:rpi-3

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    53962647   242879   0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    53962647   242879   0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:05:d1:18 brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    2262815616 6893276  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    1418557018 5001845  0       0       0       0      

Node:rpi-5

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    714697777  1592828  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    714697777  1592828  0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:37:b0:7f brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    2818520409 5782485  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    2807188040 5674868  0       0       0       0      

Node:rpi-4

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    197693045  497290   0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    197693045  497290   0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:40:da:da brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    2265075093 6984473  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    1410030494 5277670  0       0       0       0      

Node:rpi-2

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    120126969  232051   0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    120126969  232051   0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:f8:d4:0d brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    2374566443 9304249  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    705864757  13103220 0       0       0       0      

Node:rpi-6

    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    RX: bytes  packets  errors  dropped overrun mcast   
    28360220   171077   0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    28360220   171077   0       0       0       0      
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether b8:27:eb:56:c6:b1 brd ff:ff:ff:ff:ff:ff
    RX: bytes  packets  errors  dropped overrun mcast   
    1414456938 2868572  0       0       0       0      
    TX: bytes  packets  errors  dropped carrier collsns 
    958900142  2588900  0       0       0       0      

Phillip Tribble
phillip.tribble@gmail.com

stamster, 2017-07-14:
Well, I got that kernel 5 days ago and still there were dropped packets.
Now I've fetched:
```
*** Updating kernel modules
 *** depmod 4.9.37-v7+
 *** depmod 4.9.37+
```
Let's see...

koppenho, 2017-07-16:
Let me add my stats: RX drop was about 4% with unpatched rasbian-jessie.
After updating Jessie to 4.9.37-v7+ the drop count "dropped" to 0.005%.
I think this is an improvement.

stamster, 2017-07-17:
> 4.9.37+ #1017 Thu Jul 13 11:14:43 BST 2017 armv6l GNU/Linux
> RX packets:62278 errors:0 dropped:4203 overruns:0 frame:0

JamesH65, 2017-07-20:
Hmm, just been doing some testing on this, not seeing dropped packets on the ethernet at all. It is possible that this is environmental?

```
RX packets:828974 errors:0 dropped:0 overruns:0 frame:0
TX packets:503029 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:1142250236 (1.0 GiB)  TX bytes:43037541 (41.0 MiB)```

JamesH65, 2017-07-24:
A similar issue has just come up on the linux networking list, unrelated to Pi. Suggestion is RX queueing problem.  I'm not sure how to go further here, since I cannot see any dropped packets on the ethernet connection at all. This is a continuation of the previous stats, after leaving it over the weekend, not though with anything hammering the connection.

```
RX packets:1685234 errors:0 dropped:0 overruns:0 frame:0
TX packets:515057 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:1242494098 (1.1 GiB)  TX bytes:44171593 (42.1 MiB)
```

rusher81572, 2017-07-24:
After running my 6 pi cluster for 11 days,  I got 65 dropped packets on two nodes. Much improvement

JamesH65, 2017-07-24:
@stamster Could you give me some idea of how your network is set up? Since I (nd others) am seeing no errors on the onboard Ethernet at all, it would be interesting to know if you have some odd setup that might be causing an issue.

JamesH65, 2017-09-14:
ping @stamster Are you still seeing this? If so are there any strange things on your network that might have some effect. 

Also, I'm currently using Stretch with a 4.13 kernel and seeing no drops. It would be interesting, if you have the ability to build your own kernel and install it, to see if you still see errors with the latest kernel code.

JamesH65, 2017-09-15:
I tested with over 10GB of data sent back and forth with no dropped packets at all. This is on Stretch with a 4.13 kernel (admittedly not yet a release kernel, but will be released eventually). I'm inclined, unless I hear otherwise to close this issue.

6by9, 2017-09-15:
Just a thought, but seeing as the ethernet is on USB this may be the same root cause as #2134.

That affected 4.9 (actually from 4.8), but was fixed in 4.9.43, and 4.13.
Worth retesting with the latest 4.9 kernel.

stamster, 2017-09-15:
@JamesH65 sorry I missed notifications somehow.

Well, nothing is really odd with my network(s). No smart switches, no VLAN's. 
On primary location there's a bunch of servers, so they are 24/7 and don't have any drops, even though they have large traffic (internally). 

But on another location where there is no broadcast of any kind as it is home network w/o servers etc, RPi's all have same issue.

Bottomline old kernel on Rpi B+ still have zero drops on the same network where all new kernels and RPi flavors having the drops.

JamesH65, 2017-09-15:
@stamster  Any chance of building/running the latest kernel from our github. Using that I never see issues. Which kernel version are you running at the moment on the failing systems? Newer than 4.9.43?

stamster, 2017-09-15:
ATM: 
> Linux raspberrypi 4.9.43+ #1026 Wed Aug 16 22:26:02 BST 2017 armv6l GNU/Linux

```
eth0      Link encap:Ethernet  HWaddr b8:27:eb:xx:xx:xx
          inet addr:192.168.100.3  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:788619 errors:0 dropped:41135 overruns:0 frame:0
          TX packets:135760 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:80246523 (76.5 MiB)  TX bytes:11768548 (11.2 MiB)
```

> up 28 days

stamster, 2017-09-15:
Running now: `4.9.50+ #1035 Wed Sep 13 22:53:09 BST 2017`

One minute uptime, and:

`RX packets:586 errors:0 dropped:1 overruns:0 frame:0`
:/

JamesH65, 2017-09-18:
So weird. As stated above, I have run for hours with the latest kernel, somewhat newer than yours  (4.13), no drops at all. I do wonder what the difference is. 

Is it currently good enough to wait until 4.12/13 is released and see how that performs?

stamster, 2017-09-18:
> Kernel 4.9.50+
> up 2 days, 16:16,
>         RX packets:78308 errors:0 dropped:3855 overruns:0 frame:0

> Is it currently good enough to wait until 4.12/13 is released and see how that performs?
Sure, like I have something to lose :)

JamesH65, 2017-09-18:
Well, you could try 4.13 from our repo, seems pretty stable to me. I would
be interested to know if that exhibits the issue. Meanwhile, I'll try and
find time to set up a system with that kernel to see if I can replicate.

6by9, 2017-09-18:
I'll spend a few minutes trying to track down what conditions a dropped frame gets logged.
There are various legit reasons for dropping frames, and it's possible those only occur due to a device on one of your networks. The later kernels may have actually had a fix to correctly log these events.

6by9, 2017-09-18:
So with some googling and code digging, the most useful info is https://www.novell.com/support/kb/doc.php?id=7007165

> Beginning with kernel 2.6.37, it has been changed the meaning of dropped packet count. Before, dropped packets was most likely due to an error. Now, the rx_dropped counter shows statistics for dropped frames because of:
> - Softnet backlog full -- (Measured from /proc/net/softnet_stat)
> - Bad / Unintended VLAN tags
> - Unknown / Unregistered protocols
> - IPv6 frames when the server is not configured for IPv6
>
> If any frames meet those conditions, they are dropped before the protocol stack and the rx_dropped counter is incremented.

So my reading is that those are all harmless, it is just that packets are being delivered that there are no handlers for further up the stack. Looking at the actual ethernet driver, [the only place it flags a drop](http://elixir.free-electrons.com/linux/v4.4-rc8/source/drivers/net/usb/smsc95xx.c#L1762), it also sets the rx_error flag, but you ifconfig output isn't showing that.

You say:
> But on another location where there is no broadcast of any kind as it is home network w/o servers etc, RPi's all have same issue.

No servers, but what about IOT devices such as Sonos, smart thermostats, smart lights, or similar? They all tend to stuff out random data intended only for devices on the local network for auto-discovery.
How have you determined no broadcasts? ARP is broadcast, and without that you haven't got a workable network.

It'll be a bit of a trawl, but I suspect you'll really need to take a tcpdump of the received data until you get a drop, and then look back. 3800 dropped in 78000 over two days should actually be fairly obvious.

My gut feel is that this isn't an issue in the kernel, but probably more acurate reporting. Either that or IPv6 is disabled on the new Pi's - your ifconfig output isn't listing an IPv6 address, but I don't know which level IPv6 needs to be disabled at for the packets to be flagged as dropped.

itlksez, 2017-09-18:
I'm having the same issue on 4.9.50-v7+ (and earlier) on my RPi 3.  I have IPv6 enabled and it is fully functional.  The dropped packets are not affecting performance or throughput at all as far as I can tell.   I see 2 or 3 drops per minute.

I notice, though, that whenever I run tcpdump on eth0, I never see any drops.  Based on this observation, I suspected that if I put the interface into promiscuous mode (using **sudo ifconfig eth0 promisc** and verifying with **netstat -i**), the drops would stop, but that it not the case - the drops continue whether the interface is in promiscuous mode or not (if I'm not running tcpdump).  

Also, tcpdump is supposed to put the capture interface into promiscuous mode, and you can tell it not to use promiscuous mode by using the **-p** argument.  I notice that when running tcpdump **netstat -i** does not show the P flag regardless of whether I started tcpdump with **-p**.   I see no drops when running tcpdump whether or not I use **-p**.

JamesH65, 2017-12-04:
Might be useful to test the latest kernel on this one- we have 4.14 in the repo. There have been a lot of networking changes.

JamesH65, 2018-03-07:
Has any one in the thread had any experience with the latest kernels with regard to this issue? I'm inclined to close this issue in a month or two unless there is evidence that this is still a problem.

stamster, 2018-03-11:
```RX packets:641083 errors:0 dropped:44651 overruns:0 frame:0
          TX packets:83239 errors:0 dropped:0 overruns:0 carrier:0```

`Linux raspberrypi 4.9.79+ #1091 Tue Feb 6 12:58:13 GMT 2018 armv6l GNU/Linux`

JamesH65, 2018-03-12:
Hmm, any chance you could try the 4.14 kernel? Can be installed with rpi-update, but probably not really suitable for production equipment (although we are fairly close to release)

stamster, 2018-03-12:
> uname -a; uptime; ifconfig
```
Linux raspberrypi 4.14.24+ #1097 Mon Mar 5 16:16:18 GMT 2018 armv6l GNU/Linux
 12:23:42 up 15:06,  1 user,  load average: 0.12, 0.07, 0.05
eth0      Link encap:Ethernet  HWaddr b8:27:eb:99:20:2e  
          inet addr:192.168.100.3  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13497 errors:0 dropped:905 overruns:0 frame:0
          TX packets:4459 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1440583 (1.3 MiB)  TX bytes:397404 (388.0 KiB)
```

stamster, 2018-03-12:
Once again, only single RPi flavour working great (e.g. w/o drops) on my network is still this version:

```
Linux pibplus 4.1.19+ #858 Tue Mar 15 15:52:03 GMT 2016 armv6l GNU/Linux
 12:26:55 up 169 days,  9:57,  2 users,  load average: 0.00, 0.01, 0.05
eth0      Link encap:Ethernet  HWaddr b8:27:eb:a5:f5:ce  
          inet addr:192.168.100.30  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8886984 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6771841 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:667223990 (636.3 MiB)  TX bytes:2667069239 (2.4 GiB)
```
Which aligns with all other servers and devices - network works very well.

JamesH65, 2018-03-12:
Weird. On the ones showing drops is there a decrease in overall performance? As stated above, the number of things that can cause a dropped packet has increased, so it might just be counting stuff it simply wasn't counting before but that still happened.

pelwell, 2018-03-12:
Have you tried drop_watch? https://jvns.ca/blog/2017/09/05/finding-out-where-packets-are-being-dropped/

JamesH65, 2018-03-12:
IIRC, Dropwatch does compile and run successfully on a Pi so definitely worth trying.

MrLonko, 2018-03-12:
Hi guys,

we have deployed a nice solution to our customer using RPI3 and some are on wifi... ALOT of dropped packets :(

For example 1 min uptime :(
```
:~$ ifconfig; uptime
lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:c9:2c:10
          inet addr:10.15.9.204  Bcast:10.15.9.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fec9:2c10/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:437 errors:0 dropped:166 overruns:0 frame:0
          TX packets:387 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:57522 (56.1 KiB)  TX bytes:86780 (84.7 KiB)

 17:05:09 up 1 min,  0 users,  load average: 0.34, 0.16, 0.06
```
Linux 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux

Any ideas ?
Anyone came to any conclusion ?

MrLonko, 2018-03-12:
Guys,

Just ran rpi-update on second rpi3

and result after 1 minute:
`
$ uname -a; uptime; ifconfig wlan0;
Linux coffee 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux
 17:09:23 up 1 min,  0 users,  load average: 0.21, 0.08, 0.03
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:c3:8e:ac
          inet addr:192.168.3.202  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fec3:8eac/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:604 errors:0 dropped:85 overruns:0 frame:0
          TX packets:403 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:66145 (64.5 KiB)  TX bytes:82247 (80.3 KiB)
`

Will leave it over night and see if there is still such difference.

BR,

JamesH65, 2018-03-12:
AIUI, most of the posts on this issue have been related to the onboard ethernet rather than Wifi. I suspect Wifi might be a environmental issue since we've not had reports on that before.

MrLonko, 2018-03-12:
We are using jessie lite. What do you propose? rpi-update?

With kind regards,

JamesH65, 2018-03-13:
You can upgrade to 4.9 simply using 
```
sudo apt update
sudo apt upgrade
```
rpi-update is not recommended unless you want really bleeding edge stuff, although currently it pulls in 4.14,  which has actually had a lot of testing, and is about to become the default version. When updating, always test first of a less important system just to make sure everything works as you want.

popcornmix, 2018-03-13:
On Jessie that won't get a very recent kernel.

JamesH65, 2018-03-13:
Ah, forgot was using Jessie.

sudo apt dist-upgrade instead?

pelwell, 2018-03-13:
Go straight to 4.14 with `sudo rpi-update` - I'm not aware of any issues, and as James says it will be the new standard (and best supported) kernel very soon.

MrLonko, 2018-03-13:
Is there a way to setup a local rpi-update repository as i need to update units inside a very restricted network? 
So we are in consesus :) rpi update ? I tried both and both work for me, both have now 0-tomax5 dropped count which is amazing coming down from 2/3 of all received packages...

stamster, 2018-03-13:
You mean only for kernel? What about core packages (apt)? 
IMHO your best shot for restricted networks is using HTTP proxy. So let them open you one channel only and use that as a HTTP proxy during updates. That's how I do it with Debian/GNU systems for years in the enterprise.

MrLonko, 2018-03-14:
Thank you! We got a limited time window with open network ;)

stamster, 2018-07-02:
+1 getting drops on `4.14.50+ #1122 Tue Jun 19 12:21:21 BST 2018`

`RX packets:116923 errors:0 dropped:7986 overruns:0 frame:0`

JamesH65, 2018-07-02:
Hmm. No idea what this is. Out of 4M packets sent in a currently running  test, I see 7 dropped packets. I can only think its environmental. Bad cable somewhere? Bad router? Interference?

[23 more replies]