raspberrypi/linux #1955: Non existent kernel modules requested by cups-filters

kh90909, 2017-04-09:
Quoting from: https://bugs.launchpad.net/raspbian/+bug/1518718

> There is file /etc/modules-load.d/cups-filters.conf installed by cups-filters package. Its content is following:
> 
> ~~~
> # Parallel printer driver modules loading for cups
> # LOAD_LP_MODULE was 'yes' in /etc/default/cups
> lp
> ppdev
> parport_pc
> ~~~
> 
> None of these modules exists on the RPI - it's not PC and it doesn't have parallel port. This is causing systemd-modules-load.service to fail on boot.
> 
> cups-filters-1.0.61-5+deb8u1

This causes [knock-on issues](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=174648) with iptables-persistent / netfilter-persistent.

It's not entirely clear to me that this is the right place for this issue, but a commenter on the launchpad report suggested it should be filed here. If not, feel free to close, but it would be very helpful if you could advise on the appropriate place to report this issue so that it can be fixed.
 
I personally see this issue with cups-filters-1.0.61-5+deb8u3 on kernel 4.4.38-v7+ (raspberrypi-kernel-1.20161215-1)

JamesH65, 2017-05-18:
Hmm, this is a upstream Raspbian and/or Debian packaging issue I think, so should usually be reported there. However, talking to the guy over the partition, he is quickly going to fix it. Not sure of timescale for release though.

closed 2017-05-18 by JamesH65

XECDesign, 2017-05-18:
A fixed package should be in the repo now.

Would it make sense for these modules to be enabled for USB devices which might need them or would those use different drivers?

kh90909, 2017-05-21:
Thanks for going the extra mile to get this fixed.

I'm not familiar enough to know whether it makes sense to have these modules enabled for USB devices which might need them. I guess it's a pretty narrow subset of devices...old printers that emulate a parallel port over usb?

XECDesign, 2017-05-21:
I am guessing even if those devices use those drivers, that fact that nobody has requested them, probably indicates that there's no need to worry about it.