raspberrypi/linux #1957: Please enable kernel option CONFIG_IPV6_ROUTE_INFO

heppel, 2017-04-14:
Hi,

this is a request for enhancement.

Please activate kernel option CONFIG_IPV6_ROUTE_INFO in future Raspbian kernels. Since this option depends on the CONFIG_IPV6_ROUTER_PREF option, that needs to be enabled too.

These options enable the kernel to understand additional "Route Information" options in IPv6 router advertisements. This is very helpful in complicated IPv6 environments.

Please note:

- These options are set in the default Jessie kernel for x86_64.
- Default behaviour does NOT change. Changed behaviour has to be enabled via sysctl.
- Windows accepts these additional routes by default.
- Additional resources required are minimal and limited to the ipv6 module.

If this is the wrong place for this request, please advise where to post.

Regards,
Heppel

TheDiveO, 2017-05-11:
+1, I've been hit hard when I noticed that Raspbian Linux kernels don't enable learning via MSRO (more specific IPv6 routes) in Router Advertisements. This is especially bad, as my devices are in a hierarchical network infrastructure, with nodes and multiple nodes at each level, and fully automatic learning. Without MSRO enabled, whenever an upstream router fails, communication within a hierarchy between routers fails too.

JamesH65, 2017-05-18:
Do you have any figures for impact on kernel size and performance? Given a limited resources device like the Pi we like to ensure that kernel changes for smaller use cases do not have a detrimental impact.

heppel, 2017-05-19:
The statically linked part of the kernel is not affected at all.

The ipv6 module should grow by approx. 2 kB. Dynamic data structures are only allocated if the feature is enabled via sysctl. Their size depends on the number of ipv6 routes installed via this feature. A few additional CPU cycles are spent on every ipv6 router advertisement processed by the kernel. This never happens on ipv4-only sites and typically once every few minutes on your average ipv6-enabled site. This can increase to a few packets per second on ipv6-sites using Mobile IPv6 extensions.

n1kolaa, 2017-05-20:
+1 it shoud got official ipv6 is now more used than 4

TheDiveO, 2017-05-20:
Limited resources? If these were MCUs as used in automation devices, now these are limited. But a few k, less than ten for improved next hop handling with much improved resilence against default router fails should really worth this memory.

pelwell, 2017-05-20:
I have no objections if the quoted overheads are accurate.

pelwell, 2017-05-20:
...and they do look accurate - as @heppel said, the size of Image remains unchanged (kernel*.img shrinks by a few bytes) and ipv6.ko expands by 2KB.

popcornmix, 2017-05-22:
Agreed. Added: https://github.com/raspberrypi/linux/commit/39f2896051f5c7cc733dc1e0b5791a55ceca2a8e
Will be in next rpi-update kernel.

closed 2017-05-23 by JamesH65

TheDiveO, 2017-05-23:
Thank you very much! So it eventually will get into the normal update.

popcornmix, 2017-05-26:
Should be in latest rpi-update kernel.