raspberrypi/linux #1958: Raspberry Pi Zero W Kernel

sghazagh, 2017-04-15:
Hi all,
I am not new in compiling kernels in specific for Raspberry Pi.
I have done that several times for RPi 2/3 and all have been fine.
However, all my successful compilation for Raspberry Pi Zero W do not boot the board!
I am following exactly Raspberry Kernel Compile Instruction even I have used their suggested Toolchain but the result is same.
I have used both 4.4.y and 4.9.y branches and the result were same.

I only use my own debootstrap Ubuntu 16.04.2LTS rootfs and I don't know on top of "firmware", "modules" and "kernel+device tree files and overlays" what else is required that the boot come up with following error:
```
Kernel Panic - not syncing: Attemted to kill init! exitcode=0x00000004
CPU:0 PID:1 CommL init Not Tainted 4.9.20-RPI-Linux #2
Hardware name: BCM2835
.
.
.
---[end Kernel Panic - not syncing: Attemted to kill init! exitcode=0x00000004
```
Can anyone for God sake help me please  :cry:

dlech, 2017-04-15:
Ubuntu requires armv7 (RPi 2 or 3), but RPi 0 is only armv6. Init is probably crashing because of an illegal instruction or something like that. So, you either need to use Rapsbian (works on armv6) or use RPi 2/3 if you have to have Ubuntu.

sghazagh, 2017-04-15:
Thank you for your reply.
But I have tried to use BusyBox as rootfs to test , I get the same result!

That doesn't make sense!
Is it same for busybox rootfs as well?

JamesH65, 2017-04-15:
If you install Raspbian then put your kernel on instead, does that work?

sghazagh, 2017-04-15:
No 
I replaced Raspbian kernel,device tree, overlays and Modules/Firmwares with my compiled one and that even give me the same error and stops at  boot !!!

sghazagh, 2017-04-15:
Guys, 
You know that both RPi1 and RPi Zero (W) are ARMv6.
I could bring up Raspbian on RPi1 with my own kernel about a year ago when I was playing with my old RPi 1.
Therefore, I also have tried to use my 4.9.y compiled kernel to boot my old RPi1 and it gave me the same error.

I am pretty sure there is somethings wrong with Kernel or maybe bootloader as I was able to use my own compiled version on RPi1 before and now it does not work on it either!.

sghazagh, 2017-04-17:
@dlech,
I think you were right.
I double checked again and I could boot to Raspbian with my own kernel/modules/firmware.

My compiled kernel do not boot Ubuntu rootfs and I think you already mentioned that above!
I wish I could have my Ubuntu rootfs on RPi Zero W but if it is not possible, then I stick to Raspbian...

Thanks anyway...

sghazagh, 2017-04-19:
All,
I also have tried to "debootstrap" Debian 8 Jessie for rootfs.
The result was same.
So none of my rootfs work with my kernel to boot:
Ubuntu 16.04.2
Debian 8 Jessie
Even BusyBox
My compiled kernel only works with Raspbian rootfs extracted from Raspbian Image?!
Any one knows why?

popcornmix, 2017-04-19:
Debian is built for arm v7 (so only Pi2/Pi3 can run the code).
The whole reason for raspbian existing is to rebuild debian for armv6 so it can run on Pi0/Pi1.
You can't run code compiled for armv7 on a Pi0/Pi1.

sghazagh, 2017-04-19:
Thanks for clarifying this.
I did know that the ARMv7 rootfs do not work on ARMv6 and vice versa.
But I thought Debian supports both ARMv6 and ARMv7!!!

What other option do we have for ARMv6 rootfs from most common Linux distributions???

sghazagh, 2017-04-20:
FINALLY!!!
I just have realized that ARMv6 is called "armel" in Debian.
So I just made a rootfs based on "armel" instead of "armhf" and I could boot the RPi0.

Why I was making the ARMHF in first place? because, I was confused by "hf" and the end of cross-compiler attribute for RPi0/1 Kernel.

closed 2017-04-20 by sghazagh

dlech, 2017-04-20:
Actually, [armel](https://wiki.debian.org/ArmEabiPort) is armv4t, not armv6. And it uses soft float instead of hard float. As you have seen, armel does work on all models of RPi, *but* it will run much slower because it is not taking advantage of the hardware floating point and other features of the newer ARM processors.

[Raspbian](https://www.raspbian.org/) uses the "armhf" identifier like official Debian armhf, but it is rebuilt for armv6 instead of armv7l.

You can pass the Raspbian archive URL `http://archive.raspbian.org/raspbian` as the `MIRROR` option of debootstrap to bootstrap your own Raspbian rootfs that will work on all models of RPi.