raspberrypi/linux #1959: DMA transfer error

andersthomson, 2017-04-15:
Hi,

the attached serial capture shows ERRORs in DMA transfers. A look in the source suggests that this is a known possibility, but the errors reported by em28174 at the same time indicates that the error handling does not bring the device back into working order. 

Any ideas what's causing this?

Em28174 failing ic2 writes is a sure indicator of the tv-capture usb device failing, and I've a script to cause a reboot when it occurs. It should be noted that when I only run with one tv-capture device (and not two as in this case) the rig is stable.

[log_27141.txt](https://github.com/raspberrypi/linux/files/923727/log_27141.txt)

P33M, 2017-05-18:
This is probably fixed with latest rpi-update kernel. Please retest.

JamesH65, 2017-05-22:
@andersthomson Please retest and report back with the results.

andersthomson, 2017-05-23:
Hi,

I seem unable to trigger this one again, so you can close it. I'll reopen if needed. Tx.

closed 2017-05-23 by pelwell